ghaerr/elks #1175: German Umlauts are not correctly shown

ConiKost, 2022-02-20:
**Description**
Typing any of those characters are not shown correctly.
German Umlauts: ```ä```, ```ö```,```ü```,```Ä```,```Ö```,```Ü```,
Additionally, ```$```, ```ß``` and ```~```are also wrongly shown.

**Configuration**
- Version (commit or release): Git Master
- Emulator or HW platform: NuXT

**How to reproduce ?**
Type any mentioned characters.

**Raw data**
See screenshot: https://i.imgur.com/iAvSn3F.jpg
Since the screenshot cannot replicate it: The text is white blinking.
**Additional information**
Kernelconfig:
```
#
# Automatically generated by make menuconfig: don't edit.
#

#
# Kernel & Hardware
#

#
# System
#

CONFIG_ARCH_IBMPC=y
# CONFIG_ARCH_8018X is not set
# CONFIG_ARCH_PC98 is not set
CONFIG_CPU_8086=y
# CONFIG_COMPAQ_FAST is not set
CONFIG_HW_VGA=y
CONFIG_HW_SERIAL_FIFO=y
CONFIG_HW_KEYBOARD_BIOS=y
CONFIG_HW_VIDEO_HOC=y
CONFIG_APM=y

#
# Kernel settings
#

# CONFIG_SMALL_KERNEL is not set
CONFIG_BOOTOPTS=y
# CONFIG_STRACE is not set
CONFIG_IDLE_HALT=y
# CONFIG_ROMCODE is not set
CONFIG_EXPERIMENTAL=y
CONFIG_FARTEXT_KERNEL=y

#
# Networking Support
#

CONFIG_SOCKET=y
CONFIG_NANO=y
CONFIG_INET=y
CONFIG_UNIX=y

#
# Filesystem Support
#

CONFIG_MINIX_FS=y
# CONFIG_ROMFS_FS is not set
CONFIG_FS_FAT=y
CONFIG_FS_DEV=y
# CONFIG_ROOT_READONLY is not set
# CONFIG_FS_RO is not set
CONFIG_FULL_VFS=y
CONFIG_32BIT_INODES=y
CONFIG_FS_EXTERNAL_BUFFER=y
CONFIG_FS_NR_EXT_BUFFERS=64
CONFIG_FS_XMS_BUFFER=y
CONFIG_FS_NR_XMS_BUFFERS=1024
CONFIG_FS_XMS_INT15=y
CONFIG_PIPE=y
CONFIG_EXEC_COMPRESS=y
CONFIG_EXEC_MMODEL=y
CONFIG_EXEC_MMODEL=y

#
# Drivers
#

#
# Block device drivers
#

CONFIG_BLK_DEV_BIOS=y
CONFIG_BLK_DEV_BFD=y
# CONFIG_BLK_DEV_BFD_HARD is not set
CONFIG_TRACK_CACHE=y
CONFIG_BLK_DEV_BHD=y
CONFIG_IDE_PROBE=y
# CONFIG_BLK_DEV_FD is not set
# CONFIG_BLK_DEV_HD is not set
# CONFIG_DMA is not set
CONFIG_BLK_DEV_RAM=y
CONFIG_RAMDISK_SEGMENT=0
CONFIG_RAMDISK_SECTORS=128
CONFIG_BLK_DEV_SSD_NONE=y
# CONFIG_BLK_DEV_SSD_TEST is not set
# CONFIG_BLK_DEV_SSD_SD8018X is not set
CONFIG_BLK_DEV_CHAR=y

#
# Character device drivers
#

CONFIG_CONSOLE_DIRECT=y
# CONFIG_CONSOLE_BIOS is not set
# CONFIG_CONSOLE_8018X is not set
# CONFIG_CONSOLE_HEADLESS is not set
CONFIG_KEYBOARD_SCANCODE=y
# CONFIG_CONSOLE_SERIAL is not set
CONFIG_EMUL_ANSI=y
# CONFIG_EMUL_VT52 is not set
# CONFIG_EMUL_NONE is not set
CONFIG_EMUL_ANSI_PRINTK=y
# CONFIG_KEYMAP_BE is not set
CONFIG_KEYMAP_DE=y
# CONFIG_KEYMAP_DV is not set
# CONFIG_KEYMAP_ES is not set
# CONFIG_KEYMAP_FR is not set
# CONFIG_KEYMAP_IT is not set
# CONFIG_KEYMAP_SE is not set
# CONFIG_KEYMAP_UK is not set
# CONFIG_KEYMAP_US is not set
CONFIG_CHAR_DEV_RS=y
CONFIG_CHAR_DEV_LP=y
# CONFIG_CHAR_DEV_CGATEXT is not set
CONFIG_CHAR_DEV_MEM=y
CONFIG_CHAR_DEV_MEM_PORT_READ=y
CONFIG_CHAR_DEV_MEM_PORT_WRITE=y
CONFIG_PSEUDO_TTY=y
CONFIG_DEV_META=y

#
# Network device drivers
#

CONFIG_ETH=y
CONFIG_ETH_NE2K=y
# CONFIG_ETH_WD is not set
CONFIG_ETH_BYTE_ACCESS=y

#
# Userland
#

# CONFIG_APPS_BY_IMAGESZ is not set
CONFIG_APP_ASH=y
CONFIG_APP_SASH=y
CONFIG_APP_BUSYELKS=y
CONFIG_APP_CGATEXT=y
CONFIG_APP_SCREEN=y
CONFIG_APP_CRON=y
CONFIG_APP_DISK_UTILS=y
CONFIG_APP_FILE_UTILS=y
CONFIG_APP_SH_UTILS=y
CONFIG_APP_SYS_UTILS=y
CONFIG_APP_ELVIS=y
CONFIG_APP_MINIX1=y
CONFIG_APP_MINIX2=y
CONFIG_APP_MINIX3=y
CONFIG_APP_MISC_UTILS=y
CONFIG_APP_MTOOLS=y
CONFIG_APP_NANOX=y
CONFIG_APP_OTHER=y
# CONFIG_APP_BYACC is not set
# CONFIG_APP_M4 is not set
# CONFIG_APP_TEST is not set
CONFIG_APP_KTCP=y
CONFIG_APP_MAN_PAGES=y

#
# Target image
#

CONFIG_IMG_MINIX=y
# CONFIG_IMG_FAT is not set
# CONFIG_IMG_RAW is not set
# CONFIG_IMG_ROM is not set
# CONFIG_IMG_FD2880 is not set
CONFIG_IMG_FD1440=y
# CONFIG_IMG_FD1232 is not set
# CONFIG_IMG_FD1200 is not set
# CONFIG_IMG_FD720 is not set
# CONFIG_IMG_FD360 is not set
# CONFIG_IMG_HD is not set
CONFIG_IMG_DEV=y
CONFIG_IMG_BOOT=y
CONFIG_APPS_COMPRESS=y
CONFIG_IMG_EXTRA_IMAGES=y
```

ghaerr, 2022-02-20:
Hello @ConiKost,

Are you using a german keyboard to generate these characters, and is your machine equipped with a ROM containing the associated code page for the same characters? ELKS (and the IBM PC BIOS) only uses a single code page for characters, and these may be country-specific (and thus differ across machines built in various countries).

Note: ELKS uses scan code tables (elks/arch/i86/char/KeyMaps/keys-de.h for German) to produce characters for the codepage included in German BIOS ROMs. As the source shows below, these codes may not match those of the american IBM PC:
```
/* Keymap:DE:Deutsch:German */
#if defined(CONFIG_KEYMAP_DE)

/**************************************************************
 * German Keyboard adapted from the spanish layout            *
 * by Klaus Syttkus (no steady e-mail address right now)      *
 * Some characters might not display correctly on your screen *
 * and wont do so on the elks console...                      *
 **************************************************************/
```
You might want to compare the character code(s) being generated by your keyboard, and the associated BIOS character set in the system you're testing on. The US version of the IBM PC uses codepage 437.

Thank you!

ConiKost, 2022-02-20:
Hello @ghaerr,

> Are you using a german keyboard to generate these characters, and is your machine equipped with a ROM containing the associated code page for the same characters? ELKS (and the IBM PC BIOS) only [...]

I am using a real german keyboard. If I do connect it to my daily Linux/Windows box, all chars work. So I don't think, it's a keyboard problem.

> You might want to compare the character code(s) being generated by your keyboard, and the associated BIOS character set in the system you're testing on. The US version of the IBM PC uses codepage 437.

Do I have a chance to print those charcodes on ELKS directly?

ghaerr, 2022-02-21:
> So I don't think, it's a keyboard problem.

What I am saying is that the key-de.h file above translates the keyboard scancodes to a transmitted byte value. If the transmitted value from the keystroke causes the remote system to use the character with the intention desired, then the keyboard translation file is correct and doesn't need to be modified. However, that's an entirely different thing than having the system display a keyboard character or received character properly. ELKS does not do any character or code page translation, and the direct console just writes they byte value into the machine's text display buffer without translation.

> On DOS on NuXT, it's also all fine.

DOS performs codepage translations, so we would need to know the codepage it's configured as in order to determine how this might be working on DOS.

> Do I have a chance to print those charcodes on ELKS directly?

You mean display the keyboard characters correctly? ELKS doesn't perform any scan code <-> BIOS character ROM translation, it uses the hardware text mode (BIOS character ROM) in order to display any console characters in text mode.  Without knowing the NuXT character ROM codepage (or whether there are variations of them), it is hard to comment on why this isn't working, but I suspect its a codepage mismatch.

I will have to ask some of our international users how their systems are configured to properly display byte codes generated from non-US keyboards (ELKS key-xx.h scan code translation files) to hardware text mode without output translation.

Code page translation can get messy fast. In order to proceed, we need to know more about your machine's input (keyboard) and output (display) code pages, and the desired codepage translation. The solution, if coded, would likely be a 128-byte translation array for character values from 80h through FFh, using some known codepage translation.

ConiKost, 2022-02-21:
> ELKS does not do any character or code page translation, and the direct console just writes they byte value into the machine's text display buffer without translation.

Ah, okay, that makes it more clear for me.

> DOS performs codepage translations, so we would need to know the codepage it's configured as in order to determine how this might be working on DOS.

All I know, that my german DOS uses Codepage 850.

> Without knowing the NuXT character ROM codepage (or whether there are variations of them), it is hard to comment on why this isn't working, but I suspect its a codepage mismatch.

Where is such character ROM stored? Video card?

ghaerr, 2022-02-24:
Hello @ConiKost,

After researching more about this issue, it seems this is not likely a bug, but rather a possible need for enhancements, depending on whether the hardware actually can support it.

It appears that DOS (and MINIX) have the ability to load a font file into the video card, which allows for a different display of a glyph than the hard-coded character glyph in ROM somewhere. I'm pretty sure your actual problem is that you're wanting to display codepage 850 on a NuXt machine that has codepage 437 in ROM. I don't know what codepage the ELKS german keyboard implements.

> Where is such character ROM stored? Video card?

You'll have to research that for your machine. And I am wondering whether the EGA/CGA cards that ELKS supports allow for RAM font loading, versus VGA, where it is functional. The technical specifications for your machine should show how a soft character set might work and what the load address (usually 4K bytes of 8x16 characters) would be.

Thank you!

ConiKost, 2023-08-03:
Sad to hear :-/

ghaerr, 2023-08-03:
> Sad to hear :-/

Yes, ELKS pretty much expects code page 437 for its keyboard input and console output. While it's possible someone might contribute translation code pages and new scan code tables for the keyboard and console drivers, these days it would probably be a better idea to design UTF-8 input/output and corresponding translation tables for unicode.

You're welcome to contribute an updated optional scancode table for German keyboards, which would allow for typed-in characters to display properly (but only on systems that have both German keyboards and CP 850 BIOS character ROM glyphs). Any files produced using this format would be CP 850-specific, and would not display properly on other ELKS systems, however. ELKS currently displays properly on systems with German keyboards and CP 437 (US IBM PC) BIOS character ROMs.

ConiKost, 2023-08-06:
> ou're welcome to contribute an updated optional scancode table for German keyboards, which would allow for typed-in characters to display properly (but only on systems that have both German keyboards and CP 850 BIOS character ROM glyphs). 

Well, I guess, this won't help, FWICS, no video cards exists with CP850.

ghaerr, 2023-08-06:
> Well, I guess, this won't help, FWICS, no video cards exists with CP850.

OK, I see. Thus it seems that in order to support this feature, the video card will need to support soft character sets and the character glyphs will have to be replaced from CP 437 to CP 850, just like DOS apparently does. Then, the ELKS keyboard table would have to be switched to an 850-compatible set when the feature is active. It is interesting to compare this to the current day, when almost no systems use hard character glyphs anymore - even text consoles are almost always emulated graphically. I suppose soft character sets in video cards was one of the first steps in this movement.

ConiKost, 2023-08-06:
Well, maybe someone will able to provide something in future for that :-)