ghaerr/elks #1196: [net] Attempting TCP access outside LAN directly after boot hangs system

ghaerr, 2022-03-01:
Opening new issue from discussion in [Add DNS nameserver #1160](https://github.com/jbruchon/elks/commit/f429953537cc603bfde43a05735b841e405ac3a6#commitcomment-67564843).

Issues are:
- Attempting TCP connect outside of LAN directly after boot hangs system on real hardware. However, works if a LAN host is connected to first, and on QEMU.
- On connect outside of LAN, router blocks traffic and replies with ICMP packet "Destination not reachable". Problem is that TCP level connect doesn't know about it, and waits 10 seconds before returning.

ghaerr, 2022-03-01:
Hello @Mellvik,

> Turns out that telnet to a random (presumably nonexistent) IP address also cause a hard hang.
> So - like you alluded to yesterday - this has nothing to do with DNS

I think the problem is likely not that of connecting to a random or nonexistent IP address, but rather, what's happening under the hood when ktcp tries to send a packet outside the local area network directly after boot, which is hanging the system. After a connection to a host on the LAN, then there seem to be no problems connecting outside the LAN (which is what nslookup does talking to a DNS server). Since this is all handled under-the-hood with ARP requests and replies, we need to see what's happening there, using 'arp' and a packet analyzer, if possible.

> telnet 1.1.1.1
hangs hard and is obviously an external address,
telnet 192.168.10.48
works fine - even though it is external to the ELKS LAN (10.0.2.x) … (it's my LAN; the other side of the Raspi ).

When you say "hangs hard", I assume that you've tried accessing the real hardware console and can't login, the system is crashed, correct? You can't ^C from the serial port, nor does login continue running on the console. This is distinct from a hard ktcp hang, which would still allow console logins and processes to run.

> but (I am guessing) in the code that replaces the destination address with the gateway address.

There is no code that replaces the destination address with the gateway address; the packet addresses are never rewritten. Instead, I'm thinking the problem may be in the link layer code - ktcp uses ARP to route the original packet to the gateway, I presume. Thus, if the ARP table or mechanism is initialized incorrectly, this could be the cause. We are going to need to see debug information on ARP and Ethernet packets.

When ktcp tries to send an IP packet on ethernet, its lower level writes an ARP request on ethernet and waits for a reply. We had early problems with this, and ktcp can hang waiting for a reply, that's why I'm concerned whether a full system hang, or just a ktcp hang (of all network programs) is occurring. Ktcp does some funny business with holding and dropping IP packets, while waiting for an ARP response. I'm guessing the problem lies near here.

However, you're saying that ktcp is always sending a single IP packet (a TCP SYN connect packet), so it seems at least the initial ARP is working, although we may need to see the actual ethernet packet header(s). What we don't know are what subsequent responses may be occurring. These could be 1) an ICMP response from your router that ktcp isn't treating properly for some reason, 2) an additional ARP reply that is damaging the internal ARP table for some reason, or 3) perhaps a different ethernet packet being received that is corrupting ktcp.

I suggest turning on DEBUG_ARP and DEBUG_ETH, doing ^P, then net start, then run 'arp', then the failing app. Then do it all over again, but connect to the LAN, then run 'arp' to see what the ARP table looks like and any differences found, along with an external packet trace, if possible.

Thank you!

ghaerr, 2022-03-01:
> On connect outside of LAN, router blocks traffic and replies with ICMP packet "Destination not reachable". Problem is that TCP level connect doesn't know about it, and waits 10 seconds before returning.

Earlier, I thought we might be able to shorten the 10 second connect timer in this case, where an ICMP packet is returned, rather than a TCP RST (which will always be the case if the remote host actually gets a TCP SYN with no server listening on the requested port).

I no longer think we can fix this (easily) - the problem is that the ICMP layer knows nothing about what TCP connects may be outstanding, and can't communicate "upstairs" to a particular TCP control block, because it doesn't have the TCP endpoints (src and dest IP addr), which the TCP level requires. If the ICMP return destination unreachable packet contains the TCP source and destination addresses, then this could be made to work.

We need to do some more research on what TCP stacks do when receiving an ICMP response, and how that gets tied to the IP layer, and then the TCP layer...

Mellvik, 2022-03-01:
@ghaerr,

> I no longer think we can fix this (easily) - the problem is that the ICMP layer knows nothing about what TCP connects may be outstanding, and can't communicate "upstairs" to a particular TCP control [...]
The ICMP error reply package contains the IP header of the packet causing the error (actually, in many cases there is more - at least a few bytes from the rest of the packet), so I think your initial thinking about how to handle this was OK.

—M

***@***.***:/home/helge# tcpdump -X -i wlan0 host ams.mymayday.com
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
10:03:46.653248 IP 192.168.10.12.33226 > ams.mymayday.com.telnet: Flags [S], seq 4159433141, win 29200, options [mss 1460,sackOK,TS val 2045249765 ecr 0,nop,wscale 7], length 0
	0x0000:  4510 003c c92b 4000 4006 f844 c0a8 0a0c  ***@***.***@..D....
	0x0010:  b23e fc48 81ca 0017 f7eb e9b5 0000 0000  .>.H............
	0x0020:  a002 7210 7663 0000 0204 05b4 0402 080a  ..r.vc..........
	0x0030:  79e8 08e5 0000 0000 0103 0307            y...........
10:03:46.695981 IP ams.mymayday.com > 192.168.10.12: ICMP host ams.mymayday.com unreachable - admin prohibited, length 68
	0x0000:  4500 0058 aba3 0000 3501 60c6 b23e fc48  E..X....5.`..>.H
	0x0010:  c0a8 0a0c 030a 7660 0000 0000 4500 003c  ......v`....E..<
	0x0020:  c92b 4000 3306 0555 c0a8 0a0c b23e fc48  .***@***.***>.H
	0x0030:  81ca 0017 f7eb e9b5 0000 0000 a002 7210  ..............r.
	0x0040:  7663 0000 0204 05b4 0402 080a 79e8 08e5  vc..........y...
	0x0050:  0000 0000 0103 0307                      ........

Mellvik, 2022-03-01:
@ghaerr,

this is very inefficient communication, we need to find a better way to do this.

When I include the steps of the debugging in a problem report, the purpose is to eliminate extra rounds of questions so as to speed up the process. Also, it is under the assumption that you read the entire message before responding.

Further, there is no way we're always going to use the same expressions and terms and terminology. That would be very efficient but not possible. So - in order to communicate with reasonable efficiency we need to make some assumptions about the knowledge level of the individual on the other end. I think we've acquired that over the past few years, and attempt to practice accordingly.

IOW (and a great example) - it's a waste of time to explain client routing mechanisms to someone who's been teaching and writing books on TCP and IP for 30+ years.

We can do better than this.

-M

ghaerr, 2022-03-01:
Hello @Mellvik,

> Hopefully you can make some sense of this [@ghaerr](https://github.com/ghaerr) --
> We can do better than this.

I am sorry - just trying to help. In this case, my (typically) long explanation about "under the hood stuff" was to detail a more specific list of things of my knowledge of ktcp that might cause it  to hang the system, not to educate you about TCP/IP. Added detail was included to clarify that our ktcp doesn't rewrite packets, as was suggested may be the cause.

> Also, it is under the assumption that you read the entire message before responding.

I had read your response quite a few times and am struggling to come up with why the system may be hanging. Your initial theory was the problem likely in C library, while I'm guessing now likely in link layer.

> this is very inefficient communication

This seems about par with many networking problems we've solved - lots of back and forth has been required to actually determine the cause. Frankly, a single IP-level packet trace showing one outgoing packet and no reply is not enough information for my to identify the problem, and we've both been guessing.

Thank you!

ghaerr, 2022-03-01:
> The ICMP error reply package contains the IP header of the packet causing the error (actually, in many cases there is more - at least a few bytes from the rest of the packet)

Thanks - with the IP src and dest addresses, the TCP control block can be found and the kernel notified, resulting in an immediate connect error return to the application. I'll get that done.

ghaerr, 2022-03-03:
Hello @Mellvik,

> On connect outside of LAN, router blocks traffic and replies with ICMP packet "Destination not reachable". Problem is that TCP level connect doesn't know about it, and waits 10 seconds before returning.

I think I have a way to code up ktcp's ICMP processing to make this work, but I don't know how to get my router, or any external host, to issue this ICMP message. Do you know a way to do that?

It would be nice to be able to this test before posting a PR, or post an untested PR and have you test on your LAN ?

Thank you!

Mellvik, 2022-03-03:
> telnet ams.mymayday.com
> telnet pogostick.net
>
> Thanks. I don't have any working ELKS ethernet over here (never did get a card working for my desktop 386). So I have to test on QEMU. From my laptop running QEMU, both of these connections return immediately with "Connection Refused", meaning that we got a TCP RST, rather than ICMP packet in response.
>
> I suppose I'll go ahead and post a possibly fix, with debug information, enabling you to test, so we can wrap this up.

OK; it seems QEMU is translating the response, not very surprising.
It also probably explains why the hard hang on external first address doesn't show up.

How's your Portable 386 doing BTW?

Yes, I'll test, no problem.

-M

ghaerr, 2022-03-04:
> How's your Portable 386 doing BTW?

It's doing great, I have two now actually, but only got the first one working with a new CMOS battery and (temp) floppy emulator. I haven't figured which card I might insert into the expansion chassis for networking, so it's still off any network. I use a different machine for SLIP testing.

> It seems QEMU is translating the response, not very surprising.

I somehow doubt that QEMU slirp would be rewriting an ICMP packet into a TCP RST... ? But who knows. I hope to push a PR for your testing shortly.

> It also probably explains why the hard hang on external first address doesn't show up.

Could be - but I'm thinking these two issues are unrelated - the first issue doesn't generate an ICMP packet, just hangs connect for reasons yet completely unknown, the other is a request to terminate a connection on an ICMP dest unreachable packet.

Mellvik, 2022-03-04:
> How's your Portable 386 doing BTW?
>
> It's doing great, I have two now actually, but only got the first one working with a new CMOS battery and (temp) floppy emulator. I haven't figured which card I [...]
OK; let me be the money on your back on this one: I seem to remember seeing on a picture a ne2k card kust like one of mine, with the 3 leds on top towards the connector end of the card? And possibly even a wd8003 card from some stash?
My memory has been found to have holes recently (), so it's quite possible these images come from somewhere else??

—M

ghaerr, 2022-03-08:
Moving conversation from https://github.com/jbruchon/elks/pull/1207#issuecomment-1061767868 back to here.

Following is @Mellvik's comment:

I finally narrowed down this one. More specifically - I found the reason, not the solution.

The most unlikely of possible culprits: sync.

Commenting out SYNC in `/bootopts` - or setting SYNC=0  eliminates the problem.
Needless to say - it surely beats me. Unless somehow `net_port` gets overwritten in `init` which would surely result in a hard hang. Then again, why would that happen only if some network actions happen in a certain sequence.

It seems more likely that changing `sync_interval` in `init` somehow triggers a bug in the ne2k driver, which I haven't found yet. I'm still looking. The placement of the variables in `init` makes me suspicious.

Still looking for that one.

ghaerr, 2022-03-08:
Hello @Mellvik,

Are you saying that when a "sync=<nonzero>" is set in /bootopts, a telnet connect outside your LAN hangs the system (first time only), while, with the same setting, a telnet connect within your LAN always works?

Definitely strange. What is the sync value, is it 30? Have 30 seconds elapsed before the first-try failure? Perhaps set sync=360 to understand whether a firing `alarm` has anything to do with this.

If a local LAN connection always works, I would say that sync is unlikely the problem, and that we may need an ARP and ETH trace in order to further debug.

ghaerr, 2022-03-08:
Given what you're saying, it seems possible that `alarm` is buggy, although I studied it again and can't see how.

It might be useful to display when an alarm is occurring (in init, even though somehow it might be interrupting another process, say ktcp). Or, set the sync/alarm value to a bigger number, to see what that does.

The file elks/kernel/sys2.c has a debug statement that could be changed to display when the alarm is occurring:
```
static void alarm_callback(int data)
{
        struct task_struct *p = (struct task_struct *)data;

        debug("kernel ALARM pid %d for %d\n", current->pid, p->pid);
        send_sig(SIGALRM, p, 1);
}
```
This will display the pid for the process that happens to be running when the alarm fire, and the pid that the alarm is for (which should be 1).

Mellvik, 2022-03-08:
Thanks @ghaerr - 
yes, that's a good idea! I'll start off with that.

-M

Mellvik, 2022-03-08:
OK; this may be interesting.

I enabled the suggested debug output and set SYNC=10

The output (regularly) is like this

ELKS built from commit 0ce5063a
Tue Mar 08 18:44:09 2022
sys_alarm 0
sys_alarm 0

sys_alarm 0
sys_alarm 10

ELKS 0.5.0

login: kernel ALARM pid 0 for 1
sys_alarm 10
kernel ALARM pid 0 for 1
sys_alarm 10
kernel ALARM pid 0 for 1
sys_alarm 10
kernel ALARM pid 0 for 1
sys_alarm 10
kernel ALARM pid 0 for 1
sys_alarm 10

Then, immediately after the templet command that would normally hang (and still does),
I get sys_alarm 2

Then hang. Looks like this with my extra debug output:

# telnet kernel ALARM pid 0 for 1
sys_alarm 10
ams.mymayday.com
sys_alarm 2
enter eth_route
eth_write ok
Exit eth_route1 — first return from eth_route
<— HANG

Before this I made a telnet connection into the ELKS system to see if the alarm output was stable. It was, always 'sys_alarm 10'.

@ghaerr - does the 'sys_alarm 2' tell you something?

—M

ghaerr, 2022-03-08:
> does the 'sys_alarm 2' tell you something?

The sys_alarm 2 is the alarm(2) system call in_resolv(), the DNS Resolver library routine, is doing before it connects to DNS to resolve your ams.mymayday.com name. It appears that the DNS connect is not returning.

I would recommend we leave out DNS from testing, and only use numeric addresses, to keep things simple. You had suggested that the problem could be duplicated by accessing a numeric address outside the LAN (while any inside always worked).

> enter eth_route
eth_write ok
Exit eth_route1 — first return from eth_route
<— HANG

This apparently is the DNS packet going out, then system hang, exactly like the telnet to a numeric non-LAN address does.
Thus, we're no farther ahead in knowledge, because that's a description of the original problem - a hang on a non-LAN address.

Please test larger sync numbers (=180 or higher) to ascertain whether it is the sync setting, or the alarm itself causing problems.
Also please confirm that telnet to numeric local-LAN addresses always work, regardless of sync settings.
Setting DEBUG_ARP and DEBUG_ETH will really help so we can see what's happening on the wire prior to hang.

Thank you!

ghaerr, 2022-03-08:
I just realized something - the current implementation of alarm only allows for a single alarm to be active at a time - and we have two, a telnet connect, and the init sync timer, when enabled. In this implementation, any subsequent sys_alarm call will overwrite the previous value.

SO - what I think is happening is not necessarily a system hang, but possibly just a connect to a system (which is always off-LAN, since local-LAN always respond) which doesn't return from a connect - because there is no response from that system. That is occurring because the telnet alarm call is being replaced by another init alarm call, nullifying the telnet alarm.

This doesn't answer why you're saying that the system is totally hung, rather than just not returning from connect, like it used to, but nonetheless I think this is the problem.

I will think about how the kernel can implement multiple alarms simultaneously, since it seems we need it!

Thanks for your debugging!

ghaerr, 2022-03-08:
@Mellvik - I have confirmed the above to be the problem, and in addition, for reasons yet unknown, the kernel is entering a tight loop sending the SIGALRM signal to telnet, effectively hanging the system.

Fix coming, after I've figured out exactly why the signal loop is occurring.

Mellvik, 2022-03-08:
That's excellent news @ghaerr.
I was about to write up a detailed descriptioon of the events before the hang, which now seems moot. I have been barking up the wrong tree(s), one of them the ARP angle: It turned out that the last packet out before the hang was always from ARP, either an ARP reply or a delayed packet. Kept me busy for a while :-)

Thank you!!

-M

closed 2022-03-15 by ghaerr