ghaerr/elks #1199: IBM PS/2 L40 SX does not finish booting

eloydegen, 2022-03-02:
**Description**
The IBM L40 SX does not finish the boot sequence

**Configuration**
- Version (commit or release): v0.5.0
- Emulator or HW platform: https://en.wikipedia.org/wiki/IBM_PS/2_L40_SX with 6 megabytes RAM 

**How to reproduce?**
The problems always happens on my L40 SX. To reproduce, `dd` the `fd1440-fat.img` onto the floppy and press F1 to boot from it.

**Picture of the laptop when it stops**
[image]

ghaerr, 2022-03-02:
Hello @eloydegen,

Hmmm, I wonder if this is because of an XMS incompatibility on the PS/2? (XMS buffers were erroneously left turned on in the v.0.5.0 release).

Can you try this [fd1440-fat.img](https://github.com/jbruchon/elks/files/8173579/fd1440-fat.img.zip) and see how that works? It has CONFIG_FS_XMS_BUFFER turned off.

Thank you!

eloydegen, 2022-03-14:
Hi @ghaerr, that fixed the issue! Thank you