ghaerr/elks #1201: [FAT] slow ls on FAT

Mellvik, 2022-03-03:
We've visited this issue before, and this is more of a test report than a problem - at this time.

A directory in a mount FAT file system, approx 500MB in size, contains some 250 files.
If that directory is the current directory, `ls` takes 14.5 seconds to run. Slow but not all that bad.
If the current directory is one above, say `ls zz`, the time almost quadruples, to 42 seconds.
Going one more level up, the time is about the same, but `ls` is having (non-fatal) `malloc` problems (this is with the old heap setting).

What's surprising is that the listing from `ls` is sorted, which I read as "all filenames must have been read into ls before the sort", but the output is line-oriented, meaning one-line-then-wait-5-secs-new-line-5-secs etc. How's this possible unless the sorting is in the filesystem itself?

For the record, `ls /bin` on the Minix boot floppy takes 0.7 seconds with 106 files, and seems to gain significantly from the XMS cache, while the FAT directory access does not seem to be affected by the cache at all.

Again, this is a test report, not a bug report.

-M

```
# time ls
1000b     1001b     1024b     1025b     1200b     128b      1459b     
1999b     1byte     2000b     2001b     511b      513b      599b      
600b      601b      767b      768b      769b      999b      arp       
aslookup  aup       banner    basename  bc        cal       cat       
chmod     chown     cksum     clear     clock     console   cp        
cut       date      dd        decomp16  df        diff      dirname   
du        echo      ed        env       false     fdisk     fdtest    
fgrep     file      find      fsck      ftp       ftp.c     ftpd      
ftpd.c    ftpd.o    ftpget    ftpput    grep      hd        head      
httpd     httpget   init      kill      kilo      ktcp      ln        
login     logname   lp        ls        makeboot  makefile  man       
meminfo   mesg      mined     miniterm  mkdir     mkfifo    mkfs      
mknod     more      mount     mouse     mv        net       netstat   
nslookup  od        p         partype   passwd    poweroff  pp        
printenv  proto     ps        pwd       pwdauth   reboot    remsync   
rm        rmdir     sash      sed       sercat    sh        shutdown  
sleep     sort      sum       sync      synctree  sys       tail      
tar       tee       telnet    telnetd   test      tget      time      
tinyirc   touch     tr        true      tty       umount    uniq      
unreal16  urlget    uudecode  uuencode  v         vi        vim       
wc        which     who       whoami    x         x1        x2        
x4        x5        xab       xac       xacc      xad       xae       
xaf       xag       xah       xaj       xak       xal       xam       
xan       xao       xap       xaq       xar       xargs     xas       
xat       xau       xav       xaw       xax       xaz       xba       
xbb       xbc       xbd       xbe       xbf       xbg       xbh       
xbi       xbj       xbk       xbl       xbm       xbn       xbo       
xbp       xbq       xbr       xbs       xbt       xbu       xbv       
xbw       xbx       xby       xbz       xca       xcb       xcd       
xce       xcf       xch       xci       xcj       xck       xcl       
xcm       xcn       xco       xcp       xcq       xcs       xct       
xcu       xcv       xcw       xcx       xcy       xcz       xda       
xdb       xdc       xdd       xde       xdf       xdg       xdh       
xdi       xdj       xdk       xdl       xdm       xdmm      xdn       
xdo       xdp       xdq       xdq1      xdr       xds       xdt       
xdu       xdv       xdvdr     xftp      xx        xxx       xxxx      
yes       yyy       yyyxxx    yyyy      yyy  

Real    0m14.630s
# cd ..
# time ls zz
1000b     1001b     1024b     1025b     1200b     128b      1459b     
1999b     1byte     2000b     2001b     511b      513b      599b      
600b      601b      767b      768b      769b      999b      arp       
aslookup  aup       banner    basename  bc        cal       cat       
chmod     chown     cksum     clear     clock     console   cp        
cut       date      dd        decomp16  df        diff      dirname   
du        echo      ed        env       false     fdisk     fdtest    
fgrep     file      find      fsck      ftp       ftp.c     ftpd      
ftpd.c    ftpd.o    ftpget    ftpput    grep      hd        head      
httpd     httpget   init      kill      kilo      ktcp      ln        
login     logname   lp        ls        makeboot  makefile  man       
meminfo   mesg      mined     miniterm  mkdir     mkfifo    mkfs      
[... 53 lines cut ...]
sleep     sort      sum       sync      synctree  sys       tail      
tar       tee       telnet    telnetd   test      tget      time      
tinyirc   touch     tr        true      tty       umount    uniq      
unreal16  urlget    uudecode  uuencode  v         vi        vim       
wc        which     who       whoami    x         x1        x2        
x4        x5        xab       xac       xacc      xad       xae       
xaf       xag       xah       xaj       xak       xal       xam       
xan       xao       xap       xaq       xar       xargs     xas       
xat       xau       xav       xaw       xax       xaz       xba       
xbb       xbc       xbd       xbe       xbf       xbg       xbh       
xbi       xbj       xbk       xbl       xbm       xbn       xbo       
xbp       xbq       xbr       xbs       xbt       xbu       xbv       
xbw       xbx       xby       xbz       xca       xcb       xcd       
xce       xcf       xch       xci       xcj       xck       xcl       
xcm       xcn       xco       xcp       xcq       xcs       xct       
xcu       xcv       xcw       xcx       xcy       xcz       xda       
xdb       xdc       xdd       xde       xdf       xdg       xdh       
xdi       xdj       xdk       xdl       xdm       xdmm      xdn       
xdo       xdp       xdq       xdq1      xdr       xds       xdt       
xdu       xdv       xdvdr     xftp      xx        xxx       xxxx      
yes       yyy       yyyxxx    yyyy      yyy  

Real    0m44.320s
#
```

Mellvik, 2022-03-03:
Additional observation - only somewhat related:

Root can be unmounted at any time - with obvious consequences ...

ghaerr, 2022-03-03:
> ls /bin on the Minix boot floppy takes 0.7 seconds with 106 files, and seems to gain significantly from the XMS cache, while the FAT directory access does not seem to be affected by the cache at all.

> If that directory is the current directory, ls takes 14.5 seconds to run. Slow but not all that bad.

> If the current directory is one above, say ls zz, the time almost quadruples, to 42 seconds.

Explaining what is likely the issue behind these three slightly different scenarios is a bit complicated, but 1) the FAT filesystem uses internal buffers extensively for directory searches of which there are only 12. In addition, directory searches always start from the beginning and involve lots of EXT/XMS buffers (or new disk reads) being copied into L1 buffers every time, also starting from the beginning of the directory, every time. 2) In the third case above, the internal FAT "namei" routine, which converts a string to a filename or directory, is doing double-duty since it has to essentially chase down another path segment, doing the same thing (non-cached) directory search to find "zz" (again), then searching each file within that directory, all over again, all uncached.

> "all filenames must have been read into ls before the sort", but the output is line-oriented, meaning one-line-then-wait-5-secs-new-line-5-secs etc. How's this possible unless the sorting is in the filesystem itself?

There are cases where `ls` does a `stat` system call (e.g. ls -l) on each line, after sorting the list, in order to get more details. This could be the reason, as the stat call ends up doing a full-path "namei" as described above.

The FAT filesystem implementation doesn't attempt any caching, and relies on L1 buffers (of which are extremely limited) in order to do almost all its directory work. Any directory outside the root directory will be slower.

> Root can be unmounted at any time - with obvious consequences ...

Not quite following you here: are you saying `umount /` is causing a problem?

Mellvik, 2022-03-03:
> 
> Root can be unmounted at any time - with obvious consequences ...
> 
> Not quite following you here: are you saying umount / is causing a problem?
> 
> 
The issue is that a file system with open files should not umount under any circumstance (which IIRC works for filesystems mounted into the root file system, like on /usr or /mnt).
And since the root fs will always have open files, it should block umounts regardless.

So 'umount /' should always return File System Busy IMHO. The system crashes immediately after the command.

—M

ghaerr, 2022-03-03:
> The issue is that a file system with open files should not umount under any circumstance (which IIRC works for filesystems mounted into the root file system, like on /usr or /mnt).
And since the root fs will always have open files, it should block umounts regardless.

We do support the unmounting of the root filesystem using `mount -o remount,ro", (and rw) which allows for `fsck` on the root filesystem. In all cases, a sync is performed beforehand. I'm pretty sure even this variant would be disallowed if any files are open rw on the device.

> So 'umount /' should always return File System Busy IMHO. The system crashes immediately after the command.

Agreed, this is a new bug - it didn't used to allow it. I will add it to the list. I'm wondering whether this happens on MINIX root as well, but will find out.

Mellvik, 2022-03-03:
Thanks @ghaerr,
> ls /bin on the Minix boot floppy takes 0.7 seconds with 106 files, and seems to gain significantly from the XMS cache, while the FAT directory access does not seem to be affected by the cache at [...]
OK, I think 'it's complicated' is a reasonable  characterization. Anyway there seem to be two optimizations possible w/o going into the OS code (i.e. in ls): 1) cwd into the directory being listed - while listing it, and 2) skip the extra stat after the names have been sorted and rearranged. I don't see how this can do any good except trying to detect files removed during sort, which isn't very likely on ELKS; and no catastrophe if it happened.

If ls w/o options could run as fast for the rest of the listing as it does for the first 4-5 lines (20 or so files), we'd be good. And if that seed could be maintained even if we're listing a subdirectory, we'd be even 'gooder'.

—M

ghaerr, 2022-03-03:
> skip the extra stat after the names have been sorted and rearranged. 

Yes, it already should and may be - but `ls` itself is, shall we say, "overly complicated". Adding the seemingly simple '-t' option required quite a bit of thought, which is where I observed the stat issue(s) in the first place. I've added checking that to my list.

> If ls w/o options could run as fast for the rest of the listing as it does for the first 4-5 lines (20 or so files), we'd be good. 

As mentioned above, the slowness is likely due to the FAT filesystem code and it's having to "linux-ify" an otherwise alien filesystem, so I'm not optimistic here. FAT directories are fundamentally different than MINIX/UNIX/Linux filesystem directories, and ELKS just doesn't have the L1 buffer space that 32-bit systems do.

> And if that seed could be maintained even if we're listing a subdirectory, we'd be even 'gooder'.

Also not likely to change, IMO.

I'll look at `ls`, otherwise recommendation for ELKS use of FAT filesystems is: use small directories.

ghaerr, 2022-03-03:
> Case in point: (386/20, HD is a 500MB CF card)

Wow, that is very slow! I see what you mean about slowing down rapidly per line of `ls` output. Not really sure exactly what that is. Can you upload a copy of that CF card image, so I can reproduce on QEMU?

Mellvik, 2022-03-03:
> 
> 
> Case in point: (386/20, HD is a 500MB CF card)
> 
> Wow, that is very slow! I see what you mean about slowing down rapidly per line of ls output. Not really sure exactly what that is. Can you upload a copy of that CF card image, so I can reproduce on QEMU?
> 
> 
Will do that - it'll take some time though. I need to dump the image via ethernet, the card is inside the machine, not accessible.
No problem - it's done in a few hours and is always a good test of ELKS and KTCP.

Available tomorrow morning.

And yes, it's the slowing down that makes me curious. Looks like a loop that is recursive by accident.

—M

toncho11, 2022-03-04:
I agree that ls output is very slow on fat on a 8086 8 MHz. Every line takes like a 0.8 seconds to print. Can this be related to the way ls calculates how to format the output on the console?

Mellvik, 2022-03-04:
@ghaerr,
here's the image - bzip2 compressed.

https://www.icloud.com/iclouddrive/07bbWIwZnWWhpgLVCEs82zteA#cpq-hda

ghaerr, 2022-03-04:
> here's the image - bzip2 compressed.

Thanks! We'll try to unfold the mystery of why you, and now also @toncho11, are getting very slow output on FAT ls...

ghaerr, 2022-03-04:
I am testing using the FAT filesystem, but QEMU is so fast that, although the lines are output a bit slower than normal, I can't see any particular increasing slowdown on a per-line basis, as is shown in @Mellvik's video. However, surprisingly, `ls` gets an error, about 1 out of 6 tries, during `ls zz`, where an invalid filename is returned from the kernel, seemingly:
```
# ls zz
1000b     1001b     1023b     1024b     1025b     1100b     1200b     
128b      1459b     1999b     1byte     2000b     2001b     511b      
512b      513b      599b      600b      601b      767b      768b      
769b      999b      arp       aslookup  aup       banner    basename  
bc        cal       cat       chgrp     chmem     chmod     chown     
cksum     clear     clock     cmp       console   cp        cut       
date      dd        decomp16  df        diff      dirname   du        
echo      ed        env       false     fdisk     fdtest    fgrep     
file      find      fsck      ftp       ftp.c     ftp.o     ftpd      
ftpd.c    ftpd.o    ftpget    ftpput    getty     grep      hd        
head      httpd     httpget   init      kill      kilo      ktcp      
ln        login     logname   lp        ls        makeboot  makefile  
man       meminfo   mesg      mined     miniterm  mkdir     mkfat     
mkfifo    mkfs      mknod     more      mount     mouse     mv        
net       netstat   nslookup  od        p         partype   passwd    
poweroff  pp        printenv  proto     ps        pwd       pwdauth   
ramdisk   reboot    remsync   rm        rmdir     sash      sed       
sercat    sh        shutdown  sleep     sort      stty      sum       
sync      synctree  sys       tail      tar       tee       telnet    
telnetd   test      tget      time      tinyirc   touch     tr        
true      tty       umount    uniq      unreal16  urlget    uudecode  
uuencode  v         vi        vim       wc        which     who       
whoami    x         x1        x2        x3        x4        x5        
xaa       xab       xac       xacc      xad       xae       xaf       
xag       xah       xai       xaj       xak       xal       xam       
xan       xao       xap       xaq       xar       xargs     xas       
xat       xau       xav       xaw       xax       xay       xaz       
xba       xbb       xbc       xbd       xbe       xbf       xbg       
xbh       xbi       xbj       xbk       xbl       xbm       xbn       
xbo       xbp       xbq       xbr       xbs       xbt       xbu       
xbv       xbw       xbx       xby       xbz       xca       xcb       
xcc       xcd       xce       xcf       xcg       xch       xci       
xcj       xck       xcl       xcm       xcn       xco       xcp       
xcq       xcr       xcs       xct       xcu       xcv       xcw       
xcx       xcy       xcz       xda       xdb       xdc       xdd       
|q?q: No such file or directory 
xde       xdf       xdg       xdi       
# 
```
That's a definite problem, likely in the kernel, which we'll now have to chase down. @Mellvik, it would be interesting if you found that duplicating on real hardware. It happens infrequently on QEMU, but is repeatable.

I'm working on figuring a way to slow QEMU down enough to effect a way to determine where the slowdown is on FAT... any ideas?

ghaerr, 2022-03-04:
More input on `ls` on large FAT directories: it seems the above problem is within ls, not a kernel issue. Messing around with `ls` heap space caused an outright crash when set to 28000, which is very strange. When setting to max, using the following line, no errors occur:
```
       $(LD) $(LDFLAGS) -maout-heap=0xffff -o ls ls.o $(LDLIBS)
```
@Mellvik, it may be worth checking to see whether the slowdown is within `ls` itself (possibly the result of a large number of internal reallocs?), by setting max heap as above, and retrying.

Mellvik, 2022-03-04:
Very interesting  @ghaerr,
How does ls in sash compare?
I'll test in the morning.

Btw, and off topic, I've been getting '............4!' boot failures on physical fd1440 images recently, thusly falling behind. The images work on qemu and virtualbox. Any clues?

-M

Mellvik, 2022-03-04:
> xcq       xcr       xcs       xct       xcu       xcv       xcw
> xcx       xcy       xcz       xda       xdb       xdc       xdd
> |q?q: No such file or directory
> xde       xdf [...]
Very interesting indeed! No, I haven't seen this one, and I've been listing this directory like foerever. Looks like a memory leak/buffer overflow,doesn't it?
> I'm working on figuring a way to slow QEMU down enough to effect a way to determine where the slowdown is on FAT... any ideas?
> 
Not really, but now that I have virtualbox up with elks, I'll check it out...

-m
>

closed 2022-03-05 by Mellvik