ghaerr/elks #1203: mount issues

Mellvik, 2022-03-05:
Two issues popped up today while testing a newly partitioned HD (tested on VirtualBox only).

Using the same 500MB disk image that was shared the other day, now repartitioned into 4 partitions.

The first issue is minor, but possibly the trigger for the second. The new `mount` command lists mounted file systems and available space, something we've been missing forever. Now here's the thing (see the attached screenshot): `/dev/hda4` is always listed as `/dev/hda`, regardless of the sequence of mounts.

The second issue is more severe: When mounting filesystem #5, the system hangs hard. No warning, no message, just full stop. Maybe 4 mounts is max but it does need to be caught and reported somehow.

[image]

An additional issue, unrelated to `mount`, but instead related to `fsck`: Just like `mount` above, `fsck` doesn't like ´/dev/hda4`, but seem to work (or at least accept) the other file systems. See screenshot.

[image]

ghaerr, 2022-03-05:
Hello @Mellvik,

Thanks for the testing!

> /dev/hda4 is always listed as /dev/hda, regardless of the sequence of mounts.

That's a bug, I'll fix it.

> Maybe 4 mounts is max but it does need to be caught and reported somehow.

The variable NR_SUPER in include/linuxmt/fs.h is currently set to 4, should we increase it? It does use kernel data segment space, but perhaps a change to 5 would help with small cost. I'll fix the bug in the kernel not rejecting the mount when over the max allowed. Good catch!

> Just like mount above, fsck doesn't like ´/dev/hda4`, but seem to work (or at least accept) the other file systems

This is a different problem than mount, but nonetheless should work. Could you package up this 4-partitioned hard drive as an image and post it, so I could look more into this? Not sure why this is failing at this point. (And/or give the commands used to partition and mkfs it, or both :)

Thank you!

Mellvik, 2022-03-05:
Thank you @ghaerr,
> Maybe 4 mounts is max but it does need to be caught and reported somehow.
>
> The variable NR_SUPER in include/linuxmt/fs.h is currently set to 4, should we increase it? It does use kernel data segment [...]
If the cost seems reasonable, an increase to 5 would be good for testing, thank you.
> Just like mount above, fsck doesn't like ´/dev/hda4`, but seem to work (or at least accept) the other file systems
>
> This is a different problem than mount, but nonetheless should work. Could you [...]
I'll create a link to the image, no problem. there is no 10 hour ehther tranfer this time :-) .
It's actually a repartitioned copy of the image you got the other day.
-> make a copy of the image
-> boot the first image in qemu or virtual box (DOS 6.22), so that the copy is the D: drive
-> use fdisk, select D drive
-> delete the single partition
-> create new primary DOS partition , size approx (500-(64x3))MB

Exit DOS, boot ELKS with the repartitioned drive as HD, boot from floppy, use ELKS fdisk to create the 3 partitions to use for minix file systems.
 
That's about it. 

Link coming.

-M

Mellvik, 2022-03-05:
https://www.icloud.com/iclouddrive/0b9eGO3PbtKNeOcRsYMyIGgbg#greg-cpq-split

Mellvik, 2022-03-05:
Thank you @ghaerr -

this seems to be the day I run into all kinds of odd problems, and instead of creating another issue I'll continue on this one.

I've had trouble building a bootable fd1440 image for some time so I've been experimenting with configurations.
Among other things I accidentally enabled ROM FS, which turns out to cause compilation failures. Possibly known, even intentional, but I figured I'll mention it anyway.

The other oddity I ran across was that if '/dev on FAT' is disabled, XMS buffers will not be included.

Fortunately I now get bootable images and I still don't know that caused the problem (I got ……4! during boot — are these codes documented somewhere btw?).

-M

ghaerr, 2022-03-05:
As mentioned in #1204, I've now fixed what can be fixed in this issue. I'll commit that and you can test tomorrow, so we can close this issue.

>  I accidentally enabled ROM FS, which turns out to cause compilation failures.

I'll check this - it can only work on embedded systems, which are defined in emu86*.config.

> if '/dev on FAT' is disabled, XMS buffers will not be included.

Noted - Added to list and will check separately.

> I got ……4! during boot — are these codes documented somewhere btw?).

They're in elks/boot/boot_err.h. Possibly not documented very well, added to list todo when I have time.

closed 2022-03-15 by ghaerr