ghaerr/elks #1231: Incomplete build instructions (where is the cross-compiler?)

sebras, 2022-03-17:
The [wiki](https://github.com/jbruchon/elks/wiki/) has an item for "Toolchain" under "Builnding" which doesn't link anywhere, and "Building" itself links to a paragraph ending with "Full instructions are [here](https://github.com/jbruchon/elks/wiki/#Building)." linking back to itself. I'm having using this information to locate the necessary 8086 cross-compiler to build ELKS. Do you mind completing the build instructions? This is a small issue, but adding a link to compiler ought to be easy enough. :)

ghaerr, 2022-03-17:
Hi @sebras,

It appears the Wiki has a bad link, I'll fix that. The main page has full build instructions, which point to [here](https://github.com/jbruchon/elks/blob/master/BUILD.md).

Thank you!

sebras, 2022-03-17:
Ah, thank you!

closed 2022-03-19 by ghaerr