ghaerr/elks #1233: Add more artifacts (floppy images)

toncho11, 2022-03-18:
Can you please add more artifacts after each PR ?

Now I see that only the 1440 elks floppy image is available. I suppose it is the minix. 
Can we add more?
- 1440 fat
- 720 KB
- 360 KB 
- HD

In ideal case for both minix and FAT. 

This will make testing much easier. I particularly interested in the 720 KB image as I managed to put a 720 KB floppy on my IBM 5160.

Mellvik, 2022-03-18:
Out of curiosity, @toncho11 - 

why would you want a 1440k FAT image? Or a FAT boot floppy of any size?

-M

toncho11, 2022-03-18:
Hello FAT hater :) (just joking)

I am on Windows. FAT is important, so that I am able to add files to ELKS. My hard drive is a sd card and I usually do modifications to an ELKS installation. Or I verify that everything is OK. If it is a minix then I do not know of any tool on Windows that can access minix FS.

Anyway FAT allows for a number of things to be tested and mainly that support for other FSs is available in general.

ghaerr, 2022-03-18:
> Can you please add more artifacts after each PR ?

I assume you're talking about the artifacts produced by the automatic make occurring on Github? I didn't realize that one could actually grab the results of those makes...

> Can we add more?

I think we should be able to add a "cd images; make images", which would make all the sizes you're requesting. However, the FAT images require that 'mtools-4.0.23' (the DOS disk utility tools) are installed on github's VM. If that's not the case, it will fail, and I don't have a Linux box to test what the 'apt-get' command is to download that toolset. Can someone try that?

toncho11, 2022-03-18:
I think it should be OK. 

https://packages.ubuntu.com/focal/mtools 
It is available on for Ubuntu 20.04LTS

ghaerr, 2022-03-18:
> It is available on for Ubuntu 20.04LTS

Thanks @toncho11, I'll try an update on the auto-make later today. How do you grab the images produced?

toncho11, 2022-03-18:
I would go to github.com and click on "Actions" and then "Workflows" and then I click on the last successful "Run".

toncho11, 2022-03-18:
OK so you need to update the main.yaml file from

`run: 'sudo apt install bison flex texinfo libncurses5-dev libelf-dev'`

to 

`run: 'sudo apt install bison flex texinfo libncurses5-dev libelf-dev mtools'`

toncho11, 2022-03-19:
Thanks @ghaerr !

closed 2022-03-19 by toncho11