ghaerr/elks #1238: NEC PC-98 enhancements to ELKS

ghaerr, 2022-03-19:
Continued conversation from #1047.

tyama501, 2022-03-19:
Hello @ghaerr ,

I modified bios_disk_rw() and simplified bios1B.S as follows.
https://github.com/tyama501/elks/commit/ce55cce26d305cc78d5fee0aef1cced5a8919fcf

With this modification, I could read/write the partition > cylinder 1023.

So the next step is to modify hd_drive_map?

Thank you!

ghaerr, 2022-03-19:
Hello @tyama501,

> I modified bios_disk_rw() and simplified bios1B.S as follows.

Nice simplification of bios1B.S. I think, once hd_drive_map[] is modified, then your current changes to bioshd.c can also be simplified, so that hard-coded 0xA0, 0x90, 0x30 won't be required, but looked up calling map_drive.

After the full modifications, I assume in line 173 you won't be comparing against an IBM PC BIOS value of 0x80, but instead using the PC-98 PDA value. This could possibly be changed by calling map_drive at a later point.

> So the next step is to modify hd_drive_map?

Yes, I think so.

Thank you!

tyama501, 2022-04-02:
Hello @ghaerr ,

I think I am almost done except for getting boot partition.
Here is my modified code.
https://github.com/tyama501/elks/compare/master...merge_pc98

It seems I need to modify the following part of drivers/block/genhd.c to set boot_partition which will be used in bioshd_conv_bios_drive.
Do you have any idea what is the right way to set for this "if" condition? 

    if (ROOT_DEV == (0x80 | minor >> hd->minor_shift)) {
        sector_t boot_start = SETUP_PART_OFFSETLO | (sector_t) SETUP_PART_OFFSETHI << 16;
        if (start == boot_start)
            boot_partition = minor;
    }

[image]

fd is bootable.
[image]

ghaerr, 2022-04-02:
Hello @tyama501,

> I think I am almost done except for getting boot partition.
Here is my modified code.
https://github.com/tyama501/elks/compare/master...merge_pc98

Your changes look very nice. Perhaps use "<< 9" instead of "* 512" on lines 184, etc, as the emitted code should be smaller.

In line 186, (sector - 1) could result in 0xFFFF if sector == 0, you may need to say (sector - 1) & 0xFF?

> Do you have any idea what is the right way to set for this "if" condition?

I think it should just use 0xA0 which is the PC-98 BIOS value for hard disk, rather than IBM PC's 0x80. Thus just the first line would change to:
```
if (ROOT_DEV == (0xA0 | minor >> hd->minor_shift)) {
```
This could possibly be done with a \#define BIOS_HD_DRVNUM 0xA0 (or 0x80) for PC.

This should hopefully work, since it appears from your first screenshot above that BIOS drive 0xA1 is resulting in root device 0x320, when it should be 0x321.

ghaerr, 2022-04-02:
Also, the following code is not really correct:
```
#ifdef CONFIG_ARCH_PC98
    for (minor = 0; minor < NUM_DRIVES; minor++) {
	if (biosdrive == hd_drive_map[minor]) break;
    }
    ...
```
It has two issues: first, it calculates "minor" by looping through up to 8 times, while minor is only valid when 0 <= minor <= 3.
Thus, minor probably needs to be "anded with 3" outside the loop. The problem is that minor is only really useful for hard drive SCSI mapping, so perhaps only loop through 4 times.

Secondly, there is the possibly that a match isn't found (unlikely), in which case minor would be out of range.

tyama501, 2022-04-03:
Thank you @ghaerr ,

I modified like this but it seems there are a couple of problems.

#ifdef CONFIG_ARCH_PC98
    printk("\nROOT_DEV %x, minor %d, shift %d, start %ld\n", ROOT_DEV, minor, hd->minor_shift, start);
    printk("OFFSETLO %d, OFFSETHI %d\n", SETUP_PART_OFFSETLO, SETUP_PART_OFFSETHI);
    if (ROOT_DEV == (0xA0 | minor >> hd->minor_shift)) {
#else
    if (ROOT_DEV == (0x80 | minor >> hd->minor_shift)) {
#endif
        sector_t boot_start = SETUP_PART_OFFSETLO | (sector_t) SETUP_PART_OFFSETHI << 16;
        if (start == boot_start)
            boot_partition = minor;
    }

(1) SCSI_ID = 1 for boot device and there is no SCSI_ID = 0 device

ROOT_DEV = 0xA1 and  (0xA0 | minor >> hd->minor_shift) = 0xA0 so it cannot set boot_partition

[image]

(2) SCSI_ID = 1 for boot device and there is also SCSI_ID = 0 device 

ROOT_DEV = 0xA1 and (0xA0 | minor >> hd->minor_shift) = 0xA1 so it can set boot_partition
but root_device is incorrect.
( It should be 0x321)

[image]

(3) SETUP_PART_OFFSETLO/SETUP_PART_OFFSETHI

Where the SETUP_PART_OFFSETLO/SETUP_PART_OFFSETHI values are set?
Although 200 is correct value for the first partition but I couldn't find where it is set.

BTW, I purchased PC-9801BX and waiting to arrive.
It has intel 486 and can use IDE hard drive, 1.44MB floppy.
Although the IDE hard drive has been removed so I need to get one,
we can consider supporting IDE hard drive which has device address 0x80 later.

ghaerr, 2022-04-03:
Hello @tyama501,

I see, I was incorrect earlier. I'm thinking the code should read something like the following, which would use the BIOS drive mapping table to map the minor number to the BIOS PDA, which would then be compared to the boot PDA:
```
if (ROOT_DEV == (hd_drive_map[minor >> hd->minor_shift])) {
```
In this case, when ROOT_DEV (the boot BIOS drive ID) is 0xA1, it would associate that with the minor number that matches its map location, and things should work. (hd_drive_map will have to be declared external and not static). It seems this should work for both IBM PC and PC-98 without modification.

> Where the SETUP_PART_OFFSETLO/SETUP_PART_OFFSETHI values are set?
Although 200 is correct value for the first partition but I couldn't find where it is set.

These are set in the last lines of code in the FAT boot loader sect_boot_fat.h (part_offset, +2, previously stored as sect_offset, +2).

tyama501, 2022-04-06:
Hello @ghaerr ,

The modification worked for the (1) case. It could boot from 0xA1.

[image]

The (2) still have wrong root_dev which is 0x341.
I added debug message and found it comes from the code below.

bioshd.c
kdev_t INITPROC bioshd_conv_bios_drive(unsigned int biosdrive)
{
...
    if ((biosdrive & 0xF0) == 0xA0)
        partition = boot_partition;
...
    return MKDEV(BIOSHD_MAJOR, (minor << MINOR_SHIFT) + partition); (<----- Here minor = 1, and partition = 33 (0x21))
}

Should the boot_partition be 0x01 instead of 0x21?

[image]

Thank you.

tyama501, 2022-04-09:
Add codes for bios_disk_rw, Simplified int1B, Modified bootcode for PC-98 #1251

Thank you for the merging!

tyama501, 2022-04-09:
Hello @ghaerr ,

I found you have added nice basic.

PC-98 has graphic routine called LIO In ROM which originally used for BASIC.

I have written ASM codes for LSI-C before to use these directly as following.
https://github.com/tyama501/CLIO98

Is it interesting if I rewrote these codes with gcc-ia86 asm and use from your basic?

ghaerr, 2022-04-09:
Hello @tyama501,

> Is it interesting if I rewrote these codes with gcc-ia86 asm and use from your basic?

Sure, why not?! It seems everyone can have fun with an old-fashioned basic interpreter. I know I have :)

> I have written ASM codes for LSI-C before to use these directly as following.

Perhaps you can use an approach similar to the INT 1Bh C->ASM interface so that most of the code stays in C, rather than ASM. You could also pass the interrupt number itself and use self-modifying code to change the INT to what it needs to be (A7h, ACh, etc). Or keep it all in assembly.

Having graphics on PC98 (and IBM PC) would be fun. If you want to jump into this, I would suggest that you keep the BASIC statements such as PLOT and DRAW (in this case) compatible with the original Sinclair BASIC, by reading the [manual](https://worldofspectrum.net/ZXBasicManual/zxmanchap9.html), implementing the BASIC parsing in basic.c, but calling out to the specific host implementation using new host_xxx routines only, like it does now for all external interfacing. This keeps the core interpreter very portable.

@cocus has already added code to play with buttons and lights on his board, as you may have [seen](https://www.youtube.com/watch?v=XvHsd9JyTIc). I am thinking that instead of keeping that code in host.c, we could create platform-specific .c files like host-8018x.c and host-pc98.c for hardware interfacing. The Makefile could then control which files get linked in.

I plan on continuing to make a few changes to make basic.c more compatible with some Sinclair BASIC, so that we might be able to run more of the text and graphics demos available from https://github.com/ZXDunny/SpecBAS-Demos. However, those demos use a non-Dartmouth BASIC method fo string manipulation (see Chapter 8 in manual), which is different than the LEFT$/RIGHT$/MID$ method which is already implemented. So we have already departed from strict compatibility from Sinclair BASIC.

Thank you!

tyama501, 2022-04-09:
Hello @ghaerr ,

Thank you for the information.

> already added code to play with buttons and lights on his board, as you may have [seen](https://www.youtube.com/watch?v=XvHsd9JyTIc).
 
Wow, great video.

It seems the manual web page has security warning from mcafee web advisor and I couldn't see.
So I searched another website. Maybe I can reference these  site.

Thank you!

https://en.wikipedia.org/wiki/Sinclair_BASIC#Interpreters_for_the_ZX_Spectrum_family

https://worldofspectrum.org/ZXBasicManual/

ghaerr, 2022-04-09:
Yes those references look great. I have reviewed the various Sinclair documentation I've found and will update the elkscmd/basic/README.md with a list of implemented commands and functions, as well as differences between our basic and Sinclair, and a list of so-far unimplemented commands and functions. We can then use that as our reference for syntax.

For graphics modes, we will need commands to enter and exit graphics mode(s). I will attempt to add them to the README.md as well.

On another note, I tested your PC-98 changes for determining boot partition on IBM PC, and the ifdef in genhd.c will not be needed, we can use your new code. I will change this in a small update PR with the basic readme.

tyama501, 2022-04-10:
Hello @ghaerr ,

I got PC-9801BX(intel 486) and I could boot it with FD1440 so I attached movie.
This is the first time I tried FD1440 in the real PC.
The drive can read 1440/1232 so maybe I will think how to probing the disk after I play with "basic" :)

The PC also support IDE hard drive but it has been removed so I need to get one.

https://user-images.githubusercontent.com/61556504/162612668-766c19e6-5dd9-4951-ad24-8b09c71fe4f8.mp4

Thank you!

ghaerr, 2022-04-10:
Hello @tyama501,

Thanks for the movie, super nice ELKS is working well on your new system!

> The PC also support IDE hard drive but it has been removed so I need to get one.

Your recent work on keeping physical device address (PDA) in hd_drive_map should work well for IDE hard drive, with the following changes possibly needed:
- need slightly different query routine for IDE drives, and set hd_drive_map with PDA 0x80+drive.
- drive query routine should probably be made a function and passed either IDE or SCSI PDA (0x80 or 0xA0)
- a few places in bioshd.c that check for "((drive & 0xF0) == 0xA0) should instead check instead for "(drive & 0x80)" which will work for both SCSCI and IDE drive PDAs.
- possibly sector numbering change, only decrement sector number if ((drive & 0xF) == 0xA0).

> after I play with "basic" :)

I've added all the new functions I planned on, so you should be good to go for adding graphics commands/functions if you like. Our [documentation](https://github.com/jbruchon/elks/blob/master/elkscmd/basic/README.md) now shows possible syntax for unimplemented commands and functions. It shows a (non-standard) MODE command that will probably be needed in ELKS to switch between text and graphics modes, whose BIOS number will likely be different between PC-98 and IBM PC.

Thank you!

tyama501, 2022-04-16:
Hello @ghaerr ,

Thank you for the comments.

I have started to add host-pc98.c as follows but it seems I got memory problem.
https://github.com/tyama501/elks/commit/4116b4354f37e669f9359e20f6eab5daabb8bf45

LIO needs about 5.2KByte working memory but it seems failed to allocate.
[image]

If I decreased to 4KByte it can be allocated but If I tried PLOT x,y the stack overflow occurred.
Are they too big memory to handle for ELKS?

[image]

ghaerr, 2022-04-16:
Hello @tyama501,

> LIO needs about 5.2KByte working memory but it seems failed to allocate.
> If I decreased to 4KByte it can be allocated but If I tried PLOT x,y the stack overflow occurred.
> Are they too big memory to handle for ELKS?

Your program seems fine - the problem is that the standard heap allocation for ELKS programs is 4K (with an additional 4K default for stack). This should be easily fixed by adding `-maout-heap=8192` (or similar, depending on requirements) to the Makefile link line. You can see this is done for many other programs in elkscmd/*/Makefile. A stack value can be increased or decreased from the default using `-maout-stack=`.

You can also modify a programs heap or stack request at ELKS runtime by using `chmem -h 8192 /bin/basic`. Running `chmem /bin/basic` will show the current allocation, with 0 meaning default.

Thank you!

ghaerr, 2022-04-16:
> I have started to add host-pc98.c as follows 

Looks good. I can see we're going to need a better way of adding commands and functions to basic.c/basic.h. I am working on adding READ/DATA/RESTORE, and you've got PLOT and LIO98INI, using conditional compilation. The interpreter wants to have a sequential list in basic.h, so that would mean conditional compilation would always have to come last, which will likely turn into a problem.

I'll try to think up a better answer for this, perhaps adding the ability for the tokenizer to use seperate (conditionally compiled) token tables. Even with my recent work on READ/DATA etc, I'm finding having to match the token name/position/number with basic.h a bit tedious. 

Another discussion point, do you like the idea of having a separate LIO98INI statement, or should basic just automatically initialize this when compiled in? Does this command change the screen into graphics mode? If so, perhaps we should use a generic "MODE [n]" or "GRAPHICS [n]" statement, which sets one of several possible graphics modes. On the IBM PC, there are a number of possible BIOS graphics modes.

Thank you!

tyama501, 2022-04-16:
Thank you @ghaerr ,

After -maout-heap=8192, the PLOT worked at least for the emulator.
( For real PC bios I might need to add some interrupt handler that is called from inside the bios )

[image]

>Does this command change the screen into graphics mode? If so, perhaps we should use a generic "MODE [n]" or "GRAPHICS [n]" statement, which sets one of several possible graphics modes. 

I am not sure it is called graphic mode but I used int A1/A2/A3 to set 640x400 screen, views, 16 colors.
Maybe we can move these to "MODE [n]" or "GRAPHICS [n]".

Thank you!

tyama501, 2022-04-16:
Oh, is there plan to add "INK" command.

Since there is no INK command I used white color.

ghaerr, 2022-04-16:
> I am not sure it is called graphic mode but I used int A1/A2/A3 to set 640x400 screen, views, 16 colors.
Maybe we can move these to "MODE [n]" or "GRAPHICS [n]".

Yes, go ahead and rename your LIO98INI command to MODE for the time being. It can be (later?) enhanced to take a number argument, which would be used to switch between the "normal, text" mode and perhaps several graphics modes. We could use MODE 0 to mean normal text, and any other number to be system specific. (IBM PC has BIOS text modes of 3 and 7, for instance, and other numbers could be used to select various resolutions and/or colors). You can use whatever numbering you would like for selecting the PC-98 graphics mode(s).

On the IBM PC, BASIC will probably have to switch back to MODE 0 before exiting, so that the shell prompt can be seen. This can all be worked out in more detail later.

> Oh, is there plan to add "INK" command?

It looks like it is needed, if you like go ahead and implement "INK <number>", used to select the foreground drawing color.

For back porting into the source tree, I will try to devise a separate token array for these graphics commands, which will make it easier for you to maintain this work, as other non-graphics commands are added to the original token array. If each new token is associated with a separately numbered TOKENGR_xxx and associated hostgr_xxx, then we won't have to ifdef each of the calls in the main interpreter. Instead, the second token array will always be compiled in, and the interpreter will call out to each hostgr_xxx routine, which will be stubbed out in host-stubs.c for those systems that do not include graphics (like the IBM PC initially). The TOKENGR_xxx numbering could start at a fixed value, say 128, to keep things portable.

tyama501, 2022-04-17:
Hello @ghaerr ,

>On the IBM PC, BASIC will probably have to switch back to MODE 0 before exiting, so that the shell prompt can be seen. This can all be worked out in more detail later.

Actually PC-98 text vram plane and graphic vram plane are separated and
text is overlaid on the graphic so there is no text mode.
Maybe we need to clear graphic plane when exiting basic.

I will rename to MODE but setting interrupts and allocating memory should be done only once so I will check if it has been done.

I also need to add intC5 interrupt handler since the real PC LIO Bios is calling this periodically.
The stack overflow I mentioned before was occurring because I was using real PC Bios data and there was no handler.
(NekoProject emulator has the host LIO function so after I removed the data it worked)

I think I have read somewhere that this handler is prepared to handle escape sequence like Ctrl-C but 
maybe just iret is ok for now.

Do you think implementing the handler can be done with
__attribute__((interrupt)) void interrupt_handler(struct interrupt_frame* frame){}
according to the following?
If there is more good way please tell me.

https://wiki.osdev.org/Interrupt_Service_Routines#GCC_.2F_G.2B.2B
 
Thank you.

ghaerr, 2022-04-17:
Hello @tyama501,

> text is overlaid on the graphic so there is no text mode.
Maybe we need to clear graphic plane when exiting basic.

That's ok, I think better to just leave screen as-is when exiting. At least, only MODE command should change screen. For PC-98, looks like no changes will be required. We can worry about IBM PC at time someone implements that.

> I think I have read somewhere that this handler is prepared to handle escape sequence like Ctrl-C but
maybe just iret is ok for now.

I'm not following you on the ^C issue - the ELKS kernel should handle all keyboard I/O, and we need to be careful that anything added to BASIC does not interfere with what kernel expects to be handling.

> I also need to add intC5 interrupt handler since the real PC LIO Bios is calling this periodically.

That should be fine, as long as interrupt handling is only related to LIO and/or BIOS callbacks, which are never handled by kernel. Of course, should BASIC exit or crash without resetting that interrupt vector the system will be in an unstable state. This means an atexit() handler should probably be used to ensure vectors are reset.

> Do you think implementing the handler can be done with
attribute((interrupt)) void interrupt_handler(struct interrupt_frame* frame){}
according to the following?

No - I seem to remember the big problem with using C for an actual interrupt handler is that GCC always assumes that SS=DS on interrupt entry, as it sometimes generates code using (BX) or (BP) to access C variables, and each use DS and SS register respectively. Much better to write your interrupt handler in ASM, and add to project. You will need to be very careful in handler if accessing C variables for the same reason. The handler cannot make any assumptions about any register contents, and should not switch stacks, as the interrupt may occur even when BASIC is switched out by kernel and not running. My recomendation is to push as few registers as possible, and IRET as soon as possible. We can talk more about this after I understand exactly what handler has to do.

Also, when interrupt vectors are set or restored, CLI/STI should be performed for safety. We are running as user program, not as kernel, so things are definitely messier.

Thank you!

cocus, 2022-04-19:
Regarding the BASIC keywords, I checked and I think we can split them to a different source file, but the problem is the actual parsing of these keywords. For instance, we can extend the keyword array somehow (like making each source file to declare an array that lands on a specific program segment of the elf, and let the base code traverse thru all of them on runtime), but it would be quite inconvenient to do this for the parsing of arguments. It might be doable of we expect an api from the platform it'll be compiled against, some sort of hooks into the appropriate base functions of the basic interpreter.
Worst case scenario would be to flood the source with ifdefs.

I'm still a little bit curious about the video initialization stuff on the pc98 and why you can't re-initialize if needed. As @ghaerr  mentioned, if this is somehow used, it'll have to revert back to the previous non-overlaid graphics on top of text mode when breaking or exiting the interpreter

ghaerr, 2022-04-19:
> I checked and I think we can split them to a different source file, but the problem is the actual parsing of these keywords.

Agreed. I've come to the conclusion that it likely isn't worth the effort to try to split the parsing apart, and that the keywords can stay compiled in with no ifdefs, as well. We don't have enough complexity to justify writing all that code. Frankly, the interpreter is just beginning to be actually useful. (Although I've also realized there is no such thing as a truly portable BASIC program).

> Worst case scenario would be to flood the source with ifdefs.

Actually, I think the simplest and best answer is to no use ifdefs at all, for the keywords nor the parsing (or the host_xxx routines). Have the parsed command call the host_xxx function, which will be outside basic.c. The Makefile can then control whether a stub function is called (which could display an error or not implemented), or the platform-specific function used. This way, the interpreter recognizes and tokenizes all possible keywords, which would allow those BASIC programs to at least be read in and LISTed. Having different versions of the interpreter not even recognize ELKS-specific platform command and function variations will lead to a mess.

cocus, 2022-04-19:
> Actually, I think the simplest and best answer is to no use ifdefs at all, for the keywords nor the parsing (or the host_xxx routines). Have the parsed command call the host_xxx function, which will [...]

Oh, I think stubbing the functions not used on the other platforms would make a lot more sense. Not from the actual user perspective, but oh well!

ghaerr, 2022-04-19:
> Not from the actual user perspective, but oh well!

What should the stubbed function do? Display an error or ignore? On or the other would effectively be the same as if the command/function were not compiled in, from the user perspective.

cocus, 2022-04-19:
> What should the stubbed function do? Display an error or ignore? On or the other would effectively be the same as if the command/function were not compiled in, from the user perspective.

Yes, stating that keyword isn't valid should be enough

tyama501, 2022-04-19:
Hello @ghaerr , @cocus ,

I'm not a native English speaker so maybe I need some time to understand the discussion.

So the conclusion for now is not to use ifdef but use stabbud function?

I also thinking move codes of LIO98INIT to MODE1 and use MODE0 to free memory after I implemented the interrupt handler.

Thank you.

tyama501, 2022-04-19:
No, it is slow.

There is GET/ PUT function.
If we capture region to memory with GET,
PUT may be faster.

ghaerr, 2022-04-20:
Hello @tyama501,

It would be interesting to know whether most of the speed problem is within the BASIC interpreter, or the graphics draw routine(s). I picked this particular interpreter because it tokenized input and was able to run using a small amount of fixed memory. The interpreter should be fairly fast, but there may be ways to greatly increase the speed - for instance by allowing computation in integers in some cases instead of floats. The PLOT function has to convert from float to integer for every operating, and all other calculations are done in (software) float.

> If we capture region to memory with GET,
PUT may be faster.

Is this a PC-98 graphics issue? I wasn't able to read any of the documentation on PC-98 LIO.

Thank you!

tyama501, 2022-04-20:
Hello @ghaerr ,

I will take video with the real PC so you can judge whether the drawing is slow or not but somehow I'm still have trouble setting the interrupt table for iret asm code.

I wil try set it from C code later.

tyama501, 2022-04-22:
Hello @ghaerr ,

I still suffering with STACK OVER MINSTACK  when I used real LIO Rom and I'm not sure it is related to the interrupt related issue.

I traced kernel stack but it seems ok so this STACK is different with ks right?

I increased to -maout-stack=16384 but it seems no difference.

Is there anyway to debug?

[image]

[image]

Somehow I couldn't link .global asm fucntion so I wrote C code for temporary.

void int1c_handler(void)
{
    printf("int1C\n");
    __asm__ volatile ("iret");
}

    // Set interrupt handler for 0x1C
    //int1c_set();
    int1c_h_p = int1c_handler;
    intvec = (unsigned long __far *) 0x00000314;
    *intvec = (unsigned long) ((unsigned long __far *) int1c_h_p);

Thank you.

tyama501, 2022-04-22:
I mean intC5 not int1C...

ghaerr, 2022-04-22:
Another quick thought is to ensure that DS gets properly saved and restored across your `__asm__ volatile ("mov %0,%%ds;"...)` call. Disassembly of host-pc98.o using `ia16-elf-objdump -D -r -Mi8086` will show that.

tyama501, 2022-04-22:
Hello @ghaerr ,

Thank you for the advice.

>What is the value of your -maout-heap= option?

-maout-heap=6144
I needed to allocate about 5.2KB work area.

>Is there documentation on INT C5H anywhere?

I have read this on a technical book before but I could only find the article below.

http://argus.sblo.jp/article/868180.html

It says for STOP key.

>You might comment out the INT C5H call and see whether this still persists.

This STACK issue was happening before I add INTC5H.
I thought this can be solved by adding it but it didn't.

>Another quick thought is to ensure that DS gets properly saved and restored 

I will try this later.

Thank you!

ghaerr, 2022-04-23:
> -maout-heap=6144

Try increasing to 10240, as malloc has some overhead and is requesting 6144 as seen in your STRACE log. It worked, but that leaves zero margin between heap and stack.

> I have read this on a technical book before but I could only find the article below.

That article shows all registers being saved before INT call, perhaps you may have to do this also. C requires that DS, ES, BP, SI and DI be saved, and may at times consider other registers to be valid as well. This could possibly be fixed using the third parameter to the _asm volatile call to specify all registers changed, not sure. That article also shows SS and SP being saved and popped, (which is very strange as it would perform a stack switch!!), but perhaps that is required.

tyama501, 2022-04-23:
Thank you @ghaerr ,

Only saving DS, ES, BP, SI and DI worked!
https://github.com/tyama501/elks/commit/592e5fdac35429280638cd94064983bfa357219b

I started to merge your changes and removed ifdefs.
I could build this.
https://github.com/tyama501/elks/commit/8d521a37463aeca7eff4f0dd6477f8a9fd177e08

How the Makefile should be?
Now like this.

I am using both host-stubs and host-pc98.

ifeq ($(CONFIG_ARCH_8018X), y)
OBJS += host-8018x.o
else
OBJS += host-stubs.o
endif

ifeq ($(CONFIG_ARCH_PC98), y)
OBJS += host-pc98.o
endif
...

basic: $(OBJS) basic.h host.h
        $(LD) $(LDFLAGS) -maout-heap=10240 -maout-stack=6144 -o basic $(OBJS) $(LDLIBS)

Thank you!

ghaerr, 2022-04-23:
Hello @tyama501,

> Only saving DS, ES, BP, SI and DI worked!

Great news. So it was a register overwrite problem then, rather than actually stack related.

> I am using both host-stubs and host-pc98.

I think probably best to copy host-stubs.c functions into your host-pc98.c, and only use host-pc98.c for your build. Then also copy stub functions from host-pc98.c (host_mode and host_plot) back into host-stubs.c for the normal build, which will call empty stub functions for MODE, PLOT and any other platform-specific commands.

@cocus has recommended that the non-implemented stub functions display "MODE not implemented" etc in host-xxx.c files. This can be done later on the IBM PC and 8081X builds though. @cocus will also have to copy stub functions into his host-8018x.c file to get his build working again. This method involves some copying, but should allow each platform maintainer to have a single platform-specific file for their implementation. I can fix up the host-stubs.c file for IBM PC after your PR.

> How the Makefile should be?

Probably something like this:
```
ifeq ($(CONFIG_ARCH_IBMPC), y)
OBJS += host-stubs.o
endif

ifeq ($(CONFIG_ARCH_8018X), y)
OBJS += host-8018x.o
endif

ifeq ($(CONFIG_ARCH_PC98), y)
OBJS += host-pc98.o
HEAP=-maout-heap=10240
endif
...

basic: $(OBJS) basic.h host.h
$(LD) $(LDFLAGS) $(HEAP) -o basic $(OBJS) $(LDLIBS)
```
I recommend removing your -maout-stack=6144 directive as it should not be required to allocate special stack space for your build; the heap value of 10240 will cover your malloc.

Thank you!

tyama501, 2022-04-23:
Hello @ghaerr ,

OK! Good. I will do that.

Maybe I will add graphic clear function to host_cls in host-pc98.c.

Thank you.

cocus, 2022-04-24:
> @cocus will also have to copy stub functions into his host-8018x.c file to get his build working again. 

Yes, no worries. For the 8018x port we'll discuss the functions available there and not related to my particular board, but rather the generic 80c18x's.

> Only saving DS, ES, BP, SI and DI worked!
> [tyama501@592e5fd](https://github.com/tyama501/elks/commit/592e5fdac35429280638cd94064983bfa357219b)

Maybe this is a question for @ghaerr, but I think this could crash the machine because the pointer of the interrupts point to the basic's app cs:ds, doesn't the OS have a better mechanism to do this from the userspace?

I'm not sure about the PC98 specifics, but maybe the STOP key initialization and handling could be done on the kernel and process it as a signal, so that apps could consume it. Except if it's tied to the graphics in which case it might not make sense to do it.

ghaerr, 2022-04-24:
Hello @cocus & @tyama501,

>  I think this could crash the machine because the pointer of the interrupts point to the basic's app cs:ds could become invalid if the app exits non gracefully

Definitely - if the application exits due to an unhandled signal or other reason, and the (not yet coded) atexit routines which would reset the interrupt vectors not get executed, the interrupt handlers could be overwritten by the next executed application and the system crash as a result.

> maybe the STOP key initialization and handling could be done on the kernel and process it as a signal, so that apps could consume it.

I had been thinking that the kernel is the proper place to handle any interrupt callbacks. Adding your idea of handling the STOP key as a signal or keystroke is even better. This could be done as an add-on to the PC-98 console initialization code.

> Except if it's tied to the graphics in which case it might not make sense to do it.

This could be handled with a separate PC-98 config option, if needed.

@tyama501, what exactly is the function of the STOP key, and its associated handler?

Also, on another issue, I'm thinking that in your LIO98INIT routine, rather than using `malloc` to dynamically allocate memory, why not just use a statically-allocated buffer? The entire rest of BASIC is designed such that no dynamic allocations are used, and this would further simplify the routine, especially since it's never freed anyways.

What is the function of the 5200 byte buffer, is there any documentation on it?

Thank you!

ghaerr, 2022-04-26:
For some reason, I could not comment on your branch directly. Here are the small changes I'd like to see:

In host.h:
```
void host_mode(int mode);
```
In basic.c:
```
int parse_MODE() {
    getNextToken();
    int val = expectNumber();
    if (val) return val;	// error
    if (executeMode) {
        int mode = (int)stackPopNum();
        host_mode(mode);
    }
    return 0;
}
```

ghaerr, 2022-04-26:
Also @tyama501, you will have to add do-nothing stub functions host_mode and host_plot do host-stubs.c in order to not break the standard CONFIG_ARCH_IBMPC build, in your PR.

tyama501, 2022-04-26:
Thank you @ghaerr 

Oh I did it again, intC5 not int1C...

I will modify it and make PR.
(maybe tomorrow or day after)

I might add MODE 0 to go back but I think MODE 1 that I added is almost default mode (640x400, color) so it is ok to exit with it.

tyama501, 2022-04-27:
Hello @ghaerr ,

I added host_gcls() to clear graphic plane with CLS command.
I also added test98.bas to using MODE/CLS/PLOT/READ/DATA.

I will make PR for these.

[image]

Thank you.

tyama501, 2022-04-29:
Hello @ghaerr ,
I took a video with PC-9801RX21 ( intel 286, 12MHz ) so I attached it.

https://user-images.githubusercontent.com/61556504/165905837-d1f17289-1412-46f1-afec-5110913c4516.mp4

I also used "screen" command and run basic infinite loop on the other screen, then I tried to see CPU Usage in the ps.
The CPU Usage is very nice new feature! It says 95% for the basic.
The all the total seems not 100% but that is because of the calculating method?

[image]

ghaerr, 2022-04-29:
Hello @tyama501,

Thanks for the video, it's interesting to see the speed on a 12Mhz system. There's a lot going on with BASIC's floating point as well as the BIOS graphics draw, so I'm not sure what (yet) to do to speed things up.

> The CPU Usage is very nice new feature! It says 95% for the basic.
The all the total seems not 100% but that is because of the calculating method?

Yes, the CPU usage calculation is done every two seconds, and uses an exponential decay (=1/e currently), which essentially keeps 37% of the old value, and 63% of the new value. Also, when starting a new program such as `ps` that may start or terminate outside of a full 2-second window, the results may be inaccurate beyond the decay calculation.

I haven't been able to test the calculation routine outside of my own QEMU testing, and am a bit surprised at the CPU values on your test being quite a bit above 100%. I suspect this is because `ps` ran quite quickly. You might try running `ps` multiple times to see how that affects the displayed results. 

If you'd like to play with the calculation like I did originally, some much larger decay values can be used from the following table in include/linuxmt/fixedpt.h (or calculate your own decay):
```
/* exponential factor is amount kept between cycles, add new N at (1 - factor) */
//#define EXP_E     (0.3679 * (1 << FSHIFT))    /* 1e^(2sec/2sec = 36.79% factor */
#define EXP_E       753         /* 36.79% factor 1/e^(2sec/2sec) as fixed-point */
#define EXP_125     256         /* 12.50% factor (256/2048) */
#define EXP_008     16          /*  0.80% factor (16/2048) */
```
A value other than `EXP_E` could then be used in the following line in elks/arch/i86/kernel/timer.c:
```
 CALC_USAGE(p->average, EXP_E, (unsigned long)p->ticks << FSHIFT);
```
Using EXP_008 should allow for a much more immediate view of CPU usage, with less history. Setting the correct factor depends on how much instantaneous vs historical CPU usage is desired to be shown. I used the 1 / e value because that is considered correct for the manner in which CPU load average is calculated, but that's not quite the same as CPU usage.

Thank you!

tyama501, 2022-04-30:
Hello @ghaerr ,

Thank you for the explanation of CPU usage.
That is fine for me.

I added COLOR and DRAW command as follows.
https://github.com/tyama501/elks/commit/13aa164daafbd46c34900f7a28243070a9c6b21d

[image]

I also added CIRCLE command but it seems system hangs during bios call.
The C5 interrupt handler problem might be happened.
Maybe I will remove CIRCLE command after some trial.

I have some questions for command.
In the README there is third parameter for DRAW command.
DRAW x,y[,a]
What is [,a] for?

For CIRCLE command the third parameter is also optional.
CIRCLE x,y[,r]
If we drop third parameter, the previous value is used for radius?

Thank you!

tyama501, 2022-04-30:
(Oops I forgot to reject the third parameter in host-pc98 when negative...)

ghaerr, 2022-05-01:
Hello @tyama501,

> I also added CIRCLE command but it seems system hangs during bios call.

That is strange, does it always hang on CIRCLE, or sometimes work? Are there multiple sources of documentation of exactly how LIO works for circle drawing?

> In the README there is third parameter for DRAW command.
DRAW x,y[,a]
What is [,a] for?

I grabbed this from the [sinclair basic online manual](https://worldofspectrum.net/ZXBasicManual/zxmanchap17.html). It says "An extra frill with DRAW is that you can use it to draw parts of circles instead of straight lines, by using an extra number to specify an angle to be turned through...". It seems with 3rd parameter, it is an arc-drawing routine. Perhaps we should not implement 3rd parameter at all, and only have DRAW draw lines.

> For CIRCLE command the third parameter is also optional.
CIRCLE x,y[,r]
If we drop third parameter, the previous value is used for radius?

The same documentation above shows the radius parameter used at all times. Perhaps to keep things simple, we should not implement optional parameters. This will simplify both the interpreter as well as the host-pc98.c processing. If we take that option, CIRCLE would take 3 parameters and DRAW 2 parms.

> I added COLOR and DRAW command as follows.

The code looks good, but we are writing lots of code to parse optional parameters, and having multiple procedures that do nearly the same thing. Perhaps consider instead of adding seperate parse1or2IntCmd and parse2or3IntCmd functions, just enhance the existing parse2IntCmd to take a parameter of the number of integers expected and parse those parameters, and rename it "parseNIntCmd(int n)". Then pass it the expected number of parameters for each of the POSITION, PIN, PINMODE, PLOT, DRAW and COLOR commands. The interpreter will remain smaller and simpler, and also will the parameter processing in host-pc98.c.

> (Oops I forgot to reject the third parameter in host-pc98 when negative...)

Yes, that won't be needed if we take the fixed-parameter approach!

Thank you!

tyama501, 2022-05-05:
Hello @ghaerr ,

Thank you for adding CONFIG_TIME_RTC_LOCALTIME.
It is working.

Back to the Basic,
I modified parseTwoIntCmd to parseNIntCmd as follows.
https://github.com/tyama501/elks/commit/a81142c0588ed94e4e53392414d92b5b93b57224

For COLOR command,
I don't use 2nd paramater "bg" but I treated as 2 parameter cmd since IBM may use it.

For CIRCLE command,
it is still always hang so I wanted to add intC5 asm code for trying but somehow I cannot link with the error like this.

 ia16-elf-gcc -fno-inline -melks-libc -mcmodel=small -mno-segment-relocation-stuff -mtune=i8086 -Wall -Os -o basic basic.o host.o host-pc98.o intC5-pc98.o
host-pc98.o:function host_mode: error: undefined reference to 'intc5_handler!'
collect2: error: ld returned 1 exit status
gmake: *** [Makefile:36: basic] Error 1

Makefile is like this. Is there any other part I need to modify?
ifeq ($(CONFIG_ARCH_PC98), y)
OBJS += host-pc98.o intC5-pc98.o
endif

The code is like this.
intC5-pc98.S
        .arch   i8086, nojumps
        .code16
        .text

        .global intc5_handler

intc5_handler:
        iret

host-pc98.c
extern void intc5_handler(void);
...
        // Set interrupt handler for 0xC5
        intvec = _MK_FP(0, 0xC5 << 2);
        *intvec = (unsigned long) ((unsigned long __far *) intc5_handler);

ghaerr, 2022-05-05:
> Thank you for adding CONFIG_TIME_RTC_LOCALTIME.
It is working.

Thank you for your testing! I'm glad we're done with time functions :)

> I modified parseTwoIntCmd to parseNIntCmd as follows.

Looks good!

> intC5-pc98.S
.arch i8086, nojumps
.code16
.text
    .global intc5_handler
intc5_handler:
iret

That is very strange, I am not sure what the problem is.

> Makefile is like this. Is there any other part I need to modify?

That seems correct, your `ld` output seems to show all proper .o files are being included.

Perhaps use `ia16-elf-objdump -t` to display the symbol table on intC5-pc98.o as well as host-pc98.o, to see that intc5_handler symbol(s) match up.

It is possible that following statement could generate code thinking that intc5_handler is in data segment: (?)
```
 *intvec = (unsigned long) ((unsigned long __far *) intc5_handler);
```
Use `ia16-elf-objdump -D -r -Mi8086` to disassemble host-pc98.o to see precise reference to intc5_handler. Go ahead and post disasm output and I will try to help.

Thank you!

ghaerr, 2022-05-05:
> .text
.global intc5_handler
intc5_handler:

Can you post your intC5-pc98.S file? Please check to be very sure the spelling between the .global and definition of the intc5_handler symbol is the same. It seems that `as` will allow a .global of an undefined symbol, and no error will be shown, which would be the case if the names don't match.

tyama501, 2022-05-06:
Hello @ghaerr ,

ia16-elf-objdump -t intC5-pc98.o
00000000 l    d  .text  00000000 .text
00000000 l    d  .data  00000000 .data
00000000 l    d  .bss   00000000 .bss
00000000 g       .text  00000000 intc5_handler

ia16-elf-objdump -t host-pc98.o | less
00000000         *UND*  00000000 intc5_handler!
00000000         *UND*  00000000 intc5_handler&
00000000         *UND*  00000000 intc5_handler

It seems *UND*.

Attached ia16-elf-objdump -D -r -Mi8086 host-pc98.o
[objdump_intc5.log](https://github.com/jbruchon/elks/files/8637922/objdump_intc5.log)

intc5-pc98.S
	.arch	i8086, nojumps
	.code16
	.text

	.global intc5_handler

intc5_handler:
	iret

Thank you.

tyama501, 2022-05-06:
Not about circle but one more thing,

this basic program works
[BAK98.txt](https://github.com/jbruchon/elks/files/8640368/BAK98.txt)

[image]

but this program does not work.
[test98.txt](https://github.com/jbruchon/elks/files/8640356/test98.txt)

[image]

Only the difference is line number. I renumbered it.
It is strange that host_color is called 7 times not 3.

Is there some limitation for line number?

tyama501, 2022-05-06:
Sorry GOTO number should be different...

ghaerr, 2022-05-06:
Hello @tyama501,

> a16-elf-objdump -t intC5-pc98.o
00000000 l d .text 00000000 .text
00000000 l d .data 00000000 .data
00000000 l d .bss 00000000 .bss
00000000 g .text 00000000 intc5_handler

This shows a zero-length text segment. Perhaps run `ia16-elf-size` on it to check for non-zero text size.

>intc5-pc98.S
.arch i8086, nojumps
.code16
.text
.global intc5_handler
intc5_handler:
iret

Are these statements indented with at least one tab or space character? Perhaps the macros are not recognized in the first column.

> Sorry GOTO number should be different...

Ok does that fix your BASIC problem, or is it something else?

Thank you!

tyama501, 2022-05-06:
Hello @ghaerr ,

ia16-elf-size intC5-pc98.o
   text    data     bss     dec     hex filename
      1       0       0       1       1 intC5-pc98.o

ia16-elf-objdump -D -r -Mi8086 intC5-pc98.o
intC5-pc98.o:     file format elf32-i386

Disassembly of section .text:

00000000 <intc5_handler>:
   0:   cf                      iret

> Are these statements indented with at least one tab or space character?

Yes.
It seems .S file is not able to be attached here so I renamed and attached it.
[intC5-pc98.txt](https://github.com/jbruchon/elks/files/8641687/intC5-pc98.txt)

> Ok does that fix your BASIC problem, or is it something else?

Yes.
It is fixed.

Thank you.

tyama501, 2022-05-06:
If there is easier (and better) way to handle it in the kernel we can try that.

[42 more replies]

closed 2022-06-15 by ghaerr