ghaerr/elks #1242: Problem booting 1.44 image fat

toncho11, 2022-03-23:
So I finally booted ELKS from my 1.44 floppy on my IBM 5160.

But it gives an error:

`Exec failed: /bin/sh (errno 12)`
`No shell (errno 12)`

I can not login. But /bin/sh is there ... so what is the problem? 

I used [fd1440.img](https://github.com/jbruchon/elks/suites/5763223304/artifacts/191655281)

toncho11, 2022-03-23:
Or maybe it was a bad sector on the floppy...

ghaerr, 2022-03-23:
> No shell (errno 12)

Errno 12 is Out of Memory; the kernel thinks there's not enough RAM to run /bin/sh. How much do you have?

Perhaps copy or rename /bin/sash to /bin/sh, it uses less RAM. You can also login as "toor" instead of root to run /bin/sash as your shell.

toncho11, 2022-03-24:
I forgot to install my memory expansion card. So the memory was 256 kb. 
"toor" did not work. I mean it kept giving the same error. 
I added my card and everything is ok now.
`meminfo` is not very clear for normal users (but it is another topic).
`uname` works.

closed 2022-03-24 by toncho11