ghaerr/elks #1253: Obtaining CPU usage stats

cocus, 2022-04-08:
Hi, this is something I was thinking about from long tine ago, but now that I can telnet into my board, I'm sure I'd like to see how much cpu is being currently used.
Not sure if there's already a way to do so, but I have to ask it.
Also, I was thinking about improvements on the romfs version, like letting sash automatically execute an initial script, but I will asl that in another issue, if needed.

Thanks!

ghaerr, 2022-04-08:
> I'd like to see how much cpu is being currently used.

There isn't currently any implementation of CPU (or user/kernel) times at present. It could be a lot of work to add this similarly to a Linux kernel, the question would be to what real benefit on a single user system.

However, a simpler approach where the kernel timer tick (100 HZ) just increments a longword in the task structure of the currently running process would be pretty straightforward. Then `ps` could convert those timer ticks to MM:SS or something like that. That would show what processes are consuming time by running `ps` multiple times, but it wouldn't actually calculate CPU usage.

> letting sash automatically execute an initial script

`sash` will run a shell script (of simple internal/external commands only) when passed a file as an argument. Are you thinking you'd like to burn this into ROM for system startup? If that's the case, just put the commands in /etc/sashrc and they'll run when it starts.

Mellvik, 2022-04-09:
This brings up a much simpler command that I've been missing: uptime.
It should be as simple as recording boot time in a long (and ignoring time changes via the date command)?

-M;

cocus, 2022-04-19:
Well, I know the idea of this is to not run that many things at the same time, but for instance, I have to run ktcp, telnetd and now basic. I think it would make sense to know which thing is using cpu time.
So, incrementing a 32bit unsigned variable on the process control block seems quite tempting. Then you could expose these somehow, and let the user (in this case it would be a top-like app) do the calculations and obtain the cpu usage %.
I don't think this would be enough to calculate the uptime of the system, except if we consider the time of this variable on the pid 1.

ghaerr, 2022-04-19:
> Then you could expose these somehow

This doesn't have to get complicated - `ps` already works by opening /dev/kmem and reading each task's task structure to display its results. Adding a longword to each task structure that gets incremented by the HW timer would work to accumulate which task the kernel was running each tick, and an option could be added to `ps` to show the count, or some arithmetic on the count. This would end up combining 'user' vs 'system' CPU times, but it would simple and small.

> except if we consider the time of this variable on the pid 1.

Each process's longword would only accumulate when it is actually running, and /bin/init doesn't run that much, so I'm not sure how this could be made to show uptime. It would seem a simpler approach would be to just have /etc/rc.sys do a "date > /tmp/starttime" to record when the system started.

cocus, 2022-04-19:
> Each process's longword would only accumulate when it is actually running, and /bin/init doesn't run that much, so I'm not sure how this could be made to show uptime. It would seem a simpler approach would be to just have /etc/rc.sys do a "date > /tmp/starttime" to record when the system started

You're right, I don't know what I was thinking regarding that, lol.

> This doesn't have to get complicated - `ps` already works by opening /dev/kmem and reading each task's task structure to display its results. Adding a longword to each task structure that gets [...]

This makes sense. Pretty interesting. Yes, I think this would be the best and easiest way to do it. I'll try to toy with this on Qemu since flashing my 80188 is kinda annoying for testing. Should this feature be a configuration option? Ot should be something that can't be tweaked?

ghaerr, 2022-04-19:
> Should this feature be a configuration option? Ot should be something that can't be tweaked?

Probably doesn't need a configuration option: add a 'jiff_t ticks' into the task structure after 'sig'.
Then add "current->ticks++" in elks/arch/i86/kernel/timer.c::timer_tick().
Read the value using `offsetof` in ps.c and display it.

cocus, 2022-04-22:
Hi @ghaerr , I've done exactly as stated, but I think this doesn't work as expected.
I only get to increment the "ticks" field of `sh` once I run something (like `ps` and then getting back to `sh`). Also, seems like when spawning a new process, the last `ticks` value carries over. Not sure what's going on.

[image]

Diff: 
```
diff --git a/elks/arch/i86/kernel/timer.c b/elks/arch/i86/kernel/timer.c
index 9042db44..f489617d 100644
--- a/elks/arch/i86/kernel/timer.c
+++ b/elks/arch/i86/kernel/timer.c
@@ -22,6 +22,8 @@ extern void rs_pump(void);
 
 void timer_tick(int irq, struct pt_regs *regs)
 {
+    current->ticks++;
+
     do_timer(regs);
 
 #if defined(CONFIG_CHAR_DEV_RS) && (defined(CONFIG_FAST_IRQ4) || defined(CONFIG_FAST_IRQ3))
diff --git a/elks/include/linuxmt/sched.h b/elks/include/linuxmt/sched.h
index 6fc5657d..66247cf9 100644
--- a/elks/include/linuxmt/sched.h
+++ b/elks/include/linuxmt/sched.h
@@ -76,6 +76,8 @@ struct task_struct {
     sigset_t                   signal;         /* Signal status */
     struct signal_struct       sig;            /* Signal block */
 
+    jiff_t              ticks;
+
 #ifdef CONFIG_SUPPLEMENTARY_GROUPS
     gid_t                      groups[NGROUPS];
 #endif
diff --git a/elkscmd/sys_utils/ps.c b/elkscmd/sys_utils/ps.c
index 5e42d2af..dcba8d96 100644
--- a/elkscmd/sys_utils/ps.c
+++ b/elkscmd/sys_utils/ps.c
@@ -131,7 +131,8 @@ int main(int argc, char **argv)
                exit(1);
        }
 
-       printf("  PID   GRP  TTY USER STAT CSEG DSEG  HEAP   FREE   SIZE COMMAND\n");
+       printf("HEY\n");
+       printf("  PID   GRP  TTY USER STAT TICKS CSEG DSEG  HEAP   FREE   SIZE COMMAND\n");
        for (j = 1; j <= MAX_TASKS; j++) {
                if (!memread(fd, off + j*sizeof(struct task_struct), ds, &task_table, sizeof(task_table))) {
                        perror("ps");
@@ -163,6 +164,8 @@ int main(int argc, char **argv)
                                (pwent ? pwent->pw_name : "unknown"),
                                c);
 
+               printf("%5d ", task_table.ticks);
+
                /* CSEG*/
                cseg = (word_t)task_table.mm.seg_code;
                printf("%4x ", cseg? getword(fd, (word_t)cseg+offsetof(struct segment, base), ds): 0);
```

ghaerr, 2022-04-23:
Hello @cocus,

Actually, looks like a great job! It seems, other than new processes starting from 0 ticks, the report is correct - init ran a total of 9/100 seconds, sh 16/100's, then 18/100's after running ps again.

> I only get to increment the "ticks" field of sh once I run something (like ps and then getting back to sh).

That's because sh isn't actually running - it's waiting on a read and scheduled out. The idle task (0) runs when nothing else can. If one were running BASIC on another terminal, and it wasn't waiting on a read(), its tick count would be incremented every 1/100 second.

> Also, seems like when spawning a new process, the last ticks value carries over. 

Oops - forgot about fork(), it copies the previous task structure and doesn't zero the tick count. Do that in elks/kernel/fork.c::find_empty_process() after the new PID is set.

Thank you!

ghaerr, 2022-04-23:
Once you get ticks working, instead of reporting tick count or time, it might be worth reporting CPU % usage instead? This could be done using the following formulae:
```
CPU % = BUSY / (BUSY + IDLE)
```
The idle time will be the tick count for process 0, which is the first task in the task struct array. The problem will be that the IDLE time will have accumulated over the entire uptime of the kernel and then effectively averaged into the BUSY time of the application, lowering its effective CPU %. We probably need a seperate jiff_t which would be the idle task ticks when the application started, so that that could be subtracted from the total idle ticks:
```
CPU % = BUSY / (BUSY + IDLE - BUSY_START_IDLE_TICKS)
```
Even this method averages the CPU % across the entire time the application has been running. To be even more proper, the IDLE time might have to be adjusted to zero (along with an adjustment to BUSY_START_IDLE_TICKS) every second or so so that the CPU % is accurate to within the chosen time scale (last second, last 5 seconds, ... etc).

We may have to raise the kernel stack size if another 4 bytes is added to the task struct, and every byte we add costs 16 bytes total, as the task array is 16 tasks at the moment.

ghaerr, 2022-04-23:
@cocus, [here's an article (2nd answer)](https://stackoverflow.com/questions/3748136/how-is-cpu-usage-calculated) that shows another way we might calculate CPU % that may not require a separate busy_start_idle_ticks task struct entry.

cocus, 2022-04-23:
I think the issue on this case is that no one is house-keeping these numbers so they don't get "averaged". I think we could let the idle task to do this for us. Not sure how these numbers are going to be stored in % (maybe adding another element to the task struct?). Otherwise it would be possible to capture the task's ticks over a fixed amount of time, and you can easily calculate the cpu %.
I forgot that the task 0 was the idle one, which is not shown on ps.

For the article from StackOverflow, yes, we could calculate the actual CPU% but not per-process. Or at least that's what I understood.

ghaerr, 2022-04-23:
If you temporarily modify `ps` to show a line for the idle task and add the fork fix, we could take a look at the numbers and probably come up with a way to have the idle task or the timer interrupt average the numbers every second or so across the entire task array. It really depends on whether we want to see CPU time or CPU %.

Since `ps` column width is already big, I can see some ps options coming lol. (And remove CSEG and DSEG from the standard display, possibly displaying CPU and CPU% instead).

Mellvik, 2022-04-23:
my 2 cents - 
instead of removing CSEG and DSEG, hide them under a command line option so we can get them when they're useful …

-M

ghaerr, 2022-04-25:
Hello @cocus,

I think I figured a way to compute CPU % usage using the jiff_t timer tick mechanism you have working, with some additions. The idea would be to calculate the tick counts for each task as well as the idle task for 2 seconds, then using fixed point to compute CPU% = BUSY / (BUSY + IDLE) and add that back into itself using an exponential decay function every two seconds, then clearing the tick counts and starting over.

This method would be very similar to the way the Linux 1.0 kernel used fixed point and exponential decay to compute load average, discussed [in this article](https://en.wikipedia.org/wiki/Load_(computing)#CPU_Load_Computation).

Are you working on that, or would you like me to take a pass at it, starting with your work?

Thank you!

cocus, 2022-04-26:
> Are you working on that, or would you like me to take a pass at it, starting with your work?

Hi @ghaerr , Sadly I don't have enough time to continue working on this ASAP, so if you have spare cycles, please go ahead with the changes! 

> I think I figured a way to compute CPU % usage using the jiff_t timer tick mechanism you have working, with some additions. The idea would be to calculate the tick counts for each task as well as [...]

Interesting, never thought about using an exponential decay to compute this, but seems like it's viable!

closed 2022-04-27 by ghaerr