ghaerr/elks #1260: Add support for netinet/Berkeley sockets API

mikechambers84, 2022-04-14:
I'm extremely impressed with the current state of ELKS. I had last looked at it many years ago, and I just rediscovered it to find that it's very much improved since then. This is becoming a viable OS for old XT-class PCs.

I went to try to port a few small network utilities to it (HTTP servers, IRC client/server) and discovered that it's missing netinet, which many of these use. Adding this library would open up ELKS to running a wide array of different network software that it currently isn't able to.

Has this been attempted yet? Is there anything unfeasible about it, or are there any caveats? I may take a first crack at it.

ghaerr, 2022-04-14:
Hello @mikechambers84,

Thanks for your compliments and interest in ELKS! We've definitely been busy the last couple years getting the system quite stable, and now have TCP/IP networking running quite well, for most uses.

ELKS does have some of the standard C include files, including "arpa/inet.h", "netinet/in.h" and "sys/socket.h". These live in elks/libc/include. These provide mostly standard support for Berkeley sockets API, but for TCP connections only (not UDP).

Because ELKS is quite limited by a 64K data limit in both the kernel and user applications, the TCP/IP stack is implemented outside the kernel, and some important functions like gethostbyname have been implemented in smaller versions, using a "in_" prefix, e.g. in_gethostbyname(). These functions are defined in arpa/inet.h, and the source for them is in elks/libc/net/. I am sorry we don't (yet) have working man pages for those functions, but we're working on it.

ELKS currently has a very minimal, but working http server, and full featured telnet and ftp client and servers. These can all be found in elks/elkscmd/inet/. Take a quick look at them to see how easy it is to use the standard socket routines as well use as the slightly changed in_xxx routines. These now integrate well with each other and our new DNS resolver.

We welcome enhancements to httpd (elks/elkscmd/inet/httpd/httpd.c) or newly contributed network programs. Our IRC client/server, in inet/tinyirc, frankly is terrible and doesn't work well. I'd love to see a replacement or mixup. Should you decide to jump in, we would be glad to help you.

Thank you!

mikechambers84, 2022-04-15:
I'll see if I can port a few things over. I'm looking at darkhttpd and ngircd right now. I've done an ngircd port to DOS before, it's a good server.

mikechambers84, 2022-04-16:
Is there any chat room for ELKS? An IRC channel maybe?

ghaerr, 2022-04-16:
@mikechambers84, we don't have an IRC channel, but I'd be happy to chat with you on one. I'm not that familiar with IRC, and run on macOS. Would there be an easy way to set one up?

mikechambers84, 2022-04-16:
Thanks! If you'd rather me get in touch some other way if I have some porting questions, I could do that too. An general IRC channel for ELKS might be nice if you're interested though. It's very old school, but is a good way to have real-time discussions for a software project. I could always set up a channel on the libera.chat network. Let me know.

I don't use macOS very often, but I do recall LimeChat being a pretty decent IRC client.

If I can learn the ropes for compiling software for ELKS, I'll probably end up doing quite a few ports. Perhaps I'd be able to add some of the missing common header files, too. I'd like to get involved in this! Most of the things I've tried to port so far, I've failed for one reason or another. Usually headers and functions that are missing from ELKS.

Mellvik, 2022-04-16:
IRC works well when there are many participants.
With ELKS there are (still) just a few  of us, so mail is better, not having to attend to yet another channel.
Mike, you're very welcome to the club. We've made great progress recently as you pointed out. Your interest make us stronger. Your potential contributions even more so.
-M

closed 2022-06-16 by ghaerr