ghaerr/elks #1267: xms: A20 was off

astrobleem, 2022-04-28:
**Description**
- What is expected ?
Expecting elks to boot 
- What is observed ?
I have provided a screenshot but the last line we see is 
"xms: A20 was off"

the machine appears frozen
num lock does not function and caps lock does not funtion

**Configuration**
- Version (commit or release): 0.5.0 and 0.2  I am pretty sure I tried 0.4.0 images you had available also, I have not tried to build my own.
- 
- Emulator or HW platform:
Real Hardware, not emulated
Tandy 1000 EX
640k memory expansion/CF-IDE/Gotek Floppy emulator

**How to reproduce ?**
- Does the problem always happen ?
- yes
- List the steps to reproduce the problem.
download any ELKS v0.5.0 image and try to boot

**Raw data**
- Screenshot(s):
[image]

- Log file:

**Additional information**
Add any other context information about the problem here.

willing to help out in any way to provide extra info/debugging

ghaerr, 2022-04-28:
Hello @astrobleem,

Thanks for the boot problem report. The ELKS v0.5.0 boot images were incorrectly configured to assume a machine with XMS capabilities, and when they're missing, the images fail to boot. A solution is to recompile the system with XMS turned off, but we provided a FAT-based boot image for a user with this same problem in #1199.

Take a look at that issue, or [download this link](https://github.com/jbruchon/elks/files/8173579/fd1440-fat.img.zip) to see if that image boots and solves your problem.

Thank you!

astrobleem, 2022-04-28:
So sorry to create an issue when one already existed. For what it's worth I did make an attempt to look for a similar issue. 

I used the fd1440-fat.img.zip  you linked to, and encounter a new error

We see 

ELKS.3!
Press key

If I press a key the system reboots.

This loops over and over and over.

astrobleem, 2022-04-28:
[image]

ghaerr, 2022-04-28:
> ELKS.3!

The '3' means no /Linux in the root directory of the FAT (MSDOS) disk. Did you use 'dd' to copy it? Using MSDOS won't work.

What size drive are you using in the PC, 1440K drive?
Which v0.5.0 image did you use in your first test?

Thank you!

astrobleem, 2022-04-28:
[image]

fd360-fat.img
gets us to the A20 off message.

I am using a gotek floppy USB writing using USB Batch Manage Tool 1.4

[image]

....
testing with the 1.44 fat image from 0.5.0 gets ELKS.3!
[image]

360 image gets me to a20, 
720 image gets me to a20,
1440 gets me elks.3

I guess this means that my system doesnt support 1.44m images.

ghaerr, 2022-04-28:
> I guess this means that my system doesnt support 1.44m images.

Probably not, but that's ok. Here's a couple 360K images, FAT and MINIX. Try them out, they should work.
[fd360-minix.img.zip](https://github.com/jbruchon/elks/files/8585100/fd360-minix.img.zip)
[fd360-fat.img.zip](https://github.com/jbruchon/elks/files/8585102/fd360-fat.img.zip)

astrobleem, 2022-04-28:
yeehaw!

[image]

ghaerr, 2022-04-28:
> yeehaw!

Well, glad we were able to solve that problem! It is as I suspected, the v0.5.0 images require XMS, sorry.

The 360k images don't have much on them, you might want to try configuring your GoTek to support 1.44M floppy, then use one of the 1440K images I sent you. If you want to try compiling everything from scratch, you can read the Build section of the Github README.

I may try to cut a new v0.5.1 with fixed images, as your problem did happen to another new user.

Thank you!

astrobleem, 2022-04-28:
I hope it's not asking too much, could you hook me up with a hd mbr fat 32m  image ?

I have a compact flash drive.

ghaerr, 2022-04-28:
Sure, here's the HD32 mbr and non-mbr images.
[hd32mbr-fat.img.zip](https://github.com/jbruchon/elks/files/8585178/hd32mbr-fat.img.zip)
[hd32-fat.img.zip](https://github.com/jbruchon/elks/files/8585179/hd32-fat.img.zip)

ghaerr, 2022-04-28:
> Is it possible to detect the presence of XMS without having to compile this in?

No, not on earlier (and all) machines, unfortunately. The next release will not have XMS compiled in, so this won't be an issue.

astrobleem, 2022-04-28:
that booted up very fast!!

[image]
[image]

astrobleem, 2022-04-28:
this is awesome. thank you closing the issue.