ghaerr/elks #1270: Build issue under ubuntu 22.04

cocus, 2022-04-28:
Hi,
I've set up a docker container with ubuntu 22.04 on it, and I can't build the crosscompiler toolchain (when running `./build.sh`).
This is most likely due to the gcc being a really recent one `gcc version 11.2.0 (Ubuntu 11.2.0-19ubuntu1)`.
The error I'm getting is:
```
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../binutils-src/bfd  -DBINDIR='"/workspace/cross/bin"' -I. -I../../binutils-src/bfd -I../../binutils-src/bfd/../include  -DHAVE_i386_elf32_vec -DHAVE_i386_elks_vec -DHAVE_i386_msdos_vec -DHAVE_i386_aout_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec   -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -Werror -I../../binutils-src/bfd/../zlib -g -O2 -MT plugin.lo -MD -MP -MF .deps/plugin.Tpo -c -o plugin.lo ../../binutils-src/bfd/plugin.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../binutils-src/bfd -DBINDIR=\"/workspace/cross/bin\" -I. -I../../binutils-src/bfd -I../../binutils-src/bfd/../include -DHAVE_i386_elf32_vec -DHAVE_i386_elks_vec -DHAVE_i386_msdos_vec -DHAVE_i386_aout_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -Werror -I../../binutils-src/bfd/../zlib -g -O2 -MT plugin.lo -MD -MP -MF .deps/plugin.Tpo -c ../../binutils-src/bfd/plugin.c -o plugin.o
In function 'load_plugin',
    inlined from 'bfd_plugin_object_p' at ../../binutils-src/bfd/plugin.c:424:53:
../../binutils-src/bfd/plugin.c:403:20: error: 'valid_plugin' may be used uninitialized [-Werror=maybe-uninitialized]
  403 |         has_plugin = valid_plugin;
      |         ~~~~~~~~~~~^~~~~~~~~~~~~~
../../binutils-src/bfd/plugin.c: In function 'bfd_plugin_object_p':
../../binutils-src/bfd/plugin.c:397:11: note: 'valid_plugin' declared here
  397 |       int valid_plugin;
      |           ^~~~~~~~~~~~
cc1: all warnings being treated as errors
```

Which might have been treated as a warning in older gcc versions, but not on this one.

ghaerr, 2022-04-29:
Hello @cocus,

Was this the only error and did the build complete, or are there still more similar issues?
ELKS is running an older commit of the toolchain, perhaps with some more detail we can have @tkchia comment on whether you should try a later version of the compiler or other sources to address this problem.

Thank you!

cocus, 2022-04-29:
No, these were the two ones. After that, I was able to build (except the missing stubbed functions of basic) the current ELKS repo for my 8018x

Jookia, 2022-05-03:
I hit this issue too. Maybe worth patching the toolchain?

cocus, 2022-05-03:
Hey @Jookia, if you apply the diffs on the files I've mentioned, does it works for you?

Jookia, 2022-05-03:
Yes.

ghaerr, 2022-05-03:
> Maybe worth patching the toolchain?

The toolchain built under the cross/ directory is built from specific commits numbers from https://github.com/tkchia/gcc-ia16, specified in tools/Makefile. None of the source files under cross/ are git-tracked on this repo. Thus, patches and/or changes are made upstream, which after testing, are added back to ELKS with a tools/Makefile update commit.

@cocus, perhaps you should submit your diff on the gcc-ia16 site, to see whether this may have already been addressed over there.

cocus, 2022-05-04:
> @cocus, perhaps you should submit your diff on the gcc-ia16 site, to see whether this may have already been addressed over there.

I'll open an issue on @tkchia's project and let's see if it works or not

closed 2023-08-03 by ghaerr