ghaerr/elks #1275: far function

marcin-laszewski, 2022-05-03:
I have written my first program which uses 'far' function. It shows segment and offset of 'fun_near' and 'fun_far'. Why does a segment always have the same value even when I run this program twice simultaneously? CS segment should (?) be diffrent:

--------------------------------------------
ELKS 0.5.0
login: root
# funfar
near:   3ab6:43 000043
far:    3ab6:35 000035
# funfar & funfar &
near: 3ab6:43 000043
near:   3ab6:43 000043
far:    3ab6:35 000035
far:    3ab6:35 000035
----------------------------------------------
[funfar.c.gz](https://github.com/jbruchon/elks/files/8609616/funfar.c.gz)

marcin-laszewski, 2022-05-06:
Thank you very much! I didn't know that ELKS is so smart it is able to share CS segment.

One more question: I have two programs:
funfar.c: There is a function 'farfun'. It calls this function, shows its seg:off in memory and sleep (nearly) forever.

funfar1.c: Read two argument - seg & off and call a function which has this address.

Run "funfar" in background:
    # funfar &
    funfar.c      far:    15038:53        000035
    funfar.c        fun_far

Run "funfar1" which calls far function (150028:53) from "funfar":
    # funfar1 15038 53
    funfar1.c       start   15038:53        003abe
    funfar1.c       stop

As we see it doesnt't works. Why?

marcin-laszewski, 2022-05-06:
[funfar.c.gz](https://github.com/jbruchon/elks/files/8640261/funfar.c.gz)
[funfar1.c.gz](https://github.com/jbruchon/elks/files/8640262/funfar1.c.gz)

ghaerr, 2022-05-06:
Hello @marcin-laszewski,

> As we see it doesnt't works. Why?

It seems you are running one application in the background, then attempting to far-call a function in that sleeping program from another program. That won't work and could result in a system crash, or introduce system instability. The reason is that the DS segment is different between the two applications - so the text string you passed is relative to the wrong data segment. That is equivalent to passing an invalid pointer to the puts(text) function. Depending on the compiled code, the DS register could be refreshed with the wrong value, and both programs would be messed up.

Thank you!

marcin-laszewski, 2022-05-09:
Thank you very much for your explanation.
I cannot understand what is wrong with DS: 'far function of 'funfar' is called from "fundar1" and DS doesn';t changed - it is still set as in this caller program ("funfar1").
So, what is wrong?

closed 2022-06-16 by ghaerr