ghaerr/elks #1276: Failing to boot in DOSBox-X with 286 or newer

Jookia, 2022-05-03:
**Description**
When booting with with a CPU set to 286 or newer suddenly the filesystem can't be found.

**Configuration**
- Version (commit or release): 0.5 official and b51718735aa18e1cf1d87598129685ecd447b5e4 both hd images
- Emulator or HW platform: DOSBox-X 0.83.25

**How to reproduce ?**
Build with XMS enabled.
Set the core type to 80286 or newer.

**Raw data**
- Screenshot(s): 
[image]

- Log file: None

**Additional information**
Add any other context information about the problem here.

ghaerr, 2022-05-03:
Hello @Jookia,

Interesting! Can you send or post your dosbox-x.conf file so I can see the options you're using to set the CPU type, as well as any command line options? This will allow me to duplicate the issue.

A quick look at your screenshot shows the FAT filesystem driver receiving all 0xFF's in the superblock... not sure if this has to do with using XMS buffers or something else.

Thank you!

ghaerr, 2022-05-03:
Hello @Jookia,

I think the problem here is that the v0.5.0 version of ELKS did not check properly for a 386 processor before enabling the code to use XMS buffers. In your case above, you can see on the xms: line "using unreal mode" which requires a 386 processor. When XMS buffers is not operational, it is known that the buffer contents are all 1 bits, which matches with the FAT filesystem mount attempt in your screenshot.

Using the latest source, with XMS enabled, and using cputype=286 in dosbox-x.conf, things work, with the xms: line showing XMS "disabled, requires 386":
[image]

I'm not sure the reason this may be happening using the more recent commit you mentioned, but is likely due to the ways that the XMS buffers and or INT 15h is configured. In either case, the issue is that the 286 processor is not supported for XMS, unless INT 15h is operational. If you can show still improper detection of CPU type or XMS enablement with the latest source, we can look into why the detection code may be failing.

Thank you!

Jookia, 2022-05-03:
What's latest source if not the commit I mentioned?

ghaerr, 2022-05-03:
> What's latest source if not the commit I mentioned?

My error, that is the latest. My screenshot was from the same, but there are differences between our .config files.

Jookia, 2022-05-03:
Could you paste your configuration? Mine stock dosbox-x I think with the only settings changed being the ones I noted.

ghaerr, 2022-05-03:
Here it is. I haven't seen your dosbox config since you didn't post it, but the issue remains the same: XMS isn't supported for 286 processors. This config should result in the same "xms: disabled, requires 386" provided your dosbox setup is the same.
[config.ok.zip](https://github.com/jbruchon/elks/files/8613944/config.ok.zip)

Jookia, 2022-05-03:
Using that config and stock DOSBox-X and running the commands 'imgmount 0 fdd.img -t fdd -fs none' and 'boot a:' reaches 'fd: /dev/fd0 probed' and crashes. Setting the CPU to 8086 fixes the issue.

Jookia, 2022-05-03:
[image]

ghaerr, 2022-05-03:
> Example of the new error on 286: 

It appears that xms is being properly disabled, obviously big problems with the identification of /dev/fd0 with 0 cylinders...

When I try your suggested 'imgmount 0 fdd.img -t fdd -fs none' and 'boot a:'  I get:
[image]

I normally run dosbox from within a shell script (and set the config for 1.44M rather than 2.88M floppy, my posted config is for 2.88M floppy, that could be the latest issue here):
```
./dosbox-x-sdl2 -set "dosbox quit warning=false" -fastlaunch -c "boot image/fd1440-fat.img"
```
And use dosbox-x.conf:
```
[dosbox]
debug=true

[sdl]
output=bitmap
doublescan=true

[ne2000]
ne2000=true
nicirq=12
backend=pcap

[ethernet, pcap]
realnic=lo0

[cpu]
cputype=286 ;( only for this issue)
```

Jookia, 2022-05-03:
The issue happens with that config and launch style, with fd2880.img .
However if I change the .config to be small enough and use fd1440.img it boots fine?
Could you reproduce this? If not, maybe this is a DOSBox-X bug or something.

ghaerr, 2022-05-03:
> Could you reproduce this? If not, maybe this is a DOSBox-X bug or something.

Yes, fails in the same way over here - it appears that DOSBox-X does not support 2.88Mb floppy geometry. It all works fine on QEMU and real hardware.

> However if I change the .config to be small enough and use fd1440.img it boots fine?

Good. To summarize, ELKS boots fine on DosBox-X CPU 286 with my config changed to 1.44Mb floppy w/latest source. For some reason, it didn't work with your config, which we never figured out. You might diff them if it matters. The overall failure to boot has been confirmed to be either 1) older sources not properly recognizing XMS with non-386 processor, and/or 2) DosBox-X doesn't support 2.88M floppies.

Thank you!

Jookia, 2022-05-03:
I believe the issue is 2, but also I found that setting memsize=1 I get the original error.

ghaerr, 2022-05-03:
>  I found that setting memsize=1 I get the original error.

Is that a DosBox setting? That's definitely the problem because XMS starts allocating memory at 1MB, which is not (yet) configurable. Thus, the XMS buffers were being allocated outside the PC memory being emulated!

Jookia, 2022-05-03:
Yes, that's a DOSBox-X setting at least. I assumed memsize=1 would give you 1 megabyte of memory, but I'm not entirely sure what that means. I guess that there's upper memory/high memory?

ghaerr, 2022-05-03:
The first 2^20 of memory (1Mb) is directly addressable by 8086, although on the PC, most of RAM beyond 640K is dedicated to video hardware, BIOS, and sometimes 64K or so of other memory. There are lots of names for various memory in between 640K and 1Mb+65520. To address beyond 1Mb (which memory is called extended memory (XMS)), a processor with more than 20 bits of address lines is required. Thus XMS memory starts at the 1Mb boundary.

The ELKS memory layout is documented in Documentation/text/memlayout.txt:
```
ELKS typical PC memory layout of 1M physical memory

 +--------------------  seg  off  linear size (size decimal)  --------------------+
 | interrupt vectors    0000:0000 00000  0400 (1024)
 | ...
 | BIOS data area       0040:0000 00400  0100 (256)
 | bootopts             0050:0000 00500  0100 (256)
 | kernel setup data    0060:0000 00600  0200 (512)
 | ...
 | disk track cache     0080:0000 00800  2400 (9216 = SECTOR_SIZE * 18)
 | ...
 | kernel code segment  02C0:0000 02C00  F910 (63760 typical)
 | ...
 | kernel data segment  1251:0000 12510 10000 (65536=64K)
 | ...
 | external sys buffers 2251:0000 22510       (default=64k, CONFIG_FS_NR_EXT_BUFFERS=64)
 | ...
 | application memory   3251:0000 32510       (remaining memory after system buffers)
 | ...
 | ...
 | top of 640k ram      A000:0000 A0000 A0000 (640K)
 | EGA ram              A000:0000 A0000 10000 (65536)
 | MDA monochrome ram   B000:0000 B0000  4000 (16384)
 | CGA color ram        B800:0000 B8000  4000 (16384)
 | network ROM          C000:0000 C0000 10000 (65536)
 | ramdisk memory       D000:0000 D0000 20000 (131072=128K)
 | ramdisk memory       E000:0000 E0000
 | BIOS ROM             F000:0000 F0000 10000 (65536)
 +--------------------------------------------------------------------------------+
```

Jookia, 2022-05-03:
Ok I can confirm that this issue was purely to do with memsize, with the floppy being some misconfiguration as DOSBox-X doesn't support larger geometry than 1.4MB preset.

Thanks for the explanation, that makes sense to me. Should XMS be disabled if the RAM is less than 1MB? Is that a patch you would welcome if I could write it?

ghaerr, 2022-05-03:
> Should XMS be disabled if the RAM is less than 1MB?

Yes. There are a number of caveats to using XMS, CPU type, INT 15h capabilities, special protected mode BIOSes, etc.
The v0.5.0 release should have had it disabled, and it will be disabled in the next release.

> Is that a patch you would welcome if I could write it?

No, there are too many other issues associated with making XMS work properly automatically on all PC hardware, and there are reasons why we don't want the kernel to run a RAM checker before booting, as that has its own set of problems based on varying hardware.

The ELKS configuration system is specifically designed to allow users to add/subtract features for their particular systems, the takeaway from this and other similar issues is to ensure that the base download has esoteric options that might cause system failure to default to off.

Thank you!

closed 2022-06-16 by ghaerr