ghaerr/elks #1277: MicroPython?

jedie, 2022-05-04:
Will it ever be possible to get MicroPython running?!?

cocus, 2022-05-04:
It seems to be quite modular though, so it should be possible to disable features until it fits. There's a unix port already, so it should be "easy" to port it. I see heavy usage of pointers with malloc/free, and no regards to the segmented memory model. There's a dsPIC33F variant (16 bit MCU) which although it might contain heaps of flash, it is indeed limited (and a 16 bit target). I had a glance at the config but I couldn't understand how to selectively enable or disable stuff so the binary size gets as small as possible.

closed 2022-06-16 by ghaerr

jedie, 2022-06-20:
> I couldn't understand how to selectively enable or disable stuff so the binary size gets as small as possible.

Maybe good starting point is the minimal port -> https://github.com/micropython/micropython/tree/master/ports/minimal