ghaerr/elks #1295: ELKS 0.5.0 hangs up on boot XMS problem?

sergione78, 2022-05-17:
Hello, i made a boot floppy for my commodore pc 10.
The boot seems to proceed well, but the pc hangs up on this message: xms: A20 was off
I thought there was an issue with the media, so I made another 5,25 boot disk using the other image, but nothing, same issue.
[
[image]
](url)

Later I made another boot floppy for my toshiba t1200, and experienced the same issue

[image]

but this time the boot process went on by just pressing the enter key.  The XMS was disabled. 

my cpu is 8088
stock ati video card (mda, and perhaps hercules or cga emulation)
640 kb ram
2 x 5,25 floppy double density disk drive 

even though I don't have a lot of command line experience, I really appreciate EKLS.
My goal is to use linux to access my local network. With ms-dos typing a program in gwbasic I was able to transfer the interlink program with a null-modem cable and from this I was able to download other software, including the EKLS disk image.
Any suggestions?

ghaerr, 2022-05-17:
Hi @sergione78,

The problem you are having is the same as #1267 - the ELKS v0.5.0 disks were incorrectly made with XMS support turned on, which doesn't work on a majority of older systems. You should be able to (re)download one of the latest builds [for instance here](https://github.com/jbruchon/elks/pull/1294/checks) under the "Artifacts" button on the upper right, which should work correctly.

Thank you!

sergione78, 2022-05-17:
thank you very much! I'm sorry if I didn't notice before.

Mellvik, 2022-05-18:
@ghaerr,

maybe it's time to eliminate this recurring problem by moving to 0.6.0 - thus enacting on our stated intentions after 0.5.0 to reduce the interval between releases.

—M

closed 2022-06-16 by ghaerr