ghaerr/elks #1303: Network problem

toncho11, 2022-05-30:
I am using WD driver with the new parameters in /bootopts on a 386.
The parameters are:  IRQ = 5, 300, 0XCA000.
There is no error at boot. The driver reports the MAC address. 

I configure the ktcp like that: `ktcp -b -p eth 192.168.1.82 192.168.1.254 255.255.255.0 ` which matches my network. There is still no error.

The problem is that "ping" on my laptop gives "host unreachable" and I am unable to use telnetd on ELKS. The "arp" command in ELKS is aware of the IP and MAC of my laptop, so there is some network activity.

Mellvik, 2022-05-30:
Hi @toncho11,

To me it sounds like your interrupts aren't working. The port is obviously correct since you're (presumably) getting the correct physical (MAC) address back from the card. 
When interrupts are not working, the card will appear dead to the net. I'd recommenct you double check that the card is actually set to IRQ 5 and that this IRQ is indeed available. If you have a hard disk, it may be using IRQ5, and IRQ 2 or 3 are generally better bets.

If you can peek at what's on your net cable (wireshark, tcpdump etc), you should see a single packet coming from ELKS when running a network command such as nslookup, telnet or ftp. This first (ARP request) packet doesn't need interrupts to leave the system. If you can verify this and see the packet,  you know that the card is working and the driver is working, just not the interrupts.

--M

Mellvik, 2022-05-30:
@Toncho11,
correction - since you're now on a 386, try IRQ 2 or 9.

—M

Mellvik, 2022-05-30:
> It is a jumper card with only two options:
>
> IRQ 3, 280, 0xD000
> IRQ 5, 300, 0xCA00
>
>  <https://user-images.githubusercontent.com/1851997/170946724-1fed6779-55dd-408f-8fb0-533f7c00f8c0.png>
> On the card it is written with 1 more 0 like this "0xD0000" or "0xCA000", but I think I need to remove the last 0 when configuring ?
>
> There is also a second jumper.
> This card is known to work on both DOS and ELKS, but this is the first time I use it on the 386.

OK; thanks, I get it @toncho11 -

Actually on the 386 and unless you have 2 serial ports IRQ3 should be fine, and IRQ5 too unless you have 2 parallell ports.

The 5 digit memory address is - IIRC - the linear address of the shared memory, you take of the rightmost 0 to get the segment address. The default as set in the driver is 0xCE00.

The card seems to support PnP so you can probably set other combinations of IRQ/PORT/SHMEM using one of the many DOS utilities for such configuration and moving the jumper to 'soft'.

Using the card on the 386 should be no problem unless you have many ISA cards installed with possible conflicting configurations.

As to changing bootopts and hating the screen editor (or not having one), I suggest you use sed. Say you want to change IRQ from 12 in the current bootopts to 5, you can do this:
cd /
sed -e 's/=12/=5/' < bootopts > xxx
….and then
mv bootopts bootsave
mv xxx bootopts
…. and a similar procedure for other changes. Just s/old/new/ (the sed substitute command) and make sure 'old' is unique so the change occurs where you want it to be.

Then reboot.

-M

toncho11, 2022-05-30:
I can not make it work.

There are some options for shadowing in the BIOS. There is for the BIOS, VIDEO, 640-1024 remapping and shadow for specific addresses. 
It sees my computer when I type "arp", but no ping response. 
I disabled the hdd in the BIOS. There are no other cards.
It seems that ELKS always sees IRQ 3 as serial console, no matter what.
I was testing with IRQ 5.

I am giving up.

toncho11, 2022-05-30:
When I ping the 386 I see that the "ARP replies sent" when checking the with netstat on ELKS.
But all other types of packages stay at 0: TCP , IP, ICMP, etc.
So arp is working, but nothing else.

toncho11, 2022-05-30:
The number of send/received packets TCP/IP increased while I was trying to make a connection. 
I think ktcp is not announcing ELKS on the network properly???

Mellvik, 2022-05-31:
@toncho11,
This is interesting. What happens when you attempt to make a connection fromELKS is that it sends an ARP request packet. So the arp command should at least show 1 on the ARP requests sent. 
If netstat lists ANY received packets at all, your interrupts are working and your problem is most likely the shared memory address

It's unlikely that ktcp is the culprit as it is working well with other interfaces and does not distinguish between types.

Can you post a screenshot of the boot messages and the netstat output after and attempt to connect outwards?

-M

toncho11, 2022-05-31:
I was getting 30 arp packets in both send/receive when trying to do a ping.
And with time I was able to reach 4 TCP and IP packets in both send and receive. It is obvious the TCP/IP communication is disturbed. 
My laptop keeps saying "host unreachable", but the some arp communication keeps passing.

toncho11, 2022-05-31:
[image]
Boot with all set in config files.

[image]
First netstat after boot.

[image]
Netstat after 1 ping to Elks

I think elks is not entering in the arp table of my laptop and that is why they can not communicate.

Mellvik, 2022-05-31:
Thank you @toncho11,
This helps. You card is definitely working, interrupts and all.
Also it seems you're receiving packets correctly -  since your netstat reports both an ARP cache add and a received IP packet (presumably the same packet). SO the shared mem address is OK too.

My next thought is that there is a problem with sending in the driver. I'll take a look and see if anything stands out.

Do you have tcpdump or something similar on the machine you're pinging from? It would be very helpful to see what's on the wire.

—M

Mellvik, 2022-05-31:
> I am checking: https://www.practicalnetworking.net/series/arp/arp-probe-arp-announcement/ <https://www.practicalnetworking.net/series/arp/arp-probe-arp-announcement/>.
>
> The Gratuitous ARP is sent as a [broadcast](https://www.practicalnetworking.net/stand-alone/local-broadcast-vs-directed-broadcast/), as a way for a node to announce or update its IP to MAC mapping to the entire network.
>
> The first use case is pretty straight forward, a node can use a Gratuitous ARP to update the ARP mapping of the other hosts on the network should the node’s IP to MAC mapping change. This might happen if a user manually modifies their MAC address – they retain the same IP address, but now have a new MAC address. Therefore, the ARP mapping for all the nodes which are communicating with this user must be updated.

This is good research @toncho11 - 
and I agree with you - ARP doesn't seem to work right. 

However, ELKS ARP is known to work with other interfaces so it's still more likely a driver (or possibly a card config) problem. And evidently, ELKS is receiving ARP requests and responding to them.

ELKS does not implement Gratuitous ARP - which has given me some trouble when switching between interfaces over the past months, but never more serious than 30 seconds of non-response.

Questions:

1) Have you had this interface working with ELKS or any other system recently (without using the soft config mode)?
2) When you have pinged ELKS as in the pictures and netstat reports 7 or some packets, does the arp info on ELKS look OK? (arp command)
3) What does the arp command at the other end say?
4) Is there a chance you can activate ARP debugging and recompile ktcp so we can see what happens (config file in the ktcp directory)?

I'm assuming you don't have a way to do packet tracing - tcpdump etc.

—M

toncho11, 2022-05-31:
1) Yes. I usually use this card in another computer, but it was more than a month ago. 

2) yes. It lists my laptop IP and MAC.

3) On my laptop it does not list the ELKS 386 computer. I think this is the problem. 

4) Using TCP dump on my laptop will be easier I suppose.

Mellvik, 2022-05-31:
Great, thanks @toncho11,
that narrows it down quite a bit.

So either 
- the ARP reply packets are not leaving the interface 
or
- the ARP reply packets have errors in content or format

And I agree, using tcpdump on your (linux) laptop is a fast path to figure out which of the problems we have.

Assuming you have tcpdump installed, all you need to do is to run
tcpdump -v -i <if>
where <if> is the name of the laptop interface you talk to your ELKS machine via, as displayed but the ifconfig command - e.g. eth1.
You can skip the -i option to tcpdump if the interface is eth0 (the first physical interface on the machine).

You may have to filter the traffic if you have many chatty machines on there, here's the tcpdump command I keep running all the time:
tcpdump -v not host 10.0.2.100 and not host 10.0.2.42

or you can do the opposite, specify the traffic you'd like to see instead:
tcpdump -v host 192.168.1.8

—M

toncho11, 2022-05-31:
It is both working a little bit and not working.

[image]

I added a new entry on my laptop and ping still does not work (actually it worked one time). Also we can see Elks received Telnet and Ftp connections, but finally it could not connect.

toncho11, 2022-05-31:
I even saw "login" for the telnet one time, but all the other attempts failed.

ghaerr, 2022-05-31:
Hello @toncho11 and @Mellvik,

I agree with @Mellvik's line of questioning, and it seems we could be dealing with a NIC that is possibly not sending (or receiving) packets reliably, even though it is non-interrupt driver for both sending and receiving. A tcpdump of the packets seen on the wire will help greatly. We thought that the WD8000 driver was working well, but I think this has only been verified on a much slower system that the original author used to write it. 

At one point, when I was helping debug @Mellvik's NE2K driver on a faster machine (386), I seem to remember we found that the driver was checking one of two transmitter empty bits, and switching to the other one allowed the card to start operating properly, as the problem seemed to be that the NIC send buffer was getting overrun by ktcp sending a couple packets out quickly, with one of them getting stomped. Thus, some inspection of how the WD8000 waits to send packets might be in order, OR, trying this entire setup on an older machine, to verify that the WD8000 driver in fact still works on a slower system.

Thank you!

Mellvik, 2022-05-31:
Thanks @ghaerr - the speed is indeed a valid point,
and I'll take an extra look at the WD transmit code in the morning.

Still @toncho11 - we do need some packet traces…

BTW I just got interrupts working in a 3c509 driver for ELKS, but don't wait for it to be ready for 0.6.0.

—M

toncho11, 2022-05-31:
@ghaerr 

Let's start with the original computer my Amstrad 1640. 
I need a 360kb ELKS with FAT and /bootopts configured for:
IRQ = 3, 280, 0xd000

I will also test with wireshark later.

ghaerr, 2022-05-31:
> Let's start with the original computer my Amstrad 1640.
I need a 360kb ELKS with FAT and /bootopts configured for:
IRQ = 3, 280, 0xd000

I can't fit the networking files on that floppy... they won't all fit on 360k, same problem as before.

toncho11, 2022-05-31:
[image]

I see "login", but I can not login on Telnet from my laptop.

toncho11, 2022-05-31:
So we finally got some error messages.

Mellvik, 2022-05-31:
@toncho11 ,
This looks familiar. Try disconnecting all other nodes on the network, so you have only elks and the laptop. ALso change the IP address of the elks system. It seems you may have a node (or interface) shanghaiing the address - either the elks address or the laptop's address. A tcpdump will tell you what's going on.

-M

toncho11, 2022-05-31:
Typing "arp" is required every time I click "enter" in telnet. I was able to login.

ghaerr, 2022-06-01:
> is it possible that you gave me some custom image where for example some things are configured for QEMU? I saw some error messages related to QEMU.
QEMU is commented in /bootopts though.
Maybe something at compiled level?

No, there are no special QEMU compilation options for the kernel or user programs.

> I have this "xx" file in my home folder.

Yes, that's a test shell script I had on my ELKS root system directory when I quickly built the image for you. You can ignore it.

> Typing "arp" is required every time I click "enter" in telnet. I was able to login.

> I think the arp command is also updating the arp table. It definitely does something more which fixes the problem. The arp command is unblocking something.

The arp command sends a message to ktcp, which could be causing a retry, but does not update the arp table. To see whether the "send a message" might be causing ktcp to retry, just try running "netstat" instead, to see whether this might cause the same result. However, I don't think ktcp or arp are likely defective, the issue is very much more likely in the WD8000 driver, although it could certainly be helped along by being "bumped" by arp or netstat. We really need tcpdump to see what's on the wire (or recompilation with DEBUG_ETH/ARP/etc on and ^P) in order to get more detailed error messages. 

>  Try disconnecting all other nodes on the network, so you have only elks and the laptop. ALso change the IP address of the elks system. It seems you may have a node (or interface) shanghaiing the address - either the elks address or the laptop's address. 

I definitely agree: try @Mellvik's suggestion, we could be having address duplication errors.

toncho11, 2022-06-01:
Thanks. 
nestat also fixes the problem temporally. So @ghaerr what does this mean?
I could find this card: SMC 60-600619-002 REVA. Is this card SMC/WD ?
I think what is happening is inside ELKS, so I do not see how packet sniffing will help ?
I get some "tcp retrans" messages with "(RETRY 1 cnt 3).

toncho11, 2022-06-01:
So when I type arp on ELKS it does that:

`177	61.546094	192.168.1.8	192.168.1.47	ICMP	74	Echo (ping) reply    id=0x0001, seq=230/58880, ttl=64`

Not every time though.

When I start pinging ELKS nothing happens. My laptop does not know where to send it. 
Then while pinging I do type arp and I get some replies: 
[capture_ping_and_arp_in_the_middle.zip](https://github.com/jbruchon/elks/files/8817763/capture_ping_and_arp_in_the_middle.zip)
The file is a json from Wireshark of 6 packages.

toncho11, 2022-06-01:
If I keep typing netstat (or arp) I get all the 4 replies from ELKS:
[all_replies.zip](https://github.com/jbruchon/elks/files/8817836/all_replies.zip)

This is similar to above, but with all replies.

Mellvik, 2022-06-01:
Ok, we're getting there.
Can you post ALL output from tcpdump related to the elks machine? Sent, received, broadcast, arp, anything?

The timestamps will tell if you had to give an arp or netstat cmd.

Thank you!

-M

toncho11, 2022-06-01:
These are the dumps I can do for now. Attached previously.
@Mellvik Is this the card you need SMC 60-600619-002 REVA ?

toncho11, 2022-06-02:
OK. I think the best course of action is if @Mellvik can get a SMC/WD card. This work from distance we are trying is hard (not very productive). @Mellvik I will look for a card for you. The alternative is if I send you my own card 1-2 months. I can live without it for while :).

toncho11, 2022-06-02:
What about this card https://www.ebay.co.uk/itm/175301827859

Mellvik, 2022-06-02:
@toncho11,

yes, this seems time consuming. But regardless of how we decide to move forward, we need to determine - or rather - eliminate the chance that something in your environment is causing or contributing to the problem. Which doesn't necessarily mean that something is wrong or faulty but rather a condition that is triggering a problem not otherwise easily found.

SO let's to the tracing first.
I cannot find any tool that will read your json files. Can you produce pcap files from say, tcpdump or wireshark, or just compressed text from tcpdump?

I would be happy to borrow your card for a month or two and the return it, if we decide that's the way to go. There is a card in eBay, but it's at $140, which seems excessive.

—M

toncho11, 2022-06-02:
Did you see the two cards the 3 cards I mentioned before:
SMC 60-600619-002 REVA
https://www.ebay.co.uk/itm/175301827859 (but it needs BNC to Ethernet_
https://www.ebay.fr/itm/125119298189

Unfortunately I can not devote more time on checking card. Or at least in the next weeks.

ghaerr, 2022-06-02:
Hello @Mellvik,

> I've looked through the send side of the wd driver. It doesn't look good to be honest - there is a definite danger over overrunning the transmit buffer in the middle of a transmit. (The test for transmitter busy comes after the driver has written the packet data to shared memory, instead of before.)

I don't see that - it appears the test you're referring to is a double-check, as the `wd_pack_put` function is called by `wd_write` only *after* the transmitter is flag is tested and shown empty by calling the `wd_tx_stat` function first. The `wd_write` function will sleep the process until it is an appropriate time to write a packet.

> However I don't think this is what's causing the problem here as it would require back-to-back send packets in order to pan out. So for this case, packet tracing is the only way ahead AFAIK.

Agreed. Was the possibility of having multiple local network addresses resolved, per your previous question?

Hello @toncho11,

> netstat also fixes the problem temporally. So @ghaerr what does this mean?

This means, as I suspected, that running neither arp nor netstat are part of the fix or problem, running them sends a message to ktcp and likely causes a return from a main loop wait, which causes another retry, thus seemingly, but not really, fixing the problem. I suspect @Mellvik is probably correct and that there remains internal or external setup issues with the NIC, or possibly a heretofore unseen timing-related issue. Without adequate tracing, lots of time can can be spent guessing.

Thank you!

Mellvik, 2022-06-02:
Hi @ghaerr -

> I've looked through the send side of the wd driver. It doesn't look good to be honest - there is a definite danger over overrunning the transmit buffer in the middle of a transmit. (The test for [...]
You're right, it actually seems OK.
> However I don't think this is what's causing the problem here as it would require back-to-back send packets in order to pan out. So for this case, packet tracing is the only way ahead AFAIK. [...]
No, I never got traces that I could read.

When I get the hardware, we'll know more, but @toncho11 - pcap traces from your setting would still be very useful to rule out local 'peculiarities'.

-M

toncho11, 2022-06-02:
Here are two recordings with me pinging ELKS using tcpdump for Windows.
[windump_pcap.zip](https://github.com/jbruchon/elks/files/8827914/windump_pcap.zip)

Mellvik, 2022-06-03:
Thank you @toncho11,

this was very helpful indeed.

It turns out we're back to exactly where we started: Your interface is not generating receive interrupts, quite possibly not interrupts at all. At least, ELKS is not receiving them.

What happens is that every time you poke ktcp (with arp or netstat), it polls the interface, finds incoming traffic, reads it all in one swoop, and starts responding to each and every packet. So when the other system has sent 10 ARP request w/o response, ELKS sends 10 ARP replies back when it discovers them.

An entirely different thing is that the pings being sent towards ELKS have illegal format (are they coming from Windows?) and I'm not convinced ELKS would reply even if things worked. tcpdump is not always reliable when it comes to checksums, but a checksum that is always zero as is the case here, is definitely wrong.

It's hard to remotely diagnose the absence of interrupts, so I'm reverting to my previous advice: Change the interface jumper to soft config and configure it via a utility on MSDOS.

-M

toncho11, 2022-06-03:
Thank you for your analysis.

Yes. I was on Windows 10. I used [WinDump](https://www.winpcap.org/windump/).
These interrupts depend on which parameter in: 

`IRQ 5, 300, 0xCA00` ?

Maybe there is a parameter that is not being configured correctly. Maybe two parameters are switched.

Mellvik, 2022-06-03:
@toncho11,

the IRQ is the '5' in the sequence and there is nothing wrong with the way it's being used, as evidenced by the BOOT message.

If there is a BIOS setup on your 386, it may be a good idea to take a look and see if IRQ5 is being assigned to anything in particular, and possibly change that, make it available to the ISA bus. Obviously, everything else is fine (i.e. the IOport and the shared memory address).

Also, the next time you test, use your mac instead of windows. ping on the mac works correctly and tcpdump is included (IIRC).

—M

toncho11, 2022-06-03:
No. The hdd is removed. The BIOS has only "Shadow memory" and EMS config. EMS says is dsiabled though. There is no IRQ config in the BIOS. The BIOS does not print any info at boot which IRQ is assigned to what.

Mellvik, 2022-06-18:
@toncho11 -
I just received the WD8003E card, which turns out to work great 'out of the box' with the current driver. Surprisingly well actually - the speed this 8bit card delivers is on par with the NE2k and 3C509 16 bit cards, which demonstrate that shared memory is faster than direct IO. It will be interesting to see how the 16bit version of the card performs.

Back to your problem,
I may have asked this before: Your HDD has been removed, has the HD controller been removed too - or disabled in some way, freeing up IRQ5? It seems we're back to your NIC interrupts not working correctly.

-M

toncho11, 2022-06-18:
I disabled the HDD in the BIOS as "Not installed".

Mellvik, 2022-07-22:
@toncho11,
I recently wasted a lot of time trying to get IRQ5 to work on a 386(SX) - first with a serial line, then with a NIC. I never found the reason, but using that particular IRQ not only didn't work but it made the system unstable in general. So something is going on under the hood and docs are sketchy at best. So I suggest just leaving IRQ5 alone.

I ended up using IRQ7 (after disabling the parallel port) and 10 for serial 3 and 4 and IRQ2/9 for the NIC. The NIC is an 8bit card btw. 

-M

closed 2022-09-22 by toncho11

Mellvik, 2022-10-11:
Ok,
I think the disk controller may still be active in the sense the it's polluting your irq5. If this is the only irq available, look for a jumper on the motherboard that may disable the controller entirely.

-m

Mellvik, 2022-10-11:
Thank you @toncho11,

> 
> Thanks.
> nestat also fixes the problem temporally. So @ghaerr <https://github.com/ghaerr> what does this mean?
> 
It's confirming @ghaerr's assumption - you're waking ups ktcp which probably triggers either a read, a transmit or both so that a few more packets pass through.
> I could find this card: SMC 60-600619-002 REVA. Is this card SMC/WD ?
> 
This seems to be a wd8013 variant, which is the 16 bit variant of the WD8003. I'm not familiar with the card and there are many variants out there. Will look closer at this.
> I think what is happening is inside ELKS, so I do not see how packet sniffing will help ?
> 
You're having traffic, the interesting part is on the wire, that will tell us what ELKS is sending and receiving. We don't know if there is a ELKS problem yet. Without seeing what's on the wire, we've nothing to work with. We could also enable debugging options in ELKS; but tcpdump is much easier and possibly more useful-
> I get some "tcp retrans" messages with "(RETRY 1 cnt 3).
> 
> 
YEs, this is the part that makes me think you may have bogus addresses on your net, thus the request to disconnect all other hosts.
You'll see that easily when tcpdump is activated.

—M

Mellvik, 2022-10-11:
@toncho11,

I was referring to the boot messages from ELKS. They show that your config is fine as far er the driver and ELKS are concerned.

Are you saying that the hard disk controller has been removed or just the hard disk? 

—M