ghaerr/elks #1321: NE2K in 8bit mode, lot of errors

cocus, 2022-06-15:
Hi,
I've configured a new network card, one that I've mentioned before (the UMC UM9008F) with IRQ = 3, IO = 300. On my SBC, the ISA IRQ 3 is mapped to the INT1 pin, which in turns raises a "logical" (or fake) IRQ 12.

I've started a telnet client and connected to it (I'm using the same setup as the other NE2K in 8bit mode I've always used).

This is the output that I get:
```# net start
Starting networking on eth
ktcp -b 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmaseth: RX-error, status 0x21
k 255.255.255.0
ktcp: ethernet 00.50.4e.02.f7.82 mtu 1500
Starting daemons 'telnetd' 'ftpd -d' 
# eth: RX-error, status 0x21
accept from 192.168.137.1:16847
getpeername 192.168.137.1:16847
term_init ok, pid 16
eth: RX-error, status 0x01
eth: RX-error, status 0x01
eth: RX-error, status 0x05
eth: RX-error, status 0x21
eth: RX-error, status 0x05
eth: RX-error, status 0x05
eth: RX-error, status 0x01
eth: RX-error, status 0x05
eth: RX-error, status 0x01
eth: RX-error, status 0x01
eth: RX-error, status 0x01
eth: damaged packet (8001 64), clearing buffer
ktcp: eth_process error -1 (errno 5), discarding packet
eth: TX-error, status 0x01
eth: TX-error, status 0x01
eth: TX-error, status 0x01
eth: RX-error, status 0x01
eth: rcv oflow (0x53), keep 1
eth: rcv oflow (0x53), keep 1
tcp retrans: seq 454+19 size 19 rcvwnd 63787 unack 454 rto 8 rtt 1 (RETRY 1 cnt 1 mem 39)
eth: RX-error, status 0xa5
eth: RX-error, status 0x01
eth: rcv oflow (0x53), keep 1
eth: rcv oflow (0x53), keep 1
eth: rcv oflow (0x53), keep 1
tcp retrans: seq 665+48 size 48 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 8 mem 398)
tcp retrans: seq 646+19 size 19 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 8 mem 398)
tcp retrans: seq 857+27 size 27 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 9 mem 420)
tcp retrans: seq 836+21 size 21 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 9 mem 420)
tcp retrans: seq 790+46 size 46 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 9 mem 420)
eth: RX-error, status 0x9d
tcp retrans: seq 761+29 size 29 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 9 mem 420)
eth: rcv oflow (0x53), keep 1
tcp retrans: seq 742+19 size 19 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 9 mem 420)
tcp retrans: seq 713+29 size 29 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 9 mem 420)
tcp retrans: seq 884+2 size 2 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 10 mem 488)
tcp retrans: seq 857+27 size 27 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 836+21 size 21 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 790+46 size 46 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 761+29 size 29 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 742+19 size 19 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 713+29 size 29 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
eth: rcv oflow (0x53), keep 1
tcp retrans: seq 665+48 size 48 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 646+19 size 19 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 10 mem 488)
tcp retrans: seq 886+48 size 48 rcvwnd 63595 unack 646 rto 8 rtt 1 (RETRY 1 cnt 11 mem 623)
tcp retrans: seq 884+2 size 2 rcvwnd 63595 unack 646 rto 16 rtt 1 (RETRY 2 cnt 11 mem 623)
tcp retrans: seq 857+27 size 27 rcvwnd 63595 unack 646 rto 32 rtt 1 (RETRY 3 cnt 11 mem 623)
tcp retrans: seq 836+21 size 21 rcvwnd 63595 unack 646 rto 32 rtt 1 (RETRY 3 cnt 11 mem 623)
tcp retrans: seq 790+46 size 46 rcvwnd 63595 unack 646 rto 32 rtt 1 (RETRY 3 cnt 11 mem 623)
eth: RX-error, status 0x01
tcp retrans: seq 761+29 size 29 rcvwnd 63595 unack 646 rto 32 rtt 1 (RETRY 3 cnt 11 mem 623)
tcp retrans: seq 742+19 size 19 rcvwnd 63595 unack 646 rto 32 rtt 1 (RETRY 3 cnt 11 mem 623)
[... 69 lines cut ...]
tcp retrans: seq 665+48 size 48 rcvwnd 63904 unack 646 rto 64 rtt 1 (RETRY 6 cnt 7 mem 779)
eth: RX-error, status 0x67
tcp retrans: max retries exceeded seq 665 unack 646 time 64
tcp retrans: seq 646+19 size 19 rcvwnd 63904 unack 646 rto 64 rtt 1 (RETRY 6 cnt 6 mem 711)
tcp retrans: max retries exceeded seq 646 unack 646 time 64
eth: damaged packet (8001 64), clearing buffer
ktcp: eth_process error -1 (errno 5), discarding packet
tcp retrans: seq 884+2 size 2 rcvwnd 64120 unack 646 rto 64 rtt 1 (RETRY 6 cnt 5 mem 672)
tcp retrans: max retries exceeded seq 884 unack 646 time 64
tcp retrans: seq 1049+159 size 159 rcvwnd 64120 unack 646 rto 64 rtt 1 (RETRY 6 cnt 4 mem 650)
tcp retrans: max retries exceeded seq 1049 unack 646 time 64
tcp retrans: seq 934+115 size 115 rcvwnd 64120 unack 646 rto 64 rtt 1 (RETRY 6 cnt 3 mem 471)
tcp retrans: max retries exceeded seq 934 unack 646 time 64
tcp retrans: seq 886+48 size 48 rcvwnd 64120 unack 646 rto 64 rtt 1 (RETRY 6 cnt 2 mem 336)
eth: RX-error, status 0x67
tcp retrans: max retries exceeded seq 886 unack 646 time 64
tcp retrans: seq 1208+248 size 248 rcvwnd 64120 unack 646 rto 64 rtt 1 (RETRY 6 cnt 1 mem 268)
tcp retrans: max retries exceeded seq 1208 unack 646 time 64
```

This code comes exactly from my PR adding HEX$() support on basic.

I think this could be interesting for @Mellvik.

Thanks!

Mellvik, 2022-06-15:
Thank you Santiago - you're right, this is very interesting.

I assume this is a 16 bit card that you're trying to run in 8 bit mode. That should work fine, but there is (currently) no way for the driver to figure out that the upper part of the bus is missing. So you'll have to tell it by setting the least significant bit in the IO address in /bootopts - in this case to 301 instead of 300. This is the 'force 8 bit mode' setting.

Try that first - and send a copy of the boot messages (ethernet config message). The card is obviously working fine, the interrupts are working, the driver is interacting perfectly well with the NIC because there is nothing 16 bit going on until the actual data IO.

When you have this working, we can add this card to the list of supported ethernet NICs in ELKS!

—M

Mellvik, 2022-06-21:
@cocus, what's the status on this issue - can it be closed?

cocus, 2022-06-21:
Hi @Mellvik , I'm going to attach the boot log of the two cards, using last friday's commit, since I don't want to mess with the new multi-card code as of yet.
The original card I was using didn't show any error. The new card, after being properly configured with a DOS tool on another machine to match the other card, showed a lot of errors. I've forced it to be on 8bit, since that option was set on the config file. It wasn't detected as 16 bit. I can change the address to 301 on the .h file (since there's no support for bootopts on this platform) but the message was quite clear stating it's in 8 bit mode.

Good card:
```
8018x ELKS loader built at vie 17 jun 2022 20:39:12 -03 (LEDS NE2K BQ3285)
Now jumping into ELKS...
console_init: 8018X UART
eth: NE2K (8 bit) at 0x300, irq 12, MAC 00:40:05:33:ee:34
ssd: initialization error
8018X machine, 512K base RAM.
ELKS kernel 0.6.0 (45888 text, 0 ftext, 5456 data, 38160 bss, 21918 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 446K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# ktcp -b 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmask 255.255.255.0
ktcp: ethernet 00.40.05.33.ee.34 mtu 1500
# telnetd
# accept from 192.168.137.1:56500
getpeername 192.168.137.1:56500
term_init ok, pid 7
finished waitpid, pid 7
```

Bad card, multiple attempts to run telnetd:
```
8018x ELKS loader built at vie 17 jun 2022 20:39:12 -03 (LEDS NE2K BQ3285)
Now jumping into ELKS...
console_init: 8018X UART
eth: NE2K (8 bit) at 0x300, irq 12, MAC 00:50:4e:02:f7:82
ssd: initialization error
8018X machine, 512K base RAM.
ELKS kernel 0.6.0 (45888 text, 0 ftext, 5456 data, 38160 bss, 21918 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 446K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# ktcp -b 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmask 255.255.255.0
ktcp: ethernet 00.50.4e.02.f7.82 mtu 1500
# telnetd

8018x ELKS loader built at vie 17 jun 2022 20:39:12 -03 (LEDS NE2K BQ3285)
Now jumping into ELKS...
console_init: 8018X UART
eth: NE2K (8 bit) at 0x300, irq 12, MAC 00:50:4e:02:f7:82
ssd: 1977614336K disk
8018X machine, 512K base RAM.
ELKS kernel 0.6.0 (45888 text, 0 ftext, 5456 data, 38160 bss, 21918 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 446K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# ktcp -b 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmask 255.255.255.0
ktcp: ethernet 00.50.4e.02.f7.82 mtu 1500
# eth: RX-error, status 0xa1
eth: RX-error, status 0x21
eth: RX-error, status 0x0c
eth: RX-error, status 0x25
teth: RX-error, status 0x21
elnetd
eth: RX-error, status 0x21

8018x ELKS loader built at vie 17 jun 2022 20:39:12 -03 (LEDS NE2K BQ3285)
Now jumping into ELKS...
console_init: 8018X UART
eth: NE2K (8 bit) at 0x300, irq 12, MAC 00:50:4e:02:f7:82
ssd: 1977614336K disk
8018X machine, 512K base RAM.
ELKS kernel 0.6.0 (45888 text, 0 ftext, 5456 data, 38160 bss, 21918 heap)
[... 193 lines cut ...]
# ktcp -b 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmask 255.255.255.0
ktcp: ethernet 00.50.4e.02.f7.82 mtu 1500
# eth: RX-error, status 0xa5
eth: RX-error, status 0xed
eth: RX-error, status 0xa1
eth: RX-error, status 0x25
eth: RX-error, status 0x21
eth: RX-error, status 0x25
eth: RX-error, status 0xa1
eth: RX-error, status 0x25
eth: RX-error, status 0x21
eth: RX-error, status 0x21
teleth: RX-error, status 0x21
neth: RX-error, status 0x21
etd
eth: RX-error, status 0x25
eth: RX-error, status 0x25
eth: RX-error, status 0x21
eth: RX-error, status 0x25
eth: RX-error, status 0x21
eth: RX-error, status 0x21
eth: rcv oflow (0x14), keep 1

```

ghaerr, 2022-06-21:
> I've forced it to be on 8bit, since that option was set on the config file. It wasn't detected as 16 bit. 

FYI, as I was removing the CONFIG_ETH_BYTE_ACCESS option this morning, I saw that none of the NIC drivers actually used that option - it had no effect. Any use of it seems to have been removed prior.

cocus, 2022-06-21:
> FYI, as I was removing the CONFIG_ETH_BYTE_ACCESS option this morning, I saw that none of the NIC drivers actually used that option - it had no effect. Any use of it seems to have been removed prior.

Good point. It means the driver was smart enough to detect it as 8 bit :)

Mellvik, 2022-06-21:
Thank you Santiago,
Are you seeing a lot of broadcast packets on this segment? It seems the nic is receiving traffic even before the startup process is finished?

Do you have a picture of the two interfaces? Does the one with problems have multiple attachment options (bnc, aui) by any chance?

Btw, try setting the io address to 303, lets eliminate the chance that there is a buffer memory problem.

-M

cocus, 2022-06-21:
> Are you seeing a lot of broadcast packets on this segment? It seems the nic is receiving traffic even before the startup process is finished?

It's connected to my Windows laptop, I'm using the "share internet" feature, so it's technically the only device connected to the ethernet adapter, but Windows usually sends a lot of trash.

> Do you have a picture of the two interfaces? Does the one with problems have multiple attachment options (bnc, aui) by any chance?

Yes. Both have BNC and TP. I've pre-configured both on DOS, IRQ3, address 0x300. I've then loaded Windows 98 with the generic NE2K compatible driver (and selected IRQ3 and address 0x300), and it worked fine for BOTH cards.

[image]

(top=good, bottom=bad)

> Btw, try setting the io address to 303, lets eliminate the chance that there is a buffer memory problem.

I'll try this. Is there any risk of loading the latest sources (i.e. with the multi-card first pass code in it?)

cocus, 2022-06-21:
Now that I see this closely, the UMC board doesn't have an external RAM, like the other card does. Not sure if the UMC chip has internal RAM, but it could be related

Mellvik, 2022-06-22:
Thank you @toncho11,
> ﻿
> Now that I see this closely, the UMC board doesn't have an external RAM, like the other card does. Not sure if the UMC chip has internal RAM, but it could be related
> 
The UMC chip is known to be NE2k (8390) compatible, so that should not be a problem.
What may be tha problem is that the card is configured to use the BNC (thin ethernet) connector, and not the TP (rj45).
Can you check this setting using the same configuration utility you used to check the irq/address settings?

--M
>

cocus, 2022-06-22:
> Can you check this setting using the same configuration utility you used to check the irq/address settings?

I've triple checked this with my DOS machine. I'll grab some screen captures later tonight, but I'm sure it's fine. It works fine with Win98 using the proper irq/address and it peaks about 8mbit/s doing a direct file copy to an old IDE drive, so I think it's fine. But I'll share a screenshot of that card config utility later tonight.

cocus, 2022-06-22:
> OK; good - then we know that the NIC is fine. Just to eliminate the final variables here: Are you connecting to the same network when testing ELKS as you did when testing with Windows? Same cable too?

Yes, same ethernet port on my laptop, same configuration on the Windows 10 side, set up as a internet share.

Mellvik, 2022-06-22:
> OK; thanks - I'll take another round with your listings and the driver tomorrow and see if I can come up with some ideas.
>
> —M

@toncho11,

Thinking a bit more about it, it seems you may have a hardware problem or a 'collission'. The interrupt line is 'ringing' for no obvious reason. The status codes indicate that everything is fine, there seems to be no good reason for the errors.
Is there a chance something else is also using this interrupt line? Can you test with a different irq by any chance?

-M

cocus, 2022-06-22:
> Is there a chance something else is also using this interrupt line? Can you test with a different irq by any chance?

Nothing else is using this line, since it's just one slot that I've added to my design. I could technically use another ISA IRQ but I need to change some code to allow the use of another INTx line of the CPU. I assume it might be an issue with the pull resistor I've added to the IRQ3 line, but that begs the question, why does it works with the other card? I'll experiment with this later

toncho11, 2022-06-22:
When I have time I will test the card on my Amstrad 1640. It was working fine there with both DOS and ELKS before.

Mellvik, 2022-06-23:
One more thing, @toncho11 :

I never answered your question about using the latest version of the driver: Absolutely. 

And for the testing, remove the comment slashes from the two printk statements at the beginning of the int routine:
```

	//printk("/"); <-----------HERE
	while (1) {	
		stat = ne2k_int_stat();
		//printk("%x|", stat);  <----------- and HERE
		if (!stat) break; // shortcut
```
That will tell us if the interrupts are real or not.

--M

cocus, 2022-06-23:
Hi @Mellvik , not sure if toncho11 has something to do with this, but :) 
Anyway, I've fired up the logic analyzer, and I can't see any "rogue" interrupt. I see that whenever there's a packet incoming, an IRQ fires, and the CPU reads the card (not sure which address, I just set this up to capture IRQ and "read or write to the ISA"). I've noticed the LED of the card blinks with each packet. I could grab that as well, but... not sure if it's useful.

Right now I'm going to load the latest with the uncommented printk()'s for the int debug.

cocus, 2022-06-23:
If you have the "logic 2" software from Saleae (it's free), I can share my logic captures. The one when I've connected thru telnet and run "meminfo" is quite interesting, because I can see the card holds the IRQ line for a lot of time and these correlate to the errors being spewed out on the terminal.

[image]

The first IRQ pulses were when I was typing "meminfo" on my PC. Later, about +0.4s later you can see that the IRQ was held for some time while the processor checked for it later. This might or might not be the serial console writing the actual error.
[image]
Here's a closeup.

And the console output:
```
8018x ELKS loader built at vie 17 jun 2022 20:39:12 -03 (LEDS NE2K BQ3285)
Now jumping into ELKS...
console_init: 8018X UART
eth: NE2K (8 bit) at 0x300, irq 12, MAC 00:50:4e:02:f7:82
ssd: 1977614336K disk
8018X machine, 512K base RAM.
ELKS kernel 0.6.0 (45888 text, 0 ftext, 5456 data, 38160 bss, 21918 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 446K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# ktcp -b -p ne2k 192.168.137.69 192.168.137.1 255.255.255.0
Usage: ktcp [-b] [-d] [-m MTU] [-p eth|slip|cslip] [-s baud] [-l device] [local_ip] [gateway] [netmask]
# ktcp -b -p eth 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmask 255.255.255.0
ktcp: ethernet 00.50.4e.02.f7.82 mtu 1500
# eth: RX-error, status 0x25
eth: RX-error, status 0x21
eth: RX-error, status 0x21
eth: RX-error, status 0x21
eth: RX-error, status 0x21
eth: RX-error, status 0x21
eth: RX-error, status 0x25
eth: RX-error, status 0x21
eth: RX-error, status 0x25
eth: RX-error, status 0x25

# telnetd
# accept from 192.168.137.1:21092
getpeername 192.168.137.1:21092
term_init ok, pid 8
eth: RX-error, status 0x01
eth: RX-error, status 0x05
eth: RX-error, status 0x05
eth: RX-error, status 0x05
eth: RX-error, status 0x05
eth: damaged packet (8001 64), clearing buffer
ktcp: eth_process error -1 (errno 5), discarding packet
eth: RX-error, status 0x01
eth: TX-error, status 0x01
eth: RX-error, status 0x05
eth: rcv oflow (0x53), keep 1
eth: rcv oflow (0x53), keep 1
eth: rcv oflow (0x51), keep 1
tcp retrans: seq 438+19 size 19 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 10 mem 392)
tcp retrans: seq 553+29 size 29 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 11 mem 431)
eth: rcv oflow (0x53), keep 1
tcp retrans: seq 534+19 size 19 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 11 mem 431)
tcp retrans: seq 532+2 size 2 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 11 mem 431)
tcp retrans: seq 505+27 size 27 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 11 mem 431)
tcp retrans: seq 486+19 size 19 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 11 mem 431)
tcp retrans: seq 457+29 size 29 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 11 mem 431)
eth: rcv oflow (0x53), keep 1
tcp retrans: seq 630+19 size 19 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 12 mem 480)
tcp retrans: seq 628+2 size 2 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 12 mem 480)
tcp retrans: seq 601+27 size 27 rcvwnd 63803 unack 438 rto 8 rtt 1 (RETRY 1 cnt 12 mem 480)
```
and more...

cocus, 2022-06-23:
Now, with the newer codebase, I'm getting this:
```
8018x ELKS loader built at Wed 01 Jun 2022 02:03:13 AM -03 (LEDS NE2K BQ3285)
Now jumping into ELKS...
console_init: 8018X UART
eth: ne2k (8 bit) at 0x300, irq 12: MAC 00:50:4e:02:f7:82
ssd: initialization error
8018X machine, syscaps 0x0, 512K base ram.
ELKS kernel 0.6.0 (46320 text, 0 ftext, 5520 data, 38160 bss, 21854 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 446K free
VFS: Mounted root 0x0600 (romfs filesystem).
Running /etc/rc.sys script
mount failed: No such device or address 
Welcome, welcome to city 17
Sat Jan 22 21:09:00 2000

ELKS 0.6.0

login: root
# ktcp -b -p eth 192.168.137.69 192.168.137.1 255.255.255.0
Usage: ktcp [-b] [-d] [-m MTU] [-p ne2k|wd8003|3c509|slip|cslip] [-s baud] [-l device] [local_ip] [gateway] [netmask]
# ktcp -b -p en2.168.137.69 192.168.137.1 255.255.255.0
Usage: ktcp [-b] [-d] [-m MTU] [-p ne2k|wd8003|3c509|slip|cslip] [-s baud] [-l device] [local_ip] [gateway] [netmask]
# ktcp -b -p en168.137.69 192.168.137.1 255.255.255.0
# ktcp -b -p ne2k 192.168.137.69 192.168.137.1 255.255.255.0
ktcp: ip 192.168.137.69, gateway 192.168.137.1, netmask 255.255.255.0
ktcp: /dev/ne2k mac 00.50.4e.02.f7.82 mtu 1500
# /41|0|/1|0|/1|0|/1|0|/7|eth: RX-error, status 0x25
0|/1|0|/7|eth: RX-error, status 0x21
0|/1|0|/1|0|
```
I've captured the 9 interrupts created by the card on the logic analyser as well, nothing out of the ordinary!
This is a snippet of the IRQ line and the "read or write" line for the ISA card when the last "RX-error" message was print:
[image]

cocus, 2022-06-24:
You'll like this screen capture of the serial console, the logic analyser and the remote telnet client. This time it got locked in a loop because the IRQ line was high constantly and the card never released it; although the CPU was reading or writing the ISA card by the looks of it.

[image]

ghaerr, 2022-06-24:
Hello @cocus,

Wow, that "logic 2" analyzer and your output is some cool stuff!!! I've wondered what kind of latencies there might be in the ELKS kernel handling interrupts and/or how long applications may have to wait while getting their system call executed.

Do you suppose there'd be a way we could place software interrupts (e.g. the 1-byte INT 3 instruction) in various places for even better visual display by "logic 2" between software/hardware interrupts?

Thanks for sharing, this is pretty neat stuff. Can't wait to think of ways we can possibly use this for even better tuning of our system :)

Thank you!

cocus, 2022-06-24:
@ghaerr , well, in this particular system, there are easier ways to do so. Some pins of the CPU can be used as GPIO, and writing a 0 or 1 to them is a matter of executing an `out` command. That usually gives the least amount of latency while turning a GPIO on before entering the IRQ handler, and turning it off while returning from it. I've used this on ARM MCUs (and although ARM has a bit-mapped scheme that 808x doesn't have) and this is a fantastic way of measuring time on any kind of target!

cocus, 2022-06-24:
> Also @cocus did you see my earlier note, when implementing the new NIC architecture that we now enable and disable IRQs on NIC open close. I added a disable_irq function in irq-8259.c, but I'd like to have you write an equivalent function for irq-8018x.c. Take a look at it if you have any time, it just needs to set the mask bit for the passed IRQ in the CPU, no low core vector changes. Thanks!

Yes, I still have the stub, I'm thinking what changes I need to add on the IRQ array that I have in order to support disabling IRQs. Do you think this is a problem for this case? I don't think so, but it's probably related to that IRQ line getting "stuck". Not sure :)

ghaerr, 2022-06-24:
> Do you think this is a problem for this case? I don't think so, but it's probably related to that IRQ line getting "stuck". Not sure :)

No - sorry for bringing that up on this thread. The IRQ line being stuck should have nothing to do with the kernel `disable_irq` routine, because the IRQ is never disabled during the period when ktcp/networking is accessing the driver.

>  I'm thinking what changes I need to add on the IRQ array that I have in order to support disabling IRQs.

It seems you're using the "config_word" value (always 0x06) to enable interrupts; would outputting a different value turn the interrupt off (always 0, perhaps?)

cocus, 2022-06-24:
> It seems you're using the "config_word" value (always 0x06) to enable interrupts; would outputting a different value turn the interrupt off (always 0, perhaps?)

Yes, in fact, setting bit 4 will mask the interrupt, so I can just logically-or that value (not always 0x06, that's the priority level, I've set the same one for all the INTx pins, but the UART has a lower one) with 0x8 to disable it. I'll push a PR for that in a short while.

Mellvik, 2022-06-24:
Great debugging work @cocus - and apologies to you and @toncho11 for the name mixup. Sloppy cut'n'paste if I ever saw one! Mea culpa.

And I agree with @ghaerr, your 'logic 2' stuff is fascinatingly efficient!!!

Back to the problem at hand, the debugging output was very enlightening. I didn't realize you actually had a working connection, then all the errors. Maybe I've been too fast reading your messages too :-)

I've spent a lot of time on your gif video, frame by frame - a very efficient setup indeed, including the ability to do just that, view frame-by-frame. And the IRS status printk is very enlightening indeed.

My preliminary conclusion is this:
you either have noise on your network triggering spurious interrupts, 
or the NIC is acting up - maybe because of the (forced) buffer strategy. 

Here's the thing: The situation you have at the end, where the TX errors loop and the IRQ line stays hight, should not happen. The IRQ line will be held high by the NIC for as long as there are pending bits in the ISR. In this setting the ISR is 0xa (TX OK and TX error, an impossible combination in the first place), both bits being reset by the call to ne2k_get_tx_stat() (which is being called twice [erroneously] for good measure).

So the first thing to do (unless you've tested this already) is to run the NIC with 4k buffers, not the forced 16k mode that is the default now. Set the 2nd bit in the io address in /bootopts and see if it makes a difference. If it does not I have some driver changes for you to test. For example, when you get RX errors, in more than 9 of 10 cases, the RX is fine (as per the status reg) and the error can be ignored.

As mentioned before, I think part of the challenge may be in differences between the chipsets on the NICs. They may be generally compatible, but there may be small differences with big effects… 
Judging from your setup, you know a thing or two about such things.

Let me know what you find out.

—M

cocus, 2022-06-26:
> So the first thing to do (unless you've tested this already) is to run the NIC with 4k buffers, not the forced 16k mode that is the default now. Set the 2nd bit in the io address in /bootopts and see if it makes a difference.

Well, I don't have bootopts in my machine, but I will modify it on the ports.h file; I'll set it to 0x302, that should do it right?

> As mentioned before, I think part of the challenge may be in differences between the chipsets on the NICs. They may be generally compatible, but there may be small differences with big effects… Judging from your setup, you know a thing or two about such things.

Yeah, I bet there are some differences. I'm sure not all ne2k cards are ne1k compatible, and this UMC might be one of these.

Mellvik, 2022-10-11:
Thank you.
Again I think your card is ok, there may be something with the line and its termination. It's not unusual for two pieces of hardware to act differently in the same setting if termination is off.

So you've wired irq 12 on the system side to irq3 on the bus, is that so?

Btw I'm not  exonerating  the driver just yet. There may be some quirks with initialization of this particular chip that I've missed. It would be useful to see (printk) the interrupt status register at the entry of the interrupt routine in the driver. I can send you a codeline to that effect in the morning.

-M

Mellvik, 2022-12-04:
Hi @cocus - what's the status on this issue?

Can we close it?

-M

cocus, 2022-12-05:
hi @Mellvik, I think the issue persisted when I last tried to use the system, I should test with the current code... however I didn't use the system in a long time, and I know that I have one card that "works", so... in the end, I can use that card and call it a day...
I think we can close this ticket!

closed 2022-12-05 by ghaerr