ghaerr/elks #1334: Some notes and please open discussion section...

jedie, 2022-06-19:
I tried ELKS v0.6.0 on my Atari PC1 with 8088 and 512KB RAM and seems to work fine ;)

Some notes: The boot output lists:

- "NEC V20" CPU, but it's a normal 8088 ;)
- The FDD are 720k not 360... (But i think the BIOS may be the problem ;)

On the "download" page here on github are different images, but no information what the files are. The List https://github.com/jbruchon/elks/wiki#binaries is incomplete and not the same as the files here: https://github.com/jbruchon/elks/releases

Is there a file list for 360, 720, 1.44 and 2.88 versions?!?

Think it would be also helpfull to open the github discussion, isn't it?

Bootscreen of my Atari PC1: (Note: The right side is cut off. This is due to my EGA converter and TFT combination.)

[image]

[image]

ghaerr, 2022-06-19:
Hello @jedie and thank for running ELKS and letting us know about it with screenshots and discussion :) 

> "NEC V20" CPU, but it's a normal 8088 ;)

We're aware that the CPU identification code (probably written 20+ years ago) is buggy and sometimes inaccurate. It doesn't know anything about later model (386+) CPUs either. I have been thinking of removing it entirely, except it sometimes seems useful. (Not very useful when it gives flat out bad info though in your case!).

> The FDD are 720k not 360... (But i think the BIOS may be the problem ;)

In your first screenshot, the first two displays of /dev/fd[0-1] info are received directly from the BIOS, whereas our probe routine is accurately identifying the drives as 720K. This is because the probe routine at open time attempted a read on track 79 which succeeded, which updated the kernel tables.

> Is there a file list for 360, 720, 1.44 and 2.88 versions?!?

The actual list is in [elkscmd/Applications](https://github.com/jbruchon/elks/blob/master/elkscmd/Applications), which is a bit cryptic but shows what you need. It would be possible to write some scripts to generate a more human-readable version of the same.

>  The List https://github.com/jbruchon/elks/wiki#binaries is incomplete and not the same

Thanks, looks like that needs to be updated with the 1200k and PC-98 floppies. Noted.

> Think it would be also helpfull to open the github discussion, isn't it?

How does that work, I am not familiar with it? So far, we've mostly had discussion within open issues, but welcome more discussion elsewhere.

Thank you!

jedie, 2022-06-20:
> How does that work, I am not familiar with it? So far, we've mostly had discussion within open issues, but welcome more discussion elsewhere.

Just go into project settings/General/Features and activate "Discussions"...

[image]

...

[image]

;)

jedie, 2022-06-20:
> The actual list is in [elkscmd/Applications](https://github.com/jbruchon/elks/blob/master/elkscmd/Applications), which is a bit cryptic but shows what you need. It would be possible to write some [...]

I insert a Link in the Wiki with: https://github.com/jbruchon/elks/wiki/Home/_compare/0e9e895cd51674e070570a9a4aa7eb9c05e898b2

ghaerr, 2022-06-20:
Hello @jbruchon,

Can you turn on Discussions, or give me permission to do so, as suggested by https://github.com/jbruchon/elks/issues/1334#issuecomment-1160081114? I don't have authority and there is not Settings tab for me.

Thank you!

jbruchon, 2022-06-20:
@ghaerr I did it. Let me know if you need anything else!

closed 2022-08-14 by ghaerr