ghaerr/elks #1366: New Cool Stuff for ELKS

ghaerr, 2022-07-12:
I'm opening this issue to talk about various projects I'm working on, most of which are running, but not fully integrated into the ELKS build yet. I'm interested to know how interested people might be in each. Here's the first three projects:
- ELKS remote console display
- D-Flat Text UI windowing toolkit
- Debug stack traces for ELKS executables

I'll write up each of these separately.

ghaerr, 2022-07-12:
ELKS Remote Console

This little program allows one to login through a serial or telnet connection to ELKS (using a macOS or Linux terminal program, for instance), and display what is currently displayed on the ELKS console. This could be useful to see remote kernel messages, or just see exactly what is displayed in full color, using the ROM BIOS 437 character set.

For example, here's a login from macOS Terminal over emulated serial on QEMU, just after boot:
[image]

The display is updated every second, and the program is terminated with ^C.

The `cons` program reads the PC text RAM contents directly and converts the characters into UTF-8 encoded Unicode values, along with interpreting the PC character attributes (color, reverse video, etc) and translating those into ANSI escape sequences.

The current implementation just displays the console screen, it doesn't try to insert keystrokes, etc.

ghaerr, 2022-07-12:
D-Flat Text User Interface Library

D-Flat is an old (1995 era) text user interface that allowed overlapping window-based text user interfaces to be written on DOS. I've ported it to UNIX/Linux, and now to ELKS. It is very well written, but unfortunately kind of large, and requires medium model (large code) for compilation. The underlying API is message-based, and very similar to Win32, with SendMessage and window procedures. 

Here's a screenshot of the D-Flat `memopad` program, which allows for editing files:
[image]

I'm primarily working on it as a possible TUI for the Cosmopolitan project, but thought it would be interesting to get it running on ELKS.

Now to show off the `cons` program more, here's the cons program running on serial macOS Terminal, while `memopad` happens to be running on the console:
[image]

ghaerr, 2022-07-12:
Debug library for ELKS applications

While trying to get larger ELKS programs working, I've found it quite hard to understand what is going on when the program crashes, especially with no debugger or stack tracer. I've developed a set of routines that will allow for a running ELKS program (currently applications only, but could be fitted into the kernel itself for device drivers etc) to produce a stack backtrace, along with correct lookup of the actual C symbol names.

I've also implemented Cosmopolitan-style C function logging capability, which allows for display of every C function executed, to make it possible to visualize function call flow throughout program execution. I plan on adding nanosecond-precision timing display (on 386+ CPUs only, using the RDTSC instruction) to show function latencies as well. The idea would be to add an automatic  "--ftrace" program argument that would turn on this display when an application is compiled for debug.

I took ELKS `basic` and added the stack trace capability on every function call, here are the first few functions called, along with their arguments. The display also shows the current and max stack used by the program:
```
ELKS 0.6.0

login: root
# basic
AX=0000 BX=7460 CX=0001 DX=7464 SI=7460 DI=3448 BP=7446 SP=7422
CS:IP=5540:36f0 DS=6200 ES=6200 SS:SP=6200:7422

Level Addr    BP   DI   SI   Ret  Arg  Arg2 Arg3 Arg4
~~~~~ ~~~~    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   0: 7418 => 7446 3448 7460 36f7 dead 0000 0000 0000 0000 0000 0000 0000 0000 0000 print_stack (364e,83)
   1: 7446 => 744e 2c60 7460 3448 745c 2c60 2cb0 0019 0001 7460 7464 0000 0001 7472 __cyg_profile_func_enter+35 (36f7,83)
   2: 744e => 745c 2c60 2cb0 0019 0001 7460 7464 0000 0001 7472 0000 7478 7482 748d main+b (3448,83)
   3: 745c => 0000 0001 7472 0000 7478 7482 748d 7497 74a1 74b2 0000 6162 6973 0063 _start+19 (0019,81)
   4: 0000 => 0000 0000 5825 0900 2800 2900 2b00 2d00 2a00 2f00 3d00 3e00 3c00 3c00 _start+1 (0001,81)
>main, from _start+19, stack 0/0
AX=0019 BX=0000 CX=5401 DX=0000 SI=7460 DI=2ce5 BP=7414 SP=73f0
CS:IP=5540:36f0 DS=6200 ES=6200 SS:SP=6200:73f0

Level Addr    BP   DI   SI   Ret  Arg  Arg2 Arg3 Arg4
~~~~~ ~~~~    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   0: 73e6 => 7414 2ce5 7460 36f7 dead 00a1 0019 73fe 73fe 354e 7408 1742 1c9c 5014 print_stack (364e,83)
   1: 7414 => 7442 0000 7460 2ce5 0000 3448 0000 0000 0000 735f 6174 7472 312b 0000 __cyg_profile_func_enter+35 (36f7,83)
   2: 7442 => 744e 0000 7460 345b 0000 46b8 745c 2c60 2cb0 0019 0001 7460 7464 0000 tty_isig+b (2ce5,83)
   3: 744e => 745c 2c60 2cb0 0019 0001 7460 7464 0000 0001 7472 0000 7478 7482 748d main+1e (345b,83)
   4: 745c => 0000 0001 7472 0000 7478 7482 748d 7497 74a1 74b2 0000 6162 6973 0063 _start+19 (0019,81)
   5: 0000 => 0000 0000 5825 0900 2800 2900 2b00 2d00 2a00 2f00 3d00 3e00 3c00 3c00 _start+1 (0001,81)
|>tty_isig, from main+1e, stack 25/25
```
I am also thinking of a use where a function backtrace would be displayed whenever ^C (SIGINT) is typed, to show what the program is doing at the time of interrupt. This would be very useful for debugging hung programs.

The instrumentation facilities used to implement function call tracing require a patch to the ELKS compiler toolchain, and the stack trace capabilities require some special link handling which adds debug symbol information into the current executable, both of which are rather complicated at the moment.

In addition, I have some library code that can interpret memory bytes and display as 8086 disassembly, running directly on ELKS. This could be used within an application, or added to the `hd` or a new `disasm` or monitor/debug program, for better visibility of very low-level stuff while running on ELKS. The `hd` program is going to be enhanced to display codepage 437 (ROM BIOS) characters on a terminal display, to better see binary values as part of a hex dump.

tyama501, 2022-07-12:
Hello @ghaerr ,

Thank you for very cool stuffs!
I started to port nano-X to pc-98 but it seems harder than I thought so it would be nice to have the debug library.

D-Flat is also interesting to see applications like DOS edit on ELKS but I'm not sure it will work for PC-98 too.

ghaerr, 2022-07-12:
Hello @tyama501,

> I started to port nano-X to pc-98 but it seems harder than I thought

It may be easiest to just copy nano-X/drivers/scr_bios.c to scr_pc98.c, and plan on having all graphics draw code in that single file. Rename all the functions from VGA_ to pc98_, and then start with just writing the pc98_open() and pc98_close() routines which should switch the screen to and from graphics mode, using the same code from BASIC. 

You should then be able to write the very basic pc98_drawpixel, pc98_drawhline, pc98_drawvline  and pc98_fillrect functions, which do not depend on an internal graphics format. (HAVEBLIT should be set to 0).

After that, in order to support BLIT, we would need to make a decision as to how nano-X should store the graphics data internally, which is set with the psd->app and other statements in pc98_open().

If you'd like to share what you have, I'm happy to help you get the screen driver working. The mouse driver, but default uses the serial port, which isn't yet implemented for PC-98, so you might want to compile in drivers/mou_null.c.

> D-Flat is also interesting to see applications like DOS edit on ELKS but I'm not sure it will work for PC-98 too.

What is the character set for byte values 128 - 255 in PC-98 character ROM? These won't be codepage 437, so that could require low-level changes to D-Flat for running on console. However, I am thinking having the library code automatically switch to using ANSI ESC sequences when running on telnet or serial, and having the CP 437 codes converted to unicode for that case (the same was `cons` works, basically).

Thank you!

Mellvik, 2022-07-12:
Thank you @ghaerr -
> Disasm: Sounds very useful, but I didn't understand how it may be used.
> 
> I think the most useful case would be some kind of debug/monitor that would trap to a prompt (interrupting the running program from a SIGINT), then producing a backtrace and allow the user to dump register or specify a seg:off memory address to start disassembly from. As a library routine, it would also be useful to write programs to disassemble various files at times. Another use could be instruction tracing, by allowing a single instruction to execute on real hardware, followed by an INT 3 back to the monitor. This would work because the disassembler has the ability to compute how long a given instruction sequence is.
> 
Wow, that's ambitious indeed!! I'm rather looking forward to it - exciting! 
> I'm not quite sure where to put all this - I suppose just adding the the C library would be OK, except these are all non-standard routines. So perhaps a separate library... not sure yet.
> 
My 2cents - separate library.
> I'm also working on being able to compile and link programs outside of the ELKS source tree - this is too hard currently, as there's no easy way to get all the compiler options and C library. Once this is done it will be lots easier to contribute programs to ELKS, without having to become intimately familiar with the kernel and application source trees.
> 
Agreed, and again very useful indeed!

ELKS rocks

-M

ghaerr, 2022-07-13:
Hello @tyama501,

> I have copied vgaplan4.c to vgaplan4_pc98.c and I have been modifiying ega_drawpixel to draw directly to 0xA800, 0xB000, 0xB800, 0xE000 VRAM after changing to the graphic mode

I see. I suppose it depends on how close the PC-98 graphics hardware is to EGA - EGA is terrible, and requires multiple mask and plane register to be set for each pixel. If that's the case for PC-98, perhaps staying with your present approach is better. Another thought would be to only support mono (2-color) mode initially, is that possible, then possibly not having to write to each memory plane for each pixel.

After getting running, you will want to "define NDEBUG" to remove the assert() code, as it slows things down considerably.

> For PC-98 there is no serial mouse, and the mouse is attached to 8255 I/O port.
So, maybe we need to write the driver with polling or interrupt after the screen driver worked.

When you use mou_null, this should happen automatically. Afterwards, a PC-98 mouse driver can be written.

In general, nano-X busy-loops by polling, so this shouldn't be a problem unless the polling itself is quite slow.

> It seems there are single byte katakana-characters and some others.

I can see it's definitely not codepage 437, but does include the box line draw characters. There doesn't seem to be the 50% gray shaded character that is used for the background of D-Flat windows, but that isn't a big problem, a space can be used. 

Should we get more interested in actually running D-Flat on PC-98, we can develop a translation table for the different code page, both for D-Flat use, as well as for the `cons` unicode translation. 

Thank you!

toncho11, 2022-07-14:
Hi @ghaerr,

I am for D-Flat project - it seems it can produce a nice DOS editor and some other applications as well. We can have a developer section one day: "Programming for ELSK" using:
- pure C code
- using X-nano
- using DFlat TUI
- Basic

ghaerr, 2022-07-18:
D-Flat and its text-mode user interface library demo is now running on ELKS connecting from serial or telnet terminal sessions. Here's a video showing it in operation. In this particular case, all I/O is based on ANSI terminal escape sequences, so the TUI display is available to many remote systems, using any modern terminal emulator, such as macOS Terminal or Linux xterm-256.

It also runs on the console, but this requires more setup as a mouse driver that spits out ANSI sequences needs to be installed, unless the program can be entirely operated from the keyboard.

The video is too big for Github upload, [here's a link to my dropbox for viewing](https://www.dropbox.com/s/qlssrndb007vjwa/D-Flat%20ELKS%20Demo.mov?dl=0).

toncho11, 2022-07-18:
Nice job @ghaerr !!!
So it can run natively (without any remote sessions) on ELKS, but one needs to use the keyboard shortcuts.

> D-Flat and its text-mode user interface library demo is now running on ELKS connecting from serial or telnet terminal sessions. Here's a video showing it in operation. In this particular case, all [...]

Mellvik, 2022-07-18:
Wow @ghaerr - this is impressive.

Apropos mouse: Can the interface be operated w/o a pointing device?

-M

ghaerr, 2022-07-18:
> Can the interface be operated w/o a pointing device?

Yes, I'm sure it could, but not certain as to what has been coded in D-Flat. D-Flat is a somewhat large text user interface (TUI) library, by ELKS' standards. The `memopad` demo, which includes all the TUI controls (scrollbars, listboxes, popups, comboboxes, edit boxes, etc) is 110k - huge. D-Flat is IBM SAA/CUA compliant as of their [1987 standard](https://en.wikipedia.org/wiki/IBM_Common_User_Access).

However, the real purpose of the work is mostly R&D, and under the hood portions of the following are already implemented, or being considered:
- Working medium model ELKS applications (a toolchain bug was [recently fixed](https://github.com/tkchia/gcc-ia16/issues/117) getting memopad to load properly).
- Writing TUI applications that compile and work almost identically between macOS, Linux and ELKS.
- Unicode support, which allow ELKS to run applications that can display unicode (using UTF-8) on serial or telnet connections. In particular, the ability to display BIOS character ROM codepage 437 characters external to the console is a big win.
- Possible solutions to long-standing character display problems such as #1175 because the PC ROM character set isn't the same across machines internationally.
- Enhancement of incoming and outgoing ANSI escape sequence support for ELKS console and terminal sessions, including accepting a wider ranger sequences on direct console for UNIX application compatibility, as well as ANSI mouse and window sizing support being put in library routines.
- A possible ability for an application to know whether its running on direct console or serial/telnet terminal, and to automatically perform character conversions via Unicode and/or ANSI sequences for visual compatibility. (For instance, the `cons` console display program, and planned `hd` display enhancements to see the full CP437 character set).
- Possible support for arbitrary unicode on non-console applications. This is important, as the rest of the world is now running on unicode, and ANSI sequences allow unicode display on all modern terminal emulators.
- Development of multiple layers of TUI support, from automatic mouse/keyboard support without termcap or ncurses (using ANSI/xterm standard) for building portable TUI applications, the ability to write characters/attributes to the screen directly for speed, all the way to heavier-weight libraries like D-Flat.
- Test programs that allow for easier unit-testing of various library components.
- Testing of the kernel and C library implementation by using larger pieces of software (more incompatibilities have been found).

All this said, the fact that ELKS is basically running almost all of D-Flat identically to its UNIX counterpart is a testament to how far ELKS has come in the last few years - its pretty amazing our 16-bit system can keep up and run more modern UNIX 32- and 64-bit programs, thanks to everyone's work :)

toncho11, 2022-07-19:
@ghaerr Are you planning on submitting D-Flat as part of the bin folder. Maybe only for the HDD images?

ghaerr, 2022-07-19:
Hello @toncho11,

My plan would be to consider adding the D-Flat source code as a text user interface library, along with a number of other helpful library routines, somewhere in the ELKS source tree. The library would be automatically built so that others could build their own TUI applications. The D-Flat library includes only a single demo `memopad` that displays all of its capabilities which would also be copied to the /bin directory for any images 2.88M and larger.

The real problem with most TUI and GUI libraries is that it isn't trivial to build an application using them. Frankly, it isn't real clear to me what anyone might build with a D-Flat TUI. (Given that the `kilo` and `nano` editors don't work well, we renamed the MINIX screen `mined` to `edit`, and it seems to work quite well, so we don't need a screen editor in D-Flat). Even though D-Flat is well thought-out and well-written, it is large by ELKS standards, and programs produced using the library won't fit well on floppy disks with the rest of the standard distribution.

Thus, D-Flat itself is mostly a research project, which (described above) is helping to illuminate some things that might want to be considered next for development. That said, it's opening the doors for more discussion on what we should next add to ELKS. Thank you for your comments!

ATroubleshooter, 2022-07-20:
> 
> Thus, D-Flat itself is mostly a research project, which (described above) is helping to illuminate some things that might want to be considered next for development. That said, it's opening the doors for more discussion on what we should next add to ELKS. Thank you for your comments!

Considering D-Flat, in regards to as what should be added next - probably, a simplistic lightweight twin-panel file manager, based on D-Flat?

ghaerr, 2022-07-21:
Looking around for other "small" text-based file managers, I finally found two, both based on ncurses: `nnn` (https://github.com/jarun/nnn) which is highly supported and does lots of stuff. It purports to running well on embedded systems. Here's a screenshot:
[image]

However, in reading a comparison article, `nnn` apparently uses far less memory than other Linux-based file managers, "only" 3.5MB of RAM!!! (compared to 9.8MB for Midnight Commander, and 51MB for the largest):
[image] The `nnn` executable on macOS is 131K with no options enabled.

I guess RAM is cheap.

Looking further, I find `noice` (https://github.com/RichardHyde/noice), which it turns out `nnn` is forked from. It compiled quickly on macOS, and is a 29K executable. Its development stopped in 2015, but it looks very similar, just doesn't use any color display:
[image]

The nnn source is 8800 lines, and noice is 1000. Looking at noice, it seems it could be ported to ELKS quickly, with an initial version using just the arrow keys (and/or mouse on a remote terminal), using the low-level D-Flat framework I just developed. Later, a version could be possibly inserted into a D-Flat frame to look cooler. I haven't looked at the memory usage yet for noice. I may take a pass at porting noice to ELKS if people are interested.

ghaerr, 2022-07-22:
I looked further into `noice` and found the original project (https://git.2f30.org/noice), which was last updated in 2019. It is a quite small and nifty, portable file manager.

I ported this to ELKS as our new `fm` file manager, and wrote an ncurses library emulation layer for it, which allowed `fm` to run directly on the console or from a serial/telnet terminal. The file manager is only 13K bytes, so it should run well on floppies.

Here's a screen recording of it in operation. I plan on porting the lower layer to be able to use the mouse or function keys on the console or serial/telnet terminal, and then ultimately should be pretty straightforward to put into a D-Flat window. Pretty cool!

https://user-images.githubusercontent.com/11985637/180363966-87eef0b8-ae6e-4f26-a284-e87f93935bce.mov

Thanks for the idea @ATroubleshooter :)

Mellvik, 2022-07-22:
Indeed @ghaerr - this IS cool!

> 22. jul. 2022 kl. 06:49 skrev Gregory Haerr ***@***.***>:
> 
> Here's a screen recording of it in operation. I plan on porting the lower layer to be able to use the mouse or function keys on the console or serial/telnet terminal, and then ultimately should be pretty straightforward to put into a D-Flat window. Pretty cool!
> 
> 

-M

ATroubleshooter, 2022-07-22:
Great job @ghaerr, keep it up!

ghaerr, 2022-07-26:
Update on new File Manager for ELKS and UNIX:

Significant progress has been made, the the project now compiles and runs identically on ELKS console, ELKS terminals, as well as UNIX (tested currently only on macOS Terminal). It seems to work pretty well, and for now have created its own [project](https://github.com/ghaerr/fm), before integrating directly into ELKS. 

Here's a screenshot running on macOS (which pretty much looks identical to the ELKS terminal version):
[image]

The file manager has lots of features for its small size, but here are the basic keystrokes:
```
RightArrow/Enter - Enter directory or display file
LeftArrow/Backspace - Back up one directory
Up/DownArrow or mouse scroll wheel - move cursor
a-z - goto file matching letter pressed
Z or ^C - quit
D - directories displayed first
T - sort by time
S - sort by size
C - change directory
E - edit file
. - display hidden files
~ - home directory
? - help
```
The program automatically decompresses .Z files for display, and will run `man` on man page files in compressed or normal format. `more` is used to display files.

After cloning the source tree, type `make` for the Linux version, or `make -f Makefile.elks` for the ELKS version, then type `./fm`.

I've worked hard to make the program as small as possible, which is pretty small since ncurses and regex libraries were removed. `fm` requires and ANSI (or xterm-256) terminal to run.

The ELKS version is 20k and UNIX 40k. Let me know what you think. At some later point I may wrap the output in D-Flat to look more like Midnight Commander.

Enjoy!

toncho11, 2022-07-26:
Great job @ghaerr !

toncho11, 2022-07-26:
I see edit is possible.
Here are suggestions for future versions:
- configure editor (which is the default now?) 
- copy files with progress bar
- delete and move both files and directories

Looking back I like Norton Commander, but PC Tools is the one that impressed me the most back then. PCtools is not double screen. A new window pops up and you can select directory tree or drive (I think).

ghaerr, 2022-07-26:
> configure editor (which is the default now?)

`vi` is the default, but any editor can be specified using the EDITOR= environment variable.

> copy files with progress bar
> delete and move both files and directories

Good ideas, but `fm` is currently just single-paned. I'll wait and see whether people actually use it before spending too much time on it. With the under-the-hood work on the keyboard and mouse input for D-flat, it was really good timing to work on another program that heavily used the mouse and arrow keys. D-flat doesn't yet support the mouse wheel, and I'm going to jump back into that for bit and try to get a version others can play with on ELKS or UNIX.

ghaerr, 2022-07-28:
Update on D-Flat for ELKS and UNIX:

A [separate project](https://github.com/ghaerr/dflat) has been created for D-Flat, where it can be compiled for macOS, Linux or ELKS. Things have been progressing nicely, and it now automatically sizes to the open Terminal on startup, just like the `fm` File Manager. Both projects share the same new "unikey" ANSI keyboard/mouse library, which also supports unicode character conversions (currently only for code page CP437). The screen updates are FAST.

Here's D-Flat's `memopad` running on macOS. The display looks identical when run from ELKS serial or telnet:
[image]

The new unikey library supports ANSI mouse sequences from macOS and Linux terminal programs emulating xterm-256, and I plan on adding the mouse sequence support to ELKS running QEMU over an emulated serial port. That would allow for operation of QEMU console programs using the mouse from your host computer or trackpad to operate it. On real hardware, the serial mouse is already supported, but not yet built into the unikey library, which would allow for a new class of programs to be developed for ELKS - text based user interfaces which could be sped up greatly just by using the mouse wheel for fast scrolling, for instance.

If you have time, take a look at the project and compile it. Because both D-Flat and File Manager are being developed for both UNIX and ELKS, I'm thinking that adding them as a Git submodule (i.e. a repo within the ELKS repo) in the future, which would allow for automatic compilation and inclusion on the ELKS disk images.

tyama501, 2022-07-28:
Hello @ghaerr ,

Nice! Yes I agee adding other projects as submodules to expand ELKS world.
It would be nice if more developers contribute to ELKS through those projects.

Thank you.

ghaerr, 2022-08-02:
With the console fixes in PR #1388, ELKS can now simulate the Matrix! :)

https://user-images.githubusercontent.com/11985637/182286487-20db97ca-94bb-4e60-b168-7c109295c9ea.mov

toncho11, 2022-08-02:
Impressive @ghaerr

ATroubleshooter, 2022-10-07:
> Update on new File Manager for ELKS and UNIX:
>
> Significant progress has been made, the the project now compiles and runs identically on ELKS console, ELKS terminals, as well as UNIX (tested [...]

I tried it on ELKS and it worked! And as for the MC-like appearance - I would really like to have it on ELKS. D-Flat could indeed make the difference. For example, this is how DOS Navigator looks, it's using "Turbo Vision" style interface too:
https://youtu.be/Fmauh_ERaag?t=325

It's got its own windowed viewer, editor with clipboard buffer support etc...

Mellvik, 2022-10-11:
Thank you @ghaerr - 
again - this is cool. I just had to check it out right away - and sure enough, works almost out of the box (the Makefile.elks needs to be edited to set TOPDIR).

How do I activate the menu/instruction line on the bottom?
Like kilo it does a repaint for every keystroke, which is bearable at 57600bps but still very visible. IIRC that's a hard one to fix, right?
(BTW I haven't merged recent ELKS updates, that may be part of it).

—M

ghaerr, 2022-10-11:
Hello @ATroubleshooter,

> And as for the MC-like appearance - I would really like to have it on ELKS. D-Flat could indeed make the difference.

Thanks for trying out `fm` :) I agree, I plan on enhancing it to look like MC, but wanted to see whether users find it useful in the first place. It's on my list to improve, and the D-Flat library could be a good choice for that, although it has the problem of being kind of big.

> How do I activate the menu/instruction line on the bottom?

@Mellvik, use `?` for that. I plan on adding a full screen of help, but haven't to to that yet either.

> Like kilo it does a repaint for every keystroke, which is bearable at 57600bps but still very visible. IIRC that's a hard one to fix, right?

It doesn't repaint on every keystroke, but does repaint the entire screen during each function. I'm interested in your continued comments on usability - I was thinking this could possibly be upgraded to be the front-end of a way to upgrade various portions of ELKS or to at least run shell scripts quickly.

Thank you!

ghaerr, 2023-02-27:
I've been working for a while on a visual emulator/debugger for ELKS binaries and the kernel itself. It's finally here and ready to play with. It's called [blink16](https://github.com/ghaerr/blink16) and is based off the incredible [Blink Project](https://github.com/jart/blink). 

It is still under construction but allows for 8086-only ELKS and DOS binaries to be loaded and the programs to be emulated on your desktop, translating system calls and displaying console I/O during execution.

A separate mode of running `blink16` allows 8086 PC emulation, where an ELKS image, with symbol table, is booted just as on a PC, and the entire internal operation of the ELKS kernel can be inspected and traced. Download the source and run `make elks` to demo a prebuilt ELKS image booting. (More instructions on the [site](https://github.com/ghaerr/blink16)).

Here's a screenshot of ELKS running 'banner ELKS', showing `banner` executing the `write` system call:

[image]

toncho11, 2023-02-27:
Is there video emulation? Can I run nxtetris on a contemporary x64 Linux machine by just pointing to the executable: `./blink16 nxtetris ELKS`.

tyama501, 2023-02-27:
Hello @ghaerr ,

Nice!
Can I use it to debug elksmoria if I put files in the ibm ELKS image?

ghaerr, 2023-02-27:
Hello @toncho11,

> Is there video emulation? Can I run nxtetris on a contemporary x64 Linux machine by just pointing to the executable

There is currently MDA and CGA emulation, but only for their text modes. It doesn't (yet) emulate EGA/VGA, that'd be probably a lot of work. We've got a lot of work to do adding in all the ELKS system calls as well (e.g. read/write/close are in, many others still missing).

> ./blink16 nxtetris ELKS.

The "ELKS:" at the end of my sample `banner ELKS` is for arguments to banner. So you'd only have to give the basic executable name, not followed by ELKS. `blink16` figures its a DOS executable if .exe extension if found, a boot block if .bin or .img is seen, otherwise ELKS executable.

I haven't tried running many programs yet, but it should run them until a missing system call is attempted. Any graphics output would just be ignored at this point, so you won't see anything Nano-X programs.

We've got the new ELKS symbol table format all working - all that's left is to automatically build all programs with a symbol table, which is coming shortly, but will (temporarily at least) increase the floppy space usage. The big benefit is that anyone can then execute a program and see its execution internally using the source code symbols. This will be a great way to get more comfortable with the internals of how any ELKS executable operates.

Thank you!

ghaerr, 2023-02-27:
Hello @tyama501,

> Can I use it to debug elksmoria if I put files in the ibm ELKS image?

Haha, I knew you were going to ask that!! The answer is: almost. There isn't quite support working for medium model programs as far as symbol table display, although it ought to be able to emulate the program. However, emulating without a symbol table for mnemonic display makes it hard to understand where you are in the program. There is also the problem that there are many system calls missing. 

In short - it's coming.

Thank you!

tyama501, 2023-02-27:
> In short - it's coming.

Great, thank you!

tyama501, 2023-02-28:
I found this tweet about blink16 and ELKS:)

https://twitter.com/JustineTunney/status/1629671904696205312

tyama501, 2023-03-01:
I could build with my FreeBSD Clang with this.

cc -DBLINK16=1 -I.. -Os -o $@ $^ -lz -DNOJIT=1 -lm

[image]

ghaerr, 2023-03-01:
> I could build with my FreeBSD Clang with this.
> cc -DBLINK16=1 -I.. -Os -o @^ -lz -DNOJIT=1 -lm

Wow @tyama501, nice! I hadn't realized that -DNOJIT=1 is required, I'll add that to blink16/Makefile.

Also I noticed that your terminal is running in "dark mode", I hadn't tested that as macOS Terminal runs black on white background. That looks nice. Can you expand the right side of the terminal even more? That will allow for the instruction bytes to be displayed left of the instructions themselves. I am trying to determine a way to perhaps make the memory pane views a little smaller.

Now that you're running ELKS executables, feel free to open an issue (on ghaerr/blink16) for any problems. I haven't yet tried running any medium model programs, and there are definitely issues with the symbol display, which are being worked on now. You will also find that we need quite a few more system calls to be emulated (see syscall-elks.c) in order to be useful.

We can also make changes for PC-98 specific issues if required. I think this tool will be very helpful to better understand what ELKS applications and the kernel are doing.

Thank you!

tyama501, 2023-03-01:
Hello @ghaerr ,

Well, actually I don't know disabling jit function is ok or not.
In my environment, one of the atomic_ function in the HAVEJIT got error claiming address is const, so I just disabled it. 

Thank you.

ghaerr, 2023-03-01:
> Well, actually I don't know disabling jit function is ok or not.

Its ok to disable, as blink16 replaced the blink VM with a much smaller 8086-only instruction interpreter that does not use JIT. Using -DNOJIT=1 is useful for the other files and includes that might possibly be included into the blink16 project. I have added -DNOJIT=1 and -lm to blink16/Makefile and will push in the next commit.

tyama501, 2023-03-02:
Hello @ghaerr ,

[image]

I deleted some files from fd1440.img to get space and put the elksmoria files.
I am trying to run it but it seems that keys are not working or too slow to enter while running with "c" command.

Is there way to speed up?

Thank you.

ghaerr, 2023-03-02:
> I am trying to run it but it seems that keys are not working or too slow to enter while running with "c" command.
> Is there way to speed up?

For the above screenshot emulating the ELKS kernel, you can use the capital C) command, which executes instructions but does not show the instruction and memory tracing. This command still has some issues when a program is trying to read the console. For that, you can use the capital D)isplay command, which just shows console output. A ^C brings you back from that screen to the main emulation screen. I just added stack backtrace frames support, so pull that down, the output is helpful. Shortly, I will push a commit with P)rofiling support, which is cool because it shows the functions which are taking up the most time. You can press ? to get a help screen.

> I deleted some files from fd1440.img to get space and put the elksmoria files.

I'm not sure whether elksmoria will run or not, I have not tested any medium model binaries. For sure the symbol lookup won't be working yet. We also have the problem of not supporting many ELKS kernel system calls yet in application emulation mode.

(Note that blink16 runs in one of two "modes" - an "application emulation mode", that of emulating an ELKS or DOS application, and separately, a "system" mode, that of emulating an IBM PC bare machine, which is used to boot .bin sector games or .img files like the ELKS kernel itself. When blink16 is running in "system" mode (.img or .bin files), it only shows the symbols for the kernel itself, not for any running applications. Moreover, it might be hard to debug the application that way, since the timer interrupt is causing ELKS to switch between applications. The "application mode" is active when a .exe file is specified for DOS, or for ELKS when neither .bin or .img are specified.)

Another way to speed up display during c)ontinue mode is ^T (turbo). Typing that a few times speeds the display up. Blink16 also supports S)tep and N)ext, which execute a single instruction, with N) skipping over call instructions.

toncho11, 2023-03-03:
blink16 works only on contemporary computers and it is 32/64 bit compiled, right? It is compiled with the standard gcc (not ELKS gcc16). For a second I had this idea where blink runs inside ELKS ... and it allows for the execution of DOS executables. @ghaerr this all came as I was reading that it "translates system calls". The idea is not completely crazy ... :). If it is going to be insanely slow then it could be used only on 386 and 486

tyama501, 2023-03-03:
Thank you @ghaerr ,

I see. Here is the screen after "D" and ./elksmoria
[image]

tyama501, 2023-03-03:
cat morifile/newselks works.
The previous image I posted is trying to show this file.

[image]

ghaerr, 2023-03-04:
Hello @toncho11,

> blink16 works only on contemporary computers and it is 32/64 bit compiled, right? It is compiled with the standard gcc (not ELKS gcc16). 

Yes - blink16 is a 32/64 bit application that uses a large amount of memory to emulate the 8086 instruction set, as well as save instruction history for "rewinding" execution (by using the scroll wheel). This takes a lot of RAM and won't ever be able to run on ELKS, nor any PC with less than 32MB of RAM.

>  this all came as I was reading that it "translates system calls". The idea is not completely crazy ... :). 

The ELKS ("guest") system calls have to be translated to the operating system blink16 is running on ("host").

> If it is going to be insanely slow then it could be used only on 386 and 486. 

Not sure what you're referring to here, but if its blink16 execution speed, the purpose is for debugging, not just to emulate an 8086 like QEMU does. The capital C) and D) commands allow fast execution without instruction display.

> If it saves you the job of having DOS installed and restarting, it's still worth it.

The idea is to run some DOS executables, also by translating the DOS system calls to Linux/macOS/FreeBSD, etc, but blink16 doesn't try to emulate a full DOS environment, so I'm sure many programs may not fully work.

The intent is mostly to visually see a program's execution. Blink16 also supports P)rofiling, which shows which procedures are bottlenecks in execution time.

Thank you!

ghaerr, 2023-03-04:
Hello @tyama501,

> Here is the screen after "D" and ./elksmori

I think you may be getting confused here on how to run blink16, please re-read my post above. That is, if you run:
```
./blink16 fd1440.img
```
This will execute blink16 in "system emulation mode", where the ELKS kernel (in this case) is booted from the .img file. Trying to run `./elksmori` from *within* system emulation mode won't work well - as blink16 is primarily emulating an 8086 system, rather than trying to run a particular ELKS program. In order to directly debug elksmori, you would want to type something like:
```
./blink16 elksmori
```
This would load `elksmori` directly into blink16 memory, then attempt running it and translate the ELKS application system calls to Linux/macOS/FreeBSD, etc. However, this will not yet work, as the blink16 loader won't load medium-model ELKS applications (yet). So you should test with another program. I have included `banner` and `echo`, built with -maout-symtab, which includes the symbol table, which really helps display.

The reason why, in the first case, ELKS won't run `./elksmori` is that the `fd1440.img` image has on it a very specially-prepared ELKS kernel which is very small, and is built without medium-model support. This was done because I've just barely got ELKS running in blink16. Since the ELKS kernel didn't have MM support compiled in, it actually thinks the executable is a shell script, which you can see eventually displays "Syntax error: ')' expected. That's the ash shell trying to run ./elksmoria!

I hope this all makes sense. If not, please ask more questions.

Thank you!

tyama501, 2023-03-04:
Hello @ghaerr ,

I understand the blink16 does not support medium model yet and if I do blink16 elksmori then it says "not supported medium model" .

Also the reason why I am using the system mode is that I need to put morifiles in root directory.
Is there way to put files when direct mode?

The banner is working on the direct mode.

Thank you.

ghaerr, 2023-03-04:
> Also the reason why I am using the system mode is that I need to put morifiles in root directory.
> Is there way to put files when direct mode?

Currently, the blink16 system call translation just translates `open` and doesn't change the pathname - this would mean you'd need the files in the root directory of your host system, which is not a good idea. I will have to think more about how to best handle this, and have made a note of it. For the time being, elksmoria won't run anyways since the medium model loader isn't done yet. Sounds like we still have lots of work to do to get blink16 usable, it just isn't ready yet for larger programs nor programs that use fixed pathnames. For that reason, I would suggest testing with small programs, as there is so much left to get working.

charlysan, 2023-04-17:
Sorry if this is not related to the main topic, but is there any alternative for having an assembler and debugger within ELKS? So that we could generate binaries from x86 asm code?

charlysan, 2023-04-17:
Thanks @ghaerr. I meant operating from ELKS itself. I saw the "C compiler" discussion and I understand the difficulties, but I thought that maybe for the case of an assembler operating from ELKS it could be easier than the C compiler (?).

cocus, 2023-05-20:
Hi all, this looks so cool! In fact, it's so cool, that I dusted off my 8018x and flashed it with today's tip of master.
The `text-ui` came pre-enabled by default, but that doesn't fit on my ROMFS (I think it's because the actual kernel is placed right after the ROMFS and I should move it a little bit); but I was able to copy the binaries to an SD and run them.
I was able to run the `fm` fine, although really slow (a little bit faster over ethernet for sure!), but `cons` doesn't do anything interesting, except writing colored-characters all over the screen (either via serial or telnet):
[image]
I would like to run the memopad or something cool, and feel the "d-flat" experience (which I was looking for like 4 months ago, never knew the name of this "windowing" scheme).
Do I have to do something special in order for this to work?
Thank you!

ghaerr, 2023-05-20:
Hello @cocus,

Nice to hear from you, how have you been doing?

> but cons doesn't do anything interesting, except writing colored-characters all over the screen 

The reason `cons` doesn't work on your 8018x is that it reads the IBM PC BIOS display screen area at B800:0, which appears to be either random memory or non-existent on your system. It would be possible to hack the direct console with some work, which could then run headless and then use `cons` to display the pseudo-console, but that's probably not worth the effort, and the keyboard would have to be dealt with separately.

> I would like to run the memopad or something cool, and feel the "d-flat" experience

Because of its size, D-Flat and `memopad` are built in a seperate repository, for both UNIX and ELKS. You can take a look at it over [here](https://github.com/ghaerr/dflat) and build the ELKS version using `make -f Makefile.elks`.

> The text-ui came pre-enabled by default

What are you referring to here?

Thank you!

cocus, 2023-05-20:
> Nice to hear from you, how have you been doing?
> 
Pretty good! Thank you! How are you doing? I see you were really busy with elks!

> The reason `cons` doesn't work on your 8018x is that it reads the IBM PC BIOS display screen area at B800:0, which appears to be either random memory or non-existent on your system. It would be [...]

Ah! I thought `cons` was required for memopad to run. I was mistaken. Yes, I agree, completely useless to do this on the headless targets

> What are you referring to here?
>

Oh, if I copy the 8018x.config as my main .config and run the menuconfig, I see that the `text-ui` was enabled by default on the userland. 

> Because of its size, D-Flat and `memopad` are built in a seperate repository, for both UNIX and ELKS. You can take a look at it over [here](https://github.com/ghaerr/dflat) and build the ELKS version using `make -f Makefile.elks`.
>

Ah! cool. I just built it (I had to change the path to my elks TOPDIR under the Makefile.elks) but the binary doesn't run on my board, I always get a ```
# ./memopad
0@�@^: not found
./memopad: 2: Syntax error: ")" unexpected
```
the file seems to be okay. I copied it to the SD card because it won't fit on the rootfs romfs.

ghaerr, 2023-05-20:
> the binary doesn't run on my board,

Oh, `memopad` is built in medium model, since its code size > 64k. That means the kernel needs to have the a.out medium model extension enabled: set `CONFIG_EXEC_MMODEL=y` and rebuild the kernel. (Sorry if that's a pain to reflash!)

> I see that the text-ui was enabled by default on the userland.

I'm sorry I'm being a little dense here, but I still don't know what `text-ui` is referring to, I don't recall having ever seen that term used within ELKS.

cocus, 2023-05-20:
> Oh, `memopad` is built in medium model, since its code size > 64k. That means the kernel needs to have the a.out medium model extension enabled: set `CONFIG_EXEC_MMODEL=y` and rebuild the kernel. (Sorry if that's a pain to reflash!)

Okay, I'll test this. What are the downsides of enabling this? can I do it and keep it as the default?

> I'm sorry I'm being a little dense here, but I still don't know what `text-ui` is referring to, I don't recall having ever seen that term used within ELKS.

It's the name set on the menuconfig for `tui`:
[image]

ghaerr, 2023-05-20:
> What are the downsides of enabling this? can I do it and keep it as the default?

There are no downsides. Its only a config option because of the extra space taken up in the kernel, which is small. Perhaps it should be on by default. (Though not many programs are medium model at this point, I think memopad is the only one :)

> It's the name set on the menuconfig for tui:

Geez, thanks. Yes, I named that and promptly forgot what I called it. Are you thinking it should be off by default for 8018x?

cocus, 2023-05-20:
> There are no downsides. Its only a config option because of the extra space taken up in the kernel, which is small. Perhaps it should be on by default. (Though not many programs are medium model at this point, I think memopad is the only one :)

Well, I enabled it, but it still doesn't run. Could it be because I don't have enough memory? meminfo reports 360kb free.

> Geez, thanks. Yes, I named that and promptly forgot what I called it. Are you thinking it should be off by default for 8018x?

Well, I think it might be a good idea to turn it off by default, just for the sake of reclaiming some bytes on the romfs

ghaerr, 2023-05-20:
Does trying to run memopad act like a shell script command failure, or does the shell return an exec error? If the former, the a.out magic number is not correct, which means either no kernel support or bad executable. Post your memopad and I’ll run it when I get home.

Push text ui off on your open PR and we’ll commit it there.

[20 more replies]

closed 2023-08-03 by ghaerr