ghaerr/elks #1371: Not reading bootopts

toncho11, 2022-07-20:
The kernel does not read the bootopts file. I copied the file manually from Windows. The sys command did not put this file. Actually the sys command installation failed with not finding the "rmdir" command using a 360 KB floppy.

Is this a file that should be stored on a specific place on the filesystem? How should I put it there?
Or it could be the extra "0x80" for the WD8000? I have not used a forth parameter before.

```
## boot options max 511 bytes
#console=ttyS0,57600 debug net=ne2k 3 # sercons, multiuser, networking
#QEMU=1			# to use ftp/ftpd in qemu
#TZ=MDT7
LOCALIP=192.168.1.67
#HOSTNAME=elks16
#ne2k=12,0x300,,0x80
wd8003=3,0x280,0xD000,0x80
#3c509=11,0x330,,0x80
#bufs=2500		# system buffers
#sync=30		# autosync secs
#init=/bin/init 3 n	# multiuser serial no /etc/rc.sys
#init=/bin/sh		# singleuser shell
#root=hda1 ro		# root hd partition 1, read-only
#console=ttyS0,19200 3	# serial console
```

toncho11, 2022-07-21:
At boot time I see the default values for WD, not the ones in the bootopts.

Mellvik, 2022-07-21:
Hi @toncho11,

can you post the boot messages?

The requirements for /bootopts is that it resides in the root directory, that the file size does not exceed 511 bytes and the two first chars must be ##. It looks good but check the file size anyway. 

If it has been edited in windows, it may have been saved with <CR><LF> line termination instead of just <LF>. I don't know if it makes a difference, but check.

—M

ghaerr, 2022-07-21:
Hello @toncho11,

I think the problem is that the /bootopts directory entry is not within the first 8 directory entries on the volume. That is, if the /bootopts file were created by copying after the volume already has more than 7 files in the root directory, it won't work. I have noted this, and code in setup.S will have to be changed to read more root directory sectors in order to work.

A temporary workaround for this would be to 1) rename /bootopts to /bootopts.tmp, perform an unsorted directory listing (`ls -U` on ELKS, or I think just `dir` on Windows), and delete any one of the first 8 files shown, then copy /bootopts.tmp to /bootopts; or 2) create a blank diskette and copy /linux and /bootopts to it, before running `sys` on it.

Another workaround would be to use a FAT floppy image created directly from an ELKS build, which will have /bootopts always created as the second root directory entry.

I will look into seeing whether `sys` might be able to create /bootopts first before copying other files, to possibly help with this problem.

As @Mellvik suggests, look at the boot messages - this should be showing a "/bootopts ignored:..." message to indicate that the kernel is built for /bootopts, but that the file is not found (or size is greater than 1 sector).

Thank you for the bug report!

toncho11, 2022-07-21:
So it is with LF at each line.
It says "/bootopts ignored ..."
It is 485 bytes.
ls -U shows bootopts as second entry, but I think it is still not in the first 8 files. Do we count the files inside the directories?

I think the proper solution is to copy the file during sys.

ghaerr, 2022-07-21:
> s -U shows bootopts as second entry, but I think it is still not in the first 8 files. 

I don't understand - each directory or file (that is, line item shown by ls -U) is an "entry". It needs to be in the first 8.

What do you mean by "2nd entry, but not in first 8"?

> Do we count the files inside the directories?

No, we just count directory or file entries in the root directory.

toncho11, 2022-07-21:
I formatted my drive, copied bootopts and then executed sys.
Now it works.

As a side issue "net start" says "Exec format error" ? Other commands work ok.

ghaerr, 2022-07-21:
> I formatted my drive, copied bootopts and then executed sys.
Now it works.

Great. I'll look into enhancing sys as described above, thanks.

> As a side issue "net start" says "Exec format error" ? Other commands work ok.

Hmmm... can you uncomment the "#set -x" on line 15 of /bin/net? That will show the commands it's trying to execute, and we can see what program its trying to run that isn't working.

toncho11, 2022-07-21:
It did not show more info. 

I did manually "source /etc/net.cfg" and I got kernel panic and restart

toncho11, 2022-07-21:
Interesting: 

At boot it says: /etc/rc.sys: Exec format error

toncho11, 2022-07-21:
Offf
echo $LOCALIP says "Not supported by standalone shell:..."

toncho11, 2022-07-21:
Maybe I have the IP configured, but not the mask and gateway in /etc/net.cfg ?

ghaerr, 2022-07-21:
> At boot it says: /etc/rc.sys: Exec format error
> echo $LOCALIP says "Not supported by standalone shell:..."

The problem is that you are running the `sash` standalone shell, which doesn't support shell scripts: both /etc/rc.sys and /bin/net are shell scripts. Change your config to run the ASH shell and things will work fine.

toncho11, 2022-07-21:
OK but vi has some bugs ... I have seen this before.

ghaerr, 2022-07-21:
Thanks @toncho11 for your testing, I will also look into changing the messsages from ELKS/sash so that these kinds of things are more easily understood by users, rather than arcane technical messages like "Exec format error". Perhaps "Executable format error - use bash shell".

> echo $LOCALIP says "Not supported by standalone shell:..."

Please list those too.

ghaerr, 2022-07-21:
How do you save and exit edit ??????

^X

toncho11, 2022-07-21:
How do I change to ash?

toncho11, 2022-07-21:
There is no ash in /bin

ghaerr, 2022-07-21:
Use `make menuconfig`:
[image]

toncho11, 2022-07-21:
But the default is sash and it does not allow me to run scripts?

ghaerr, 2022-07-21:
@toncho11, you can either build your own using menuconfig as described above, or mount a distribution floppy and copy /bin/sh to your floppy.

> But the default is sash and it does not allow me to run scripts?

No, the default shell is ash (bash). Sash is built for tiny systems and doesn't support bash shell scripting. It seems your .config is setup for sash. You may want to copy ibmpc-1440.config to .config, then run menuconfig to set 360k floppy, or just pick up one of the 360k floppy images from any of the last commits, which are automatically built.

toncho11, 2022-07-21:
I recompiled with ash selected, but in /target there is no ash.

ghaerr, 2022-07-21:
> I recompiled with ash selected, but in /target there is no ash.

That is fine, the bash shell is default, and named /bin/sh. If sash is also selected, it will be named /bin/sash (otherwise it is /bin/sh). You should be good to go.

ghaerr, 2022-07-21:
No updated kernel, just replace /bin/sh should do it.

toncho11, 2022-07-21:
I am too tired ... I can not make it work.
I can not test the network because I have sh that is sash instead of ash ...

toncho11, 2022-07-22:
@Mellvik this is the same card that did not work with my 386 #1303 .

Mellvik, 2022-07-22:
Thanks - I was wondering about that.

What's the NIC setup you're using now? BTW - thanks for reminding me, I'm going to add a comment in #1303.

-M

toncho11, 2022-07-22:
WD IRQ 3, 280, D000

toncho11, 2022-07-22:
running sash confuses people in my opinion - it is limited and you don't realize the problem in the beginning.
Thank you @ghaerr !

closed 2022-07-22 by toncho11

Mellvik, 2022-07-22:
It is important - as I also pointed out above - to realize that ash and sash are different tools for different purposes. It is perfectly normal for linux and unix systems to have several shells - serving different purposes and requirements.

Size may at one point have been an important advantage for sash. This is - as @ghaerr points out - no longer the case (changing #defines in the source code changes this if desired). sash has a small memory footprint, is very fast and includes a lot of utilities that enables broad functionality on very small systems. For a 360k/512MB/8086 or 8088 system it's pure gold, the builtin commands opening up 90k bytes of the floppy for other programs.
The lack of scripting is the price to pay and may be moot for such resource-tight systems.

IOW - and IMHO - sash is great for what it does.

BTW I just noticed this morning, the sas man page is totally out of date and mostly misleading. It's on my list.

—M

toncho11, 2022-07-22:
Yes, but with sash you can not run `net start`. 
The problem is that you do not see the difference on the console until something goes wrong. The prompt is the same.
There can also be a message when sash is started saying "Warning: bash scripting is disabled".

Mellvik, 2022-07-22:
> Yes, but with sash you can not run net start.
That's right. If you're in a position to run net start you would not be using sash.

That's the point. If you have enough resources to run ethernet, using sash as the main shell is nonsensical. There's (AFIC) nothing in ELKS that encourages or forces such use.

Why is this an issue?

-M

ghaerr, 2022-07-23:
> If you're in a position to run net start you would not be using sash.

Actually, if ELKS is configured for only `sash` and not `ash`, this problem will arise, which is what happened with @toncho11.

I've left everything as-is for `sash`, but changed the "Exec format error" message to "Exec format error (install ash to use shell scripts)" to hopefully better inform the user in the future, in #1373.

Mellvik, 2022-07-23:
> I've left everything as-is for sash, but changed the "Exec format error" message to "Exec format error (install ash to use shell scripts)" to hopefully better inform the user in the future, in #1373 <https://github.com/jbruchon/elks/pull/1373>.
> 
> 
That's good - all or most error messages in sash are terse - to say the least. For a reason but also hard to understand at times.

I don't know if it would be useful, but it's obviously easy enough to change the sash command prompt to (say) 'sash# ' instead of just the hash.

Also - and for the record - sash does not prevent scripts like net and sys from running. They run just fine as long as /bin/sh is ash. 

—M

ghaerr, 2022-07-23:
> That's good - all or most error messages in sash are terse - to say the least

In this particular case, the errors are coming from /etc/perror, (the `perror` text database), so the text was changed there. It looks like `sash` uses `perror` for all error strings.

> For a reason but also hard to understand at times.

These could be changed, but they are also used system-wide, so they are kind of generic.

> sash does not prevent scripts like net and sys from running. They run just fine as long as /bin/sh is ash.

Yes, this is a little-known enhancement that's pretty nice. Most users have no idea that the ash shell is forked when running shell scripts or any file redirection commands from the sash shell.  

In summary, it's probably a bad idea for a new user to configure the system with sash only, but there really isn't any warning against them doing so and what the consequences of that are.

Mellvik, 2022-10-11:
I'm happy to hear that @toncho11!! Congrats!

> 
> Everything works now (after I got some sleep).
> So the "bash" version of sh is 48kb, the "sash" version is 36kb. I used the one from the 360kb floppy and it worked.
> 
Yes, that's about right. Remember though that the two shells are very different animals. While /bin/sh (which is really ash, not bash) is a compact universal shell, sash is great for tight memory situations, not the least because many of the standard commands are builtin and don't need to be in the /bin directory. Such as ls, cp, mv, chmod, mount … - take a look at the man page for a complete list, 25 commands in total. This makes a big difference when creating a 360k boot image. sash does not have the visual command line editing recently added to ash, but it does have (csh/bash style) command history which - when you learn how to use it, does about the same thing.
> @Mellvik <https://github.com/Mellvik> Network works with WD driver. I tried telnet for now.
> This is for me: one needs to configure the IP and WD in the bootopts and then the mask, the gateway and the link layer in net.cfg (and has bash as shell).
> 
This is kind of messy right now with addresses and more in 3 different places. It's on my list to come up with suggestions for that and ideas are welcome. The setup command (script) @ghaerr wrote a couple of months back is a good starting point. What's particularly confusing right now is that the net interface in /bootopts may be different from the one in net.cfg (link=) so 'net start' w/o parameters will not work.

Thank you.

—M