ghaerr/elks #1374: support more ethernet cards!

sparky4, 2022-07-24:
i have a card i would like to support. it is the Intel 8-16 ISA LAN card and the xircom parallel ethernet card and the pcima card of xircom. which is called the Xircom buscard ethernet 10/100 adapter. i have computers that use these cards.

Sasytoremember, 2022-08-03:
Might i suggest cards from Qemu Virt Manager.

Mellvik, 2022-12-05:
Hi @sparky4 & @Sasytoremember,

We'd love to see support for more Ethernet cards in ELKS, contributions would be very welcome. We've  come a long way on this and many other fronts recently and have a lot of useful experience to share with contributors.

A primary prerequisite for developing a new driver is the availability of a physical card. Also, it is advantageous in order for a new driver to be generally useful that is uses the ISA bus. And obviously the development skills to take on such a challenge. 

@ghaerr, I suggest we close this issue now.

-M

closed 2022-12-05 by ghaerr

sparky4, 2023-03-14:
im willing to donate some cards