ghaerr/elks #1412: [fm] Could not edit file

toncho11, 2022-08-20:
I am testing the new tui fm in an emulator.
I tried shift+e to produce "E" in order to edit a file, but the system hangs every time. Just viewing the file is OK though. 
I am using the latest hd32mbr-fat.img.

ghaerr, 2022-08-20:
Hello @toncho11,

Thanks for using and testing `fm`, you may be the first :)

When you type 'E', does anything display in the last line? By default, `vi` is run (which should be in your path). I tried removing `vi`, and got the following, but no hang:
[image]

Is it possible the emulated system is running short on RAM? I can't (yet) duplicate what you're seeing.

Thank you!

toncho11, 2022-08-21:
I am using https://copy.sh/v86/ as an emulator and it seems to be some problem with IO: 

[image]

toncho11, 2022-08-21:
So it is related to the ELKS image (or the emulator). It does work on a minix image, but not on fat image.

toncho11, 2022-08-21:
No man page for "edit".

ghaerr, 2022-08-21:
> So it is related to the ELKS image (or the emulator). It does work on a MINIX image, but not on a FAT image.

You're saying ELKS itself works with the emulator on MINIX images, or `fm` works on MINIX images? I'm getting confused as to whether the issue here is `fm` or the emulator used to run ELKS. I'm not sure if you've used this emulator before, but its likely a setup issue with it, or the infamous CHS mismatch for hard drive images and the BIOS expectation, which would require a custom ELKS HD image.

> No man page for "edit".

This is the MINIX editor `medit`, perhaps there's a man page out there for it on the MINIX distribution... I can check on that.

toncho11, 2022-08-21:
I am using the same emulator for both, but the fat version does not work. Have you tried fat images with QEMU and fm? Actually the fat image with this web emulator work ok apart editing a file. For me it

ghaerr, 2022-08-21:
> Have you tried fat images with QEMU and fm? 

No, but I just did - and things are acting very strangely. Not sure yet, but something's definitely wrong. Sometimes the cursor will move on up/down arrow, sometimes not; I notice that all files are green (marked as executable), this may be an issue with being able to edit the file. Finally, I saw a stack overflow and then the system crashed. Meanwhile, everything runs great on MINIX fs!

Now that I think of it, the FAT filesystem doesn't keep UNIX-style read/write/execute for files, and all files are marked as 'rwx'; this could be an issue for `fm`, but there's definitely other problems. Stay tuned - I'll report back with more findings.

Thanks for your testing!!

toncho11, 2022-08-25:
Hi @ghaerr,

I am testing the same way on https://copy.sh/v86
It gives the same problem - I/O error with dev and block as above for any "text" file I try to open in /etc
It is either fm or badly constructed fat .img file?

toncho11, 2022-08-25:
So it seems that with: `export EDITOR=edit` it does work. 
With default settings (vi) it does not. It is something related to the execution of vi.

toncho11, 2022-08-25:
OK. There is some problem when opening files with vi. 
It is not the same each time. Some times it works, most of the time it does not. 
It might depend if you are inside the /etc folder or not when accessing hosts.

toncho11, 2022-08-25:
I try vi /etc/hosts and it works.
I try a second time the same command and it halts.

toncho11, 2022-08-25:
So now it is more "vi on FAT problem".

toncho11, 2022-08-25:
OK it is a problem, not with vi, but rather with write operations.
I did `cp ./issue ./issue2` and on the second time I get an error.
It must be the emulator in the end, because on QEMU it always works.

ghaerr, 2022-08-25:
> But I think that QEMU is very fast. Fast enough to hide some error messages or to compensate for the problem???

No, the error messages are scrolled, so they would not be overwritten.

> It must be the emulator in the end, because on QEMU it always works.

It seems the emulator is possibly setup improperly, as I/O errors are very likely not ELKS, but incompatibility between the emulated hardware. Hard drive CHS would be the first guess.

> I am using the latest hd32mbr-fat.img.

Have you tried using a floppy image? This would remove CHS and hard drive setup from being a variable.

> It is either fm or badly constructed fat .img file?

There's no question that the previous `fm` had a stack overflow problem, but I think the continuing problems you are seeing now are a result of the HD image and your emulator incompatibility or setup.

toncho11, 2022-08-25:
I tried with the floppy images. It still gives the same problems.

toncho11, 2022-08-25:
The emulator  https://copy.sh/v86 uses : SeaBios 1.12
QEMU uses SeaBios 1.16.

ghaerr, 2022-08-25:
I've duplicated the problem, but don't have an answer yet, very strange.

Both the XMS (which uses unreal mode and may be improperly emulated), as well as the non-XMS versions of ELKS on MINIX filesystem work fine, but both FAT versions will crash as soon as `vi` is executed. I'm running the fd1440-fat.img images.

I'm not sure how to even start debugging this. We could be dealing with an instruction emulation issue (we had this issue on the PCjr emulator with 8086 instructions that weren't implemented on the PCjr using an early 8088 chip, but that was fixed). 

ELKS lower level doesn't operate differently for FAT, but I'm not seeing any I/O errors. If you get any on I/O errors on the 1440K floppy, I'd like to see them.

ghaerr, 2022-08-25:
Further testing shows that large program fails: on the latest build, trying to run `fm`, `cp`, or `fsck` with no arguments displays "file memory too large". I can produce some debug kernels that show the file exec progress to look further into this.

I would guess at this point we're dealing with a memory incompatibility of some kind. I have the emulator memory set to 640K.
[image]

toncho11, 2022-08-25:
It is not just vi.
I copy a file - its ok.
I copy it a second time - it crashes. 

So one difference is that on the floppy it always halts without error - while on the hdd it does both - no error or it prints a message bioshd IO error - usually multiple.

toncho11, 2022-08-25:
https://github.com/copy/v86#compatibility

Actually not everything works with this emulator!

toncho11, 2022-08-25:
Here: https://github.com/copy/v86

It says explicitly that these are missing:

Task gates, far calls in protected mode
Some 16 bit protected mode features
Single stepping (trap flag, debug registers)

ghaerr, 2022-08-25:
> Actually not everything works with this emulator!

Thank you for finding that out! I'm currently swamped with other ELKS issues right now, so I think I'll leave this on the back burner. I've noted the issues, but don't really have time to debug their emulator or figure out how to make ELKS FAT compatible with it at the moment. 

Since the ELKS MINIX version seems to work fine, perhaps we should upload a version for them to put on the website?

ghaerr, 2022-08-25:
> Task gates, far calls in protected mode
Some 16 bit protected mode features
Single stepping (trap flag, debug registers)

We don't use any of those... but apparently something in the FAT filesystem code seems to be not emulated correctly. It is hard to say, since we'd have to inspect the `ia16-elf-gcc` output. Lots of work to debug!

toncho11, 2022-08-25:
No, no - we are done with this.

PR #1417 (2022-08-25): [cmds] Fix fm stack overwrite
Should fix #1412.

Hello @toncho11,

Finding this bug turned into another day-long rabbit hunt but I finally found it: a stack overwrite beyond the length of an array which only happened to turn up on FAT filesystems. I've tested it pretty well. Let me know what you think of `fm` now that it's working, and thank you again for your testing!

I also ended up fixing up the C library `getcwd` routine, there's nothing wrong with it but I dug deep down its rabbit hole for a while, but it was written with K&R C, which is now ANSIfied.