ghaerr/elks #1429: Boot fails on minix fs if root dir is big

Mellvik, 2022-08-30:
Continuing from discussion in https://github.com/jbruchon/elks/discussions/1380#discussioncomment-3503542:

Bug in partition `bootblock` when the root directory grows beyond one block (1k).

@ghaerr, I've included the directory contents from my scroll buffer, but I don't think it's interesting anymore. The dir looks perfectly fine, and what happens is that when the directory size exceeds 1024b, boot will not find ´/linux`.

So there seems to be a bug in the boot block code that kicks in when the read loop runs more than once. I've run a few tests on this and it's consistent. >1024 = error 3.

--M

```
0ae000: 01 00 2e 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
0ae010: 01 00 2e 2e 00 00 00 00  00 00 00 00 00 00 00 00  ................
0ae020: 02 00 6c 69 6e 75 78 00  00 00 00 00 00 00 00 00  ..linux.........
0ae030: b6 00 62 6f 6f 74 6f 70  74 73 00 00 00 00 00 00  ..bootopts......
0ae040: 04 00 64 65 76 00 00 00  00 00 00 00 00 00 00 00  ..dev...........
0ae050: 2d 00 62 69 6e 00 00 00  00 00 00 00 00 00 00 00  -.bin...........
0ae060: 2e 00 65 74 63 00 00 00  00 00 00 00 00 00 00 00  ..etc...........
0ae070: 2f 00 6d 6e 74 00 00 00  00 00 00 00 00 00 00 00  /.mnt...........
0ae080: 30 00 74 6d 70 00 00 00  00 00 00 00 00 00 00 00  0.tmp...........
0ae090: 31 00 72 6f 6f 74 00 00  00 00 00 00 00 00 00 00  1.root..........
0ae0a0: 9f 00 6f 6c 69 6e 75 78  00 00 00 00 00 00 00 00  ..olinux........
0ae0b0: 00 00 6c 69 6e 75 78 2e  74 32 00 00 00 00 00 00  ..linux.t2......
0ae0c0: 00 00 6c 69 6e 75 78 2e  74 31 31 6f 6b 00 00 00  ..linux.t11ok...
0ae0d0: 00 00 6c 69 6e 75 78 2e  73 74 64 70 63 00 00 00  ..linux.stdpc...
0ae0e0: 00 00 6c 69 6e 75 78 2e  74 31 00 00 00 00 00 00  ..linux.t1......
0ae0f0: 00 00 6c 69 6e 75 78 2e  74 39 6f 6b 00 00 00 00  ..linux.t9ok....
0ae100: 00 00 6c 69 6e 75 78 2e  39 6f 6b 00 00 00 00 00  ..linux.9ok.....
0ae110: b9 00 6d 6e 74 32 00 00  00 00 00 00 00 00 00 00  ..mnt2..........
0ae120: 00 00 6c 69 6e 69 78 2e  78 31 6f 6b 00 00 00 00  ..linix.x1ok....
0ae130: 00 00 6c 69 6e 75 78 2e  31 30 00 00 00 00 00 00  ..linux.10......
0ae140: c3 00 6d 6e 74 33 00 00  00 00 00 00 00 00 00 00  ..mnt3..........
0ae150: c5 00 6d 6e 74 34 00 00  00 00 00 00 00 00 00 00  ..mnt4..........
0ae160: 00 00 6c 69 6e 75 78 2e  6f 72 69 67 00 00 00 00  ..linux.orig....
0ae170: 00 00 6c 69 6e 75 78 2e  6e 65 77 6f 6b 00 00 00  ..linux.newok...
0ae180: 00 00 6c 69 6e 75 78 2e  6e 65 77 31 62 00 00 00  ..linux.new1b...
0ae190: 00 00 6c 69 6e 75 78 31  30 00 00 00 00 00 00 00  ..linux10.......
0ae1a0: 00 00 6c 69 6e 75 78 31  31 00 00 00 00 00 00 00  ..linux11.......
0ae1b0: 00 00 6c 69 6e 75 78 2e  74 33 00 00 00 00 00 00  ..linux.t3......
0ae1c0: 00 00 6c 69 6e 75 78 2e  74 34 00 00 00 00 00 00  ..linux.t4......
0ae1d0: 00 00 6c 69 6e 75 78 2e  74 35 78 6d 73 00 00 00  ..linux.t5xms...
0ae1e0: 00 00 6c 69 6e 75 78 2e  74 36 00 00 00 00 00 00  ..linux.t6......
0ae1f0: 00 00 6c 69 6e 75 78 2e  74 37 00 00 00 00 00 00  ..linux.t7......
0ae200: 00 00 6c 69 6e 75 78 2e  74 38 6f 6b 00 00 00 00  ..linux.t8ok....
0ae210: 00 00 6c 69 6e 69 78 2e  74 39 00 00 00 00 00 00  ..linix.t9......
0ae220: 00 00 6c 69 6e 75 78 2e  74 31 30 62 61 64 00 00  ..linux.t10bad..
0ae230: 00 00 6c 69 6e 75 78 31  32 00 00 00 00 00 00 00  ..linux12.......
0ae240: 00 00 6c 69 6e 75 78 31  33 00 00 00 00 00 00 00  ..linux13.......
0ae250: 00 00 6c 69 6e 75 78 31  34 00 00 00 00 00 00 00  ..linux14.......
0ae260: 00 00 6c 69 6e 75 78 31  35 00 00 00 00 00 00 00  ..linux15.......
0ae270: 00 00 6c 69 6e 75 78 2e  74 31 32 62 61 64 00 00  ..linux.t12bad..
0ae280: 00 00 6c 69 6e 75 78 2e  74 31 31 62 61 64 00 00  ..linux.t11bad..
0ae290: 00 00 6c 69 6e 75 78 2e  74 31 33 62 61 64 00 00  ..linux.t13bad..
0ae2a0: 00 00 6c 69 6e 75 78 2e  74 31 34 62 61 64 00 00  ..linux.t14bad..
0ae2b0: 00 00 6c 69 6e 75 78 31  36 00 00 00 00 00 00 00  ..linux16.......
0ae2c0: 00 00 6c 69 6e 75 78 31  37 00 00 00 00 00 00 00  ..linux17.......
0ae2d0: 00 00 6c 69 6e 75 78 31  39 00 00 00 00 00 00 00  ..linux19.......
0ae2e0: 00 00 6c 69 6e 75 78 32  30 00 00 00 00 00 00 00  ..linux20.......
0ae2f0: 00 00 6c 69 6e 75 78 31  38 00 00 00 00 00 00 00  ..linux18.......
0ae300: 00 00 6c 69 6e 75 78 32  31 00 00 00 00 00 00 00  ..linux21.......
0ae310: 00 00 6c 69 6e 75 78 33  32 00 00 00 00 00 00 00  ..linux32.......
0ae320: 00 00 6c 69 6e 75 78 32  32 00 00 00 00 00 00 00  ..linux22.......
0ae330: 00 00 6c 69 6e 75 78 32  33 00 00 00 00 00 00 00  ..linux23.......
0ae340: 00 00 6c 69 6e 75 78 32  34 00 00 00 00 00 00 00  ..linux24.......
0ae350: 00 00 6c 69 6e 75 78 32  35 00 00 00 00 00 00 00  ..linux25.......
0ae360: 00 00 6c 69 6e 75 78 32  36 00 00 00 00 00 00 00  ..linux26.......
0ae370: 00 00 6c 69 6e 75 78 32  37 00 00 00 00 00 00 00  ..linux27.......
0ae380: 00 00 6c 69 6e 75 78 32  38 00 00 00 00 00 00 00  ..linux28.......
0ae390: 00 00 7a 7a 7a 00 00 00  00 00 00 00 00 00 00 00  ..zzz...........
0ae3a0: 00 00 6c 69 6e 75 78 32  39 00 00 00 00 00 00 00  ..linux29.......
0ae3b0: 00 00 6c 69 6e 75 78 33  30 00 00 00 00 00 00 00  ..linux30.......
0ae3c0: 00 00 6c 69 6e 75 78 33  31 00 00 00 00 00 00 00  ..linux31.......
0ae3d0: 00 00 6c 69 6e 75 78 33  33 00 00 00 00 00 00 00  ..linux33.......
0ae3e0: 00 00 6c 69 6e 75 78 33  34 00 00 00 00 00 00 00  ..linux34.......
0ae3f0: 00 00 6c 69 6e 75 78 33  35 00 00 00 00 00 00 00  ..linux35.......
0ae400: b8 c0 07 8e d8 b8 00 01  8e c0 b9 00 01 31 f6 31  .............1.1
0ae410: ff fc f3 a5 ea 19 00 00  01 bf f4 3f 8e d8 8e d0  ...........?....
0ae420: 89 fc bb 78 00 1e 06 31  c0 8e c0 26 c5 37 07 b1  ...x...1...&.7..
0ae430: 06 fc 57 f3 a5 5f 1f c6  45 04 24 1e 8e d8 89 3f  ..W.._..E.$....?
--More--
```

ghaerr, 2022-08-30:
> So there seems to be a bug in the boot block code that kicks in when the read loop runs more than once.

I suspect the problem isn't the read loop executing twice, but instead the wrong block is bring read, due to something in the root inode (inode number 1) or something like that.

Any chance you can upload a floppy image with this problem? Or does this only occur on HD? Otherwise, instructions on how to run `split` to produce this will help. I will then look into it.

Mellvik, 2022-08-30:
HI @ghaerr,

this problem is repeatable on any medium capable of holding enough inodes to blow the root directory above 64 entries.
Which includes 1440k floppies.

You can do a
split -b 1024 linux 
in the root directory, which will complain about running out of inodes if on a 1440k or smaller floppy, but it doesn't matter. by then there are more than enough files.

Then a reboot and the error is there.

-M

PR #1430 (2022-08-30): [boot] Fix MINIX boot when root directory larger than 1 block
Fixes #1429.

The MINIX boot was reading all directory blocks into the same block buffer, which caused filesystems with large root directories to overwrite the /linux directory entry and not match. The /linux and /bootopts entries still need to reside in the first directory block: this fix adds three bytes to the MINIX boot block, and we're down to 8-9 bytes left, and can't do much more!