ghaerr/elks #1436: Does not show anything with ps command

tyama501, 2022-09-05:
Hello,

[image]

It seems the ps commad does not show anything on 02 Sep build.
Did it change something recently?

toncho11, 2022-09-06:
It should show `init`, `sh` and `ps` ?

ghaerr, 2022-09-06:
> Are you saying that the image built for PC-98 (fd1232.img) via Github Contiguous Integration
> Yes, I just wrote how to get the newest image in the document and found this.

I suspect the problem is that `ps` is dependent on the internal structure of the `task_struct` array of the kernel, and reads it using /dev/kmem. The issue is likely that CONFIG_CPU_USAGE is not set the same for the kernel and ps.c compilations, thus causing the task_struct to not match. (I will have to check, but am wondering whether CONFIG_CPU_USAGE is set for a default kernel build - I know it is set for the pc98-1232.config build). 

This could probably be fixed by adding the following line in `build.sh`:
```
# Build PC-98 kernel, some user land files and image
if [ "$1" = "auto" ]; then
    echo "Building PC-98 image..."
    cp pc98-1232.config .config
    make kclean || clean_exit 9
    rm elkscmd/sys_utils/clock.o
    rm elkscmd/sys_utils/ps.o    # <-- add this line
    rm elkscmd/basic/*.o
    rm elkscmd/nano-X/*.o
    make -j1 || clean_exit 10
fi
```
I have been working towards not requiring that ELKS applications have to be built using any kernel options, but there are a few exceptions, namely `clock`, `ps`, `basic` and `nano-X`. In this case, it would be preferable to have CONFIG_CPU_USAGE turned on by default; that wasn't done because it uses another 5 words per task, which I suppose is no big deal. 

I'll post the above fix and see whether that works, then consider the CONFIG_CPU_USAGE issue.

tyama501, 2022-09-06:
I built from the latest commit and yes it worked.

[image]

Thank you.

PR #1437 (2022-09-07): [build] Fix non-working ps on PC-98 CI build
Fixes #1436.

The problem was not CONFIG_CPU_USAGE, which defaults enabled in PC-98, 8018X and IBM PC, but rather KSTACK_BYTES, which is included in the task structure, and has a different size between PC-98 and IBM PC builds. Unfortunately, the `ps` command must know the correct task_struct size per OS build in order to work, so it must be built separately for each as well.

Also corrects problem for proper `basic` build for 8018X.
closed 2022-09-07 by ghaerr