ghaerr/elks #1441: System requirements

eloydegen, 2022-10-14:
Hi,

Maybe I am looking at the wrong places, but I could not find what the minimum amount of required RAM is for ELKS. I have an IBM Portable Personal Computer with 512KiB RAM and don't know if that's a recommended amount for ELKS. I don't have a 5.25" floppy drive yet, so I can't test it yet on that machine.

ATroubleshooter, 2022-10-15:
512k is good enough to run ELKS. As far as I remember, someone here tried and ran it on 256k, although in a somewhat limited configuration.

eloydegen, 2022-10-15:
@ATroubleshooter Thank you! I'll keep this issue open because I'd like to hear some input on whether this should be listed in the README.

Mellvik, 2022-10-16:
Thank you @eloy,

I agree with you. We should add something like 'minimum system requirements' to the beginning of the README.

—M

eloydegen, 2022-10-17:
@Mellvik Nice! Would it make sense to distinguish between 'required' and 'recommended' RAM? Assuming the standard release binaries, no custom configurations. Any other system requirements apart from the RAM? Since PC-98 is supported, IBM PC compatibility is not required, but exceptions have to be explicitly supported I assume.

ghaerr, 2022-10-18:
> Would it make sense to distinguish between 'required' and 'recommended' RAM? 

The "system requirements" for ELKS vary greatly, depending on the platform and programs configured to run during boot. For instance ELKS can run on a ROM-based 8018X SBC with 128k RAM, while a PC version with no networking running /etc/init, the ASH shell and /etc/rc.sys likely requires just 256k. Add networking with the local daemons running and you'd want 512k minimum. I'd suggest perhaps describing the memory requirements of each of ELKS components instead.

That said, a goal of this project is to boot up from any (ancient) PC-compatible system with 256K RAM or more (including the still-unsolved boot problems for the PCs in #1327). However, the stock images and /etc/rc.sys programs may require more, or a specialized configuration. Some testing should show real requirements for stock images.

Thank you!

NathanGibbs3, 2022-10-18:
Maybe add an entry to the Wiki as well.

eloydegen, 2022-10-19:
Thanks for the clear answer, I will try to do some tests in the coming days.

closed 2023-06-21 by ghaerr