ghaerr/elks #1442: Location of documentation

eloydegen, 2022-10-18:
Looking through older issues I noticed that the creation of the wiki was discussed in issue #666. It's quite nice already I think!

But looking at the repository itself, it seems that there still is a `Documentation` directory, which contains different text in comparison to the wiki. Shouldn't that be in sync with the wiki? There also is a historic folder. In Git, things are never really deleted and can be brought back if needed. Would it make sense to keep those files visible if they are outdated? Should they be merged in cases where it is still up-to-date?

The `Documentation` directory also contains HTML files. What was the rationale behind that choice? In my view, Markdown would make more sense, and then use a generator for the HTML, instead of putting the HTML in the source.

Thanks!

eloydegen, 2022-10-19:
> You're trolling us, don't you? Is this a sort of DOS-attack on developers, when you're requesting all kinds of feature-creep, while they still got way more important things to do?

No, I'm not trolling. And I'm not requesting them to fix it, I want to do contribute. But I'm not going to send PRs against their vision of the project, and I have no idea what their current vision on the documentation is.

Also, in my view, documentation is not a fun aspect of software development, but it is definitely *important*, also with regard to attracting new developers which could even reduce work for current developers in the future. Having everything well-ordered is a requirement for the former in my view. I hope this clears something up. Feel free to close the issue if you or other developers think everything is fine in it's current form, I don't want to force any change of course.

ATroubleshooter, 2022-10-19:
> No, I'm not trolling. And I'm not requesting them to fix it, I want to do contribute. But I'm not going to send PRs against their vision of the project, and I have no idea what their current vision [...]

Alright.

> 
> Also, in my view, documentation is not a fun aspect of software development, but it is definitely *important*, 

No disagreement about that.

>
>also with regard to attracting new developers which could even reduce work for current developers in the future. Having everything well-ordered is a requirement for the former in my view. I hope this clears something up.

Yes, thank you.

ghaerr, 2022-10-19:
Hello @eloydegen,

Concerning ELKS' Documentation directory: the ELKS project has been around for quite some time, I think it forked from Linus' original Linux work in the early 1990's, so at least 25 years ago. It has a very long history, and only in the last few years has been maintained on GitHub. Thus, the previously contributed documentation is in various formats all predating Markdown. ELKS, after a few years of hectic, semi-finished development in the beginning, went into a long period of decline until just a few years ago.

At this point, I would not recommend deleting files in the Documentation directory, unless they're known incorrect, but you're welcome to contribute enhancements based on experience using ELKS, at any time. I've made a note to review the Documentation directory, but that sort of thing has been lower on my priority list, given all the other work ELKS has needed the last couple of years. 

We've been promoting the Wiki, rather than documentation within the source code repo, as it is so much more accessible to non-developer users. I would venture a guess that most users have never seen or read any of the source code documentation. We've also recently added man pages, available on 2880K+ disks and HD images, which greatly help the usability of ELKS for users.

If you want to jump in and write replacement documentation, that'd be great - let us know what you're thinking and I'll be glad to help you. If not, that's ok too, we can close this issue, I keep a large TODO list based on suggestions.

Thank you!

eloydegen, 2022-10-19:
Hi @ghaerr, thanks a lot for your clear answer! I'll try taking a look into reordering the documentation and try to have everything on the wiki as an ultimate end goal. The first thing I'll attempt is checking to what extend there is duplicate information and remove those duplicates if they exists. After that, I'll try to merge stuff from the Documentation folder into the wiki.