ghaerr/elks #1445: New names for Ethernet interfaces

Mellvik, 2022-11-03:
Status:
- ELKS network device names reflect the underlying NIC type - currently ne2k, 3c509, wd8003, pointing to individual drivers for each card type

Problem: 
- Increasingly, the drivers support a family of NICs, not only one, which means the old device names are misleading. The `ne2k` device may be a ne2k or a ne1k. I'm working on fixing the wd8003 driver to make it work smoothly with the wd8013 card, making the old device name confusing at best. 

Solution (suggested):
- Change the names of the `ne2k` and 'wd8003` devices to `ne` and `wd` respectively, possibly with a '0' added to make it even more obvious that this is a network interface when listing the /dev directory (and open up for future support of multiple interfaces of the same type).
- Amend `netstat` (possibly with a -i option) to list the currently configured interfaces, including the relevant info from the boot messages.

Being in the middle of this, I can include these changes to the 'project'.

--M

ghaerr, 2022-11-03:
Hello @Mellvik,

All sounds good, and probably good to with the appended '0' as well, if you like. What new name are you thinking for the 3c509 family, /dev/3c0?

Please check the /etc scripts and other locations (e.g. MSDOS FAT fake dev table etc) to ensure they're updated with the new names (if any are used) as well.

Thank you!

Mellvik, 2022-11-03:
Thanks you @ghaerr-

I actually intended to leave the 3c509 alone for now. I don't see ELKS supporting any of the other EtherLink III interfaces (Microchannel, PCMCIA), so it remains a driver for a single interface type.
That's admittedly inconsistent, so let's come up with something that can work. '3c' looks kinda weird to me and may create confusion if someone creates a driver for some other 3com interface (503, 505, ).
'ep' doesn't say much but that's what freebsd has been using forever.
They use 'ed' for the 3c503 btw.

What do you think?

—M

ghaerr, 2022-11-03:
Given your suggestion(s) to shorten the names of the devices, since they support more NICs with numbers that make sense when appended to the two-digit proposed name, AND given your suggestion that we append a '0' for the future case that we might support multiple NICs (or at least being able to select one of several installed), I suggest the following:
```
/dev/wd0
/dev/ne0
/dev/3c0
```
Additional reasons include having some sort of continuity for for our users, who may not be privy to this discussion and wonder what the device names mean when executing `ls /dev`. Keeping the first two letters of the old name, and adding a "NIC card number" seems to make the most sense to me, while providing consistency as well as a path forward for your NIC development.

Mellvik, 2022-11-03:
Good, we'll go for that.Thank you @ghaerr.-M3. nov.

closed 2022-11-03 by Mellvik