ghaerr/elks #1453: Compilation problem

Vutshi, 2022-11-23:
**Description**
- I compile the current version of ELKS and get the following error:
```
console.c: At top level:
console.c:323:13: error: conflicting types for ‘std_char’
 static void std_char(register Console * C, int c)
             ^~~~~~~~
```

**Configuration**
- Version (commit): 6b9f5ad1c627863d0d8fa5530451a86b6730724c

**How to reproduce ?**
- Compile with the attached config file

**Additional information**

[my_config.txt](https://github.com/jbruchon/elks/files/10075527/my_config.txt)

Mellvik, 2022-11-23:
Hi Denis,

you're right, it appears bios-console has not been tested for some time and is out of sync with other driver level updates.

If you change line 81 of elks/arch/i86/drivers/char/console-bios.c from

	static void std_char(register Console *, char);

to

	static void std_char(register Console *, int);

your compile should complete fine.

We'll make sure the change gets into a PR shortly.

Thank you!

—M

ghaerr, 2022-11-23:
Thanks @Mellvik for solving this, I was just starting to look at it. Would you like to submit the PR or myself?

closed 2022-11-23 by ghaerr

Vutshi, 2022-11-23:
Thank you @Mellvik. That was quick )