ghaerr/elks #1460: Memory allocation and fsck

Mellvik, 2022-11-26:
This is more of a question than a problem report -

`fsck` fails on `malloc` if the network is running, works fine when the network is stopped. 
I'm not sufficiently into ELKS memory allocation to actually decipher the `memoinfo` output, so the question goes to @ghaerr: is this what you would expect given the 'memory status' from `meminfo`? 

It seems to me the segments at `d218` and `cb68` (2nd listing) should both be large enough to accommodate `fsck`'s requirement?

```
elks16# fsck /dev/hda4
ZONES: 65535 -- fsck: Unable to allocate main memory for zone count
elks16# net stop
Stopping network
elks16# fsck /dev/hda4
elks16# fsck /dev/hda1
elks16# meminfo
  HEAP   TYPE  SIZE    SEG   TYPE    SIZE  CNT
  c4ee   SEG     16   21b2   free     224    0
  c50a   INT      6
  c51c   INT      6
  c52e   INT      6
  c540   BUFH  1536
  cb4c   SEG     16   31c0   CSEG    7136    1
  cb68   SEG     16   21c0   BUF    65536    1
  cb84   INT      6
  cb96   TTY   1024
  cfa2   TTY     80
  cffe   SEG     16   337e   DSEG    5936    1
  d01a   SEG     16   34f1   free   11872    0
  d036   free    16
  d052   SEG     16   37d7   CSEG    5456    1
  d06e   TTY     80
  d0ca   TTY     80
  d126   SEG     16   392c   DSEG    3152    1
  d142   SEG     16   39f1   CSEG    3952    1
  d15e   SEG     16   3ae8   DSEG    2528    1
  d17a   SEG     16   3c72   CSEG   44064    1
  d196   SEG     16   4734   DSEG   24896    1
  d1b2   SEG     16   4d48   free  308096    0
  d1ce   SEG     16   3b86   free    3776    0
  d1ea   free 11798
  Heap/free   15134/11814 Total mem  486624
  Memory usage  475KB total,  159KB used,  316KB free
elks16# net start wd0
Starting networking on wd0
ktcp -b -p wd0 10.0.2.16 10.0.2.2 255.255.255.0
ktcp: ip 10.0.2.16, gateway 10.0.2.2, netmask 255.255.255.0
ktcp: /dev/wd0 mac 00.00.c0.bc.8f.4b mtu 1500
Starting daemons 'telnetd' 'ftpd -d' 
elks16# fsck /dev/hda1
ZONES: 65535 -- fsck: Unable to allocate main memory for zone count
elks16# meminfo
  HEAP   TYPE  SIZE    SEG   TYPE    SIZE  CNT
  c4ee   SEG     16   21b2   free     224    0
  c50a   INT      6
  c51c   INT      6
  c52e   INT      6
  c540   BUFH  1536
  cb4c   SEG     16   31c0   CSEG    7136    1
  cb68   SEG     16   21c0   BUF    65536    1
  cb84   INT      6
  cb96   TTY   1024
  cfa2   TTY     80
  cffe   SEG     16   337e   DSEG    5936    1
  d01a   SEG     16   34f1   DSEG    5664    1
  d036   SEG     16   5970   CSEG   18720    1
  d052   SEG     16   37d7   CSEG    5456    1
  d06e   TTY     80
  d0ca   TTY     80
  d126   SEG     16   392c   DSEG    3152    1
  d142   SEG     16   39f1   CSEG    4064    1
  d15e   SEG     16   535c   DSEG   14656    1
  d17a   SEG     16   3c72   CSEG   44064    1
  d196   SEG     16   4734   DSEG   24896    1
  d1b2   SEG     16   4d48   free   24896    0
  d1ce   SEG     16   5e02   CSEG   13968    1
  d1ea   SEG     16   6c28   DSEG   57952    1
  d206   INT      6
  d218   SEG     16   7a4e   free  123680    0
  d234   SEG     16   3aef   CSEG    3952    1
  d250   SEG     16   616b   free   43984    0
  d26c   SEG     16   3653   DSEG    2528    1
  d288   SEG     16   3be6   free    2240    0
  d2a4   SEG     16   56f0   free   10240    0
  d2c0   SEG     16   36f1   free    3680    0
  d2dc   free 11556
  Heap/free   15134/11556 Total mem  486624
  Memory usage  475KB total,  271KB used,  204KB free
elks16# 
```

ghaerr, 2022-11-26:
Hello @Mellvik,

Thanks for the report!

> fsck fails on malloc if the network is running, works fine when the network is stopped.

From memory, I believe `fsck` asks for 64K memory in one chunk, so this must be that there isn't space for that allocation, I would think.

> It seems to me the segments at d218 and cb68 (2nd listing) should both be large enough to accommodate fsck's requirement?

For `cb68`, which shows 65536 free, there could be a problem in that ELKS allocates a header prior to the memory itself for tracking purposes (and free, like libc malloc/free). Thus, that extra memory could be causing the failure. However, for `d218`, that shows a single free memory segment of 123680 bytes, which is certainly large enough for a 64K allocation.

Thus, it seems we may have a bug in `fmemalloc` or the kernel `seg_alloc`. I will have to look into this. The libc `fmemalloc` function takes a long argument of bytes and converts it into paragraphs before calling the kernel `fmemalloc` function, which takes the number of paragraphs requested and calls `seg_alloc`, the main kernel memory allocator.

If you'd like to continue probing yourself, perhaps some printf/printk in the following functions would help:
elks/arch/i86/mm/malloc.c:
```
// allocate memory for process, return segment
int sys_fmemalloc(int paras, unsigned short *pseg)
{
    segment_s *seg;
    int err;

    err = verify_area(VERIFY_WRITE, pseg, sizeof(*pseg));
    if (err)
        return err;
    seg = seg_alloc((segext_t)paras, SEG_FLAG_PROG);
    if (!seg)
        return -ENOMEM;
    seg->pid = current->pid;
    put_user(seg->base, pseg);
    return 0;
}
```
libc/malloc/fmemalloc.c:
```
void __far *fmemalloc(unsigned long size)
{
    unsigned short seg;
    unsigned int paras = (unsigned int)((size + 15) >> 4);

    if (_fmemalloc(paras, &seg))
        return 0;
    return _MK_FP(seg, 0);
}
```
(I also just noticed in the above code that `errno` isn't set properly on error, but that isn't the problem you're having).

Thank you!

Mellvik, 2022-11-28:
Further investigation explains the issue. 

First, the `meminfo` output is a snapshot reliable only for the duration of the program itself. After it exits, the memory it uses is freed and the picture is (slightly) different. Sufficiently so to have me confused for a while.

Second, `fsck` is set up with a 64k heap which is allocated before the `fmemalloc` that fails, so what is required is really 2x64k blocks, which explains the failure reported above.

That said, some experimentation indicates that setting the `fsck` heap to 23000 is enough for it to run reliably on a max size `minix` FS. Unless there is something I'm missing here, it may be a good idea to adjust that.

-M

Mellvik, 2022-11-28:
Hi @ghaerr,

> I agree with your findings and conclusions @Mellvik. At some point it could make sense to move the memory tracker into library code so that it could be used to debug a running application...

That would be really helpful!
> 
> > experimentation indicates that setting the fsck heap to 23000 is enough for it to run reliably on a max size minix FS
> 
> Good finding. It was increased originally to max heap, but then not decreased when the far malloc code was added. Go ahead and submit a change with the heap set to 23000, that will help a lot on tight memory systems.
> 

Coming up

Thanks.

-M

closed 2022-11-28 by Mellvik