ghaerr/elks #1470: Floppy boot problem

Mellvik, 2022-12-04:
As of recently (no bisect yet), floppies (1440) generated from the ELKS build process fail to boot on the Compaq (386/20):

```
Direct console, scan kbd 80x25 emulating ANSI (3 virtual consoles)
ttyS0 at 0x3f8, irq 4 is a 16450
ttyS1 at 0x2f8, irq 3 is a 16550A
ttyS2 at 0x3e8, irq 5 is a 16550A
xms: A20 was off now on, using unreal mode, 1024 xms buffers, 1048576 ram
eth: ne0 at 0x300, irq 12 not found
eth: wd0 at 0x280, irq 2, ram 0xce00, (wd8003) MAC 00:00:C0:4D:F6:13, flags 0x80
eth: 3c0 at 0x330, irq 11 not found
bioshd: hda BIOS CHS 610,16,63
bioshd: hda  IDE CHS 993,16,63
/dev/hda: 993 cylinders, 16 heads, 63 sectors = 488.7 Mb
/dev/fd0: 80 cylinders, 2 heads, 18 sectors = 1440.0 kb
/dev/fd1: 80 cylinders, 2 heads, 15 sectors = 1200.0 kb
Partitions: hda:(0,1000944) 
```
At this point it stops and the floppy IO indicator keeps spinning in the upper right corner of the console.
This happens ONLY on the Compaq, not on the AMI386SX System.

When reconfiguring w/o XMS support compiled in, it boots fine.
I have a several months old boot floppy that boots file win XMS enabled.

From looking at the code and some testing it seems the call to `check_partitions` from `setup_dev` (line 362 in genhd.c) never returns.

@ghaerr - any hints as to where to look for this one (or how far back to start the bisect)? 

--M

Mellvik, 2022-12-04:
BTW - I strongly suggest we put the `ELKS built from` message back into the `/etc/motd` file.

-M

ghaerr, 2022-12-05:
Hi @Mellvik,

IIRC, XMS will work on the Compaq only if the CONFIG_FS_XMS_INT15 option is set. That is, the Compaq doesn't seem to want to work using unreal mode to directly access the XMS buffers, but it will work if the INT 15h extended memory BIOS calls are used. I think this is because the Compaq BIOS actually uses protected mode and ends up resetting the cached unreal mode descriptor limits, so we wrote the INT 15 code to be used instead. (The INT 15h code should work on all 386+ BIOSes also).

> I strongly suggest we put the ELKS built from message back into the /etc/motd file.

Ok, let me work on getting the commit message built into a kernel printk, rather than /etc/motd, as it saves 1K disk, which is nice on 360k floppies. That should still work to show which version the kernel was compiled under.

Thank you!

ghaerr, 2022-12-28:
Hello @Mellvik,

> I strongly suggest we put the ELKS built from message back into the /etc/motd file.

I just realized that we added the commit into `uname -a`. Or, you can use `uname -v` to see just the commit message and build time.

Thank you!

Mellvik, 2022-12-28:
Thank you @ghaerr -

you, I had forgotten about that and it's useful. 

Still, and I realize that this is different from even the previous way: The cases when I've needed the commit id the most are the ones when something stops before boot has completed, before init has started.

So the ultimate preference would be (for example) to add the commit ID to the line
PC/AT class machine, syscaps 0xff, 640K base ram.
… or even earlier in the boot process. 
OTOH there is the consideration that the ID is printed late enough in the process to not scroll off the screen (which is a non issue in my case because there are too many devices, and I need to switch to the serial console anyway).

-M

ghaerr, 2022-12-28:
> OTOH there is the consideration that the ID is printed late enough in the process to not scroll off the screen

You can put `uname -v` in /etc/profile for that. I've added an earlier build display to my TODO list, which I plan to get to when we next clean up the (large) display of boot messages to try to fit on just one screen.

Mellvik, 2022-12-29:
Thank you @ghaerr,
It's an interesting discussion because the boot messages - every single one - are so important for development and problem resolution. 

Looking at the messages there is not a single one I'd like to get rid of. And - realistically, unless we're talking a really bare bones system, there will be more than 24 lines (including the final getty login) anyway.

I realize my setups may be extreme - with 3-4 serial lines, XMS, 2 disks, two floppy drives, 3 ethernet interfaces plus (and easy to forget) the output from fsck'ing 4 or more file systems, net startup and getty. Leaving out net autostart and sending the console veto to tty2 has been helpful in some cases.
OTOH - since I (almost) always use serial, it's not much of a problem in my case.

Ignoring my rather odd setup, the question remains: How to get a workable solution for a plain 24line console system? 
Adding a 'terse boot messages' option to menuconfig and/or bootopts?
Add a temporary memory buffer to hold the printk's for a short period after boot, to be displayed with a command, say 'dmesg'? Possibly flushing the content to a /tmp/bootlog file or the like at startup?
Let boot messages overflow to tty2, tty3, making them easily available? (actually, sending the first 24 lines to both tty1 and tty2 may be quite simple and 48 lines should be enough in all cases except when fsck finds lots of errors). 

-M