ghaerr/elks #1482: NEC PC-98 enhancements to ELKS

tyama501, 2022-12-17:
Continued conversation from #1325

tyama501, 2022-12-17:
Hello @ghaerr ,

It seems that the problem I wrote in https://github.com/jbruchon/elks/issues/1325#issuecomment-1301873959
is not the problem for particular system but rather the serial cross cable is connected or not. 

>I find another problem that I cannot get to "login" prompt if I enabled the serial console or runlevel 3 in inittab for PC-9801RX.
>It gets hang-up before the login.
>Probably the serial driver issue.

It hangs up on PC-9801BX too at the following function if I remove the cable with inittab:3.

serial-pc98.c
static int rs_write(struct tty *tty)
...
    while (tty->outq.len > 0) {
        /* Wait until transmitter hold buffer empty */
        while (!(inb(port->io + UART_LSR) & UART_LSR_THRE))
                ;
        outb((char)tty_outproc(tty), port->io + UART_TX);
        i++;
    }

[image]

I need to think a workaround.
Thank you.

tyama501, 2022-12-17:
One idea is quit using while loop and use for-loop to check empty but the limit may depends on CPU speed.

Is there a good way to timeout checking and abort with couple of seconds?

ghaerr, 2022-12-17:
Hello @tyama501,

> It hangs up on PC-9801BX too at the following function if I remove the cable with inittab:3.

I am wondering if the PC-98 BIOS or serial chip is somehow holding transmission when the cable is disconnected. You might check any resources on that.

Another idea is to test a different bit in the UART - instead of checking the Transmit Holding Register (UART_LSR_THRE), check the Transmitter Empty bit (UART_LSR_TEMT).

Does your serial cross-cable require that DTR be held high? That is, certain systems that are wired for hardware flow control requires certain pins to be high before the UART will transmit. Cables can "cheat" this by wiring (IIRC) DSR to DTR on the computer side, rather than sending DTR from the receiver. You can read up on this on PC serial cables if not familiar. Currently, ELKS doesn't use HW flow control so the pins should be held high and not sent from the other system.

> One idea is quit using while loop and use for-loop to check empty but the limit may depends on CPU speed.

It seems using a for loop would be exactly the same in this case.

> Is there a good way to timeout checking and abort with couple of seconds?

We could do this, but it is a bad idea, as the serial data would be lost with timeout. Also, there would be quite a few timeouts as each character of the login: sequence was timed out, so I don't recommend that.

Are you certain the hang is in:
```
while (!(inb(port->io + UART_LSR) & UART_LSR_THRE))
```
and not in:
```
outb((char)tty_outproc(tty), port->io + UART_TX);
```

Thank you!

tyama501, 2022-12-17:
Hello @ghaerr ,

Thank you for the comments.

> serial chip is somehow holding transmission when the cable is disconnected

Yes, I think so too.

> instead of checking the Transmit Holding Register (UART_LSR_THRE), check the Transmitter Empty bit (UART_LSR_TEMT).

I don't know why but UART_LSR_TEMT is not working on my system.
So, I am using UART_LSR_THRE.

> Cables can "cheat" this by wiring (IIRC) DSR to DTR on the computer side, rather than sending DTR from the receiver. 

I am using a cable with connecting like this.
D-sub 9pin is PC/AT side and D-sub 25pin is PC-98 side.
Pin 7/8 for PC/AT and Pin4/5 for PC-98 are RTS/CTS so it is cheating cable.
They are also connected to Pin1 of PC-AT (CD pin) and Pin8 of PC-98 (DCD).

[image]

I will continue to check setting of the serial chip if I can ignore these pins.

Thank you.

tyama501, 2022-12-17:
Also PC/AT Pin 4, DTR is connected to PC-98 Pin6, DSR and
PC-98 Pin 20, DTR is connected to PC/AT Pin6, DSR.

Mellvik, 2022-12-18:
Hi @tyama501,Your cable picture caught my eye. This is a 'direct' null modem cable. That would work if your setup had a DTE (PC or terminal) and a DCE (modem or other communication device).  From what I can tell, you are connecting two PCs. They are both DCE devices and you need a crossover nullmodem cable in order for them to communicate.Currently it may be the case that the receive line on the AT is connected to the receive line on the PC98 and similar with the send lines. No data obviously.Since you have a null modem setup, all you have to do is (probably) to switch RX and TX at one end. Take a look at this one for reference:Crossover or "Null Modem" vs. Straight Through Serial Cabledecisivetactics.comI'm assuming you don't have a RS232 tester. Those are extremely valuable in settings like this and cost little, saving hours when debugging serial problems. -M17. des.

tyama501, 2022-12-18:
Hello @Mellvik ,

Thank you for the comments!

Actually, it is working when the cable is connected between PC/AT and PC-98, with RXD - TXD crossed.
Now, the problem is that ELKS cannot boot when the cable is disconnected.

The chip 8251 is waiting CTS and so that the buffer will not be empty.
I couldn't find the setting to ignore CTS so far.

Thank you!

Mellvik, 2022-12-18:
OK, I get it. Now here's an idea: You may have a hardware failure/error.If the system requires CTS in order to boot, there MUST be a pullup resistor on the CTS line. If the resistor is burned or broken, you'll get the symptoms you're seeing.To check, measure the resistance between the CTS line and +12V, you should read something between 3k and 10k ohms. If it's higher than that, this could well be your problem.It's also possible that the 1489 quad input buffer is partly broken and needs to be replaced. The fact that you're reading data ok indicates that it's OK though.And for good measure, check that the -12V power is OK on the machine. The only 'user' of -12V on a PC is the RS232 port(s) (and outgoing signals only so it would not directly affect the CTS line).-M18. des.

tyama501, 2022-12-18:
Thank you for the idea @Mellvik ,

I will check the hardware if I couldn't find the I/O setting with further investigation.

ghaerr, 2022-12-18:
Hello @tyama501 and @Mellvik,

Nice job @Mellvik on the DCE and DTE description. That rung a few bells from the past :)

It seems to me, if there isn't a setting available in the PC-98 UART to turn off hardware flow control on the 8251, that just rewiring the cable to connect pins 4,5 & 6 and possibly 20 on the PC-98 end should keep CTS happy and allow data to flow. 

I used to do this all the time in years past but can't remember the exact pin numbers for CTS/DTR, etc. It seems that @tyama501's cable is a crossover cable WITH hardware flow control. Pins 2 and 3 will want to stay cross-connected, and the other control pins all connected together possibly on each side of the cable in order to circumvent hardware flow control.

I will check on the 8251 data sheet to see if hardware flow control can be turned off later today.

Thank you!

Mellvik, 2022-12-18:
Thank you @ghaerr,Like you I've found the memory bank in which all the practical experiences with the 8251 severely lacking after 40 years and lack of refresh pulses.From reading the spec sheet it seems that CTS cannot be programmed, it's a 'hardwired enabler'. Its absence will prevent data from being transmitted, but should not prevent the device from being initialized/opened. So ELKS should not hang in the serial driver even if a getty or some other process is trying to write to the line. It seems to me we're still missing something here.-M

Mellvik, 2022-12-18:
One more thing on this issue @tyama501, @ghaerr:I've browsed half a dozen serial interface schematics this afternoonand found that a pullup on CTS is the exception, not the rule.So it's not likely that the presence or absence of a CTS pullup, as I suggested, is involved in the boot hang situation. OTOH a broken or damaged buffer chip may be involved.-M18. des.

ghaerr, 2022-12-19:
Hello @tyama501 and @Mellvik,

> So ELKS should not hang in the serial driver even if a getty or some other process is trying to write to the line. 

Thank you @Mellvik, I now finally understand the real problem. What is happening is that the kernel busy-loops inside the serial driver waiting for the UART transmitter buffer empty, and doesn't sleep the calling task. Thus, nothing else runs.

The reason we haven't seen this before on the IBM PC is that it uses the [8250 UART (see Modem Control Register)](https://en.wikibooks.org/wiki/Serial_Programming/8250_UART_Programming#Modem_Control_Register) for software-based hardware flow control handling, while the PC-98 uses the [8251 UART](https://www.jameco.com/Jameco/Products/ProdDS/52370INT.pdf) which does not have that register, requiring the CTS pin be low or it won't transmit.

The 8250 UART in the IBM PC will transmit a byte whenever data is passed to the chip and the transmitter buffer is empty. The PC-98 8251 however, uses an external CTS (Clear to Send) pin that must be low or the chip won't transmit the byte just passed to the UART. Basically, the IBM PC allows for software-driven hardware flow control, while the PC-98 always uses hardware flow control.

@tyama501, solving this problem for the PC-98's 8251 is less than great: see page 2-14 of the 8251 data sheet, Figure 13 Status Read Format. It says that TxRdy bit (0x01=UART_LSR_THRE) is not supposed to be affected by CTS low, but apparently is. It seems from what you're saying that the kernel busy loops waiting for TxRdy and never gets it unless the cable is plugged in.

A simple fix would be to drop characters if TxRdy not set for some set period of time. Actually, this is kind of  what the IBM PC driver does - it sends characters to the UART regardless of whether there is a cable connected, and they are lost if not connected, but still the code waits for Transmitter Buffer Empty, which always occurs after transmitting the previous character. In the PC-98 case, we might just "time out", drop the output and return:

In serial-pc98.c:
```
/* serial write - busy loops until transmit buffer available */
static int rs_write(struct tty *tty)
{
    struct serial_info *port = &ports[tty->minor - RS_MINOR_OFFSET];
    int i = 0;

    while (tty->outq.len > 0) {
       volatile int timeout = 5000; // set some timeout value determined by testing on real PC-98 hardware
        /* Wait until transmitter hold buffer empty or timeout */
        while (!(inb(port->io + UART_LSR) & UART_LSR_THRE)) {
                if (--timeout == 0) // <---- add these two lines
                   return i;
       } 
       outb((char)tty_outproc(tty), port->io + UART_TX);
        i++;
    }
    return i;
}
```
The initial value of `timeout` should be the number of PC-98 while loop cycles it takes to send two characters at, say, 300 baud. This will then cause the serial write to timeout based on CTS not being low, instead of actually waiting for a couple characters to be output. If not calculating the value precisely, test by sending PC-98 output at 300 baud with the cable connected, and make sure no characters are dropped, while also being low enough that the wait time is not excessive when the cable is not connected.

Try that out and see whether this fixes your problem.

Thank you!

Mellvik, 2022-12-19:
Thank you @ghaerr -
the devil is in the details - as always, and the difference between the 8250 and 8251 certainly didn't catch my attention. Good catch. Which admittedly sent me back to the (old) idea to make serial output interrupt driven (i.e. buffered) In which case we would not have had this problem.

Anyway, I keep coming back to the pull-up which would also solve the problem without any software mods. It does require some soldering experience though - and a 10k resistor at hand. See attached schematic.

-M

[Deluxe RS-232 Program Pak Schematic (Tandy).pdf](https://github.com/jbruchon/elks/files/10258797/Deluxe.RS-232.Program.Pak.Schematic.Tandy.pdf)

tyama501, 2022-12-19:
Hello @ghaerr , @Mellvik ,

Wow, thank you for the precise investigation.

@ghaerr ,
I will try the timeout codes.
The return value of rs_write is not used in ntty.c,
is that why i++ is not needed when the timeout to drop the character?

@Mellvik ,
Thank you for the schematic.
It is ok for me to pull-up or just connecting the cable for permanently,
but I will try to modify the driver to avoid some users get trouble :)

Thank you!

tyama501, 2022-12-19:
BTW, the last pc-9821 in year 2000 had Celeron 433MHz, but I won't consider that for the timeout...

tyama501, 2022-12-24:
Hmmm,

I tried 
if (--timeout == 0)
   return i;

and
if (--timeout == 0)
   break;

but both didn't solve the problem...

tyama501, 2022-12-24:
Oh, just setting inittab 3 is OK with the break. (solved the problem.)

If I set console= in bootopt then it got stuck again.

ghaerr, 2022-12-24:
Hello @tyama501,

> If I set console= in bootopt then it got stuck again.

This is because the serial console uses a different routine for output, which is also hanging:
```
/* note: this function will be called prior to serial_init if serial console set*/
void rs_conout(dev_t dev, int c)
{
    struct serial_info *sp = &ports[MINOR(dev) - RS_MINOR_OFFSET];

    while (!(inb(sp->io + UART_LSR) & UART_LSR_THRE))
        continue;
    outb(c, sp->io + UART_TX);
}
```
You will need to add the timeout there as well.

Continue testing with the above fix and then go ahead and send a PR, thanks!

tyama501, 2022-12-24:
Oh, I see thanks!

tyama501, 2022-12-25:
Hello @ghaerr ,

I tested the various timeout with PC-9801BX, intel 486SX 20MHz, Baud rate 300.

cable disconnect test
timeout = 1000 : initab3 only OK, console=ttyS0,300 3 OK
timeout = 10000 : initab3 only OK, console=ttyS0,300 3 OK
timeout = 30000 : initab3 only OK, console=ttyS0,300 3 slow but OK

cable connected test, transmit to Windows PC
timeout = 1000 : console=ttyS0,300 3 NG
timeout = 10000 : console=ttyS0,300 3 NG
timeout = 30000 : initab3 only OK, console=ttyS0,300 3 OK

It seems that timout = 30000 is good for now.
We may support 30 times faster CPU If the baud rate is 30 times faster like 9600bps and it would be enough. 

I will make PR after merging upstream.

Thank you.

tyama501, 2022-12-25:
Additional test. For PC-9801RX, intel 286 12MHz
cable disconnect
timeout = 30000 : console=ttyS0,9600 3 OK but very slow to timeout. It takes about 2 minutes to boot...

cable connected test, transmit to Windows PC
timeout = 30000 : initab3 only OK, console=ttyS0,9600 3 OK

tyama501, 2022-12-25:
Hmmm For PC-9801UV21, V30 10MHz, it doesn't work connected nor disconnected.
Maybe this is the first time I try this machine.

It seems that the miniterm 9600bps works the communication,
but the console, getty won't work. That's strange.

tyama501, 2022-12-25:
> but the console, getty won't work. That's strange.

If I slowdown to B=1200bps the getty with inittab3 worked.
I may be satisfied with this.

If I set console on bootopt not shown any.

tyama501, 2022-12-25:
Hello @ghaerr , @Mellvik ,

I posted the movie to login PC-9801UV21 from Windows TeraTerm 1200bps,

https://twitter.com/tyama50/status/1606967893958496257

Thank you for your assistance :)

ghaerr, 2022-12-25:
Hello @tyama501,

Thanks for your testing. I can see it's tough to pick a value that will work well for different CPUs and baud rates. We can use 30000 for now, since it works though isn't ideal. 

For longer term, if we move to interrupt-driven serial output, that won't help your system if the CTS pin is not low (the getty or login program would hang until CTS, but also so would the serial console). So we need to decide if we should continue to discard output, like we do now, when the cable is not connected. If we allow output to be discarded, another option would be to determine the actual delay time per character for each baud rate, and then sleep the task by setting a kernel timer (similar to how it's done with BIOS polling keyboard every 8ms, see elks/arch/i86/drivers/char/kbd-poll-pc98.c) to delay before trying again and then discarding the character.

Thank you!

tyama501, 2023-02-04:
Hello @ghaerr ,

The elksmoria worked with the real PC-98 but clock -w make crash the PC again.
Has the clock command or clib changed recently?

tyama501, 2023-02-04:
I merged the latest commit and rebuilt with the new tool chain but clock -w still crash PC with strange display.

tyama501, 2023-02-06:
The problem can reproduce with DosBox-X too.
The date showing "2002" because clock -s is not working.

[image]
 

> Hello @ghaerr ,
> 
> The elksmoria worked with the real PC-98 but clock -w make crash the PC again. Has the clock command or clib changed recently?

ghaerr, 2023-02-06:
Hello @tyama501,

> I merged the latest commit and rebuilt with the new tool chain but clock -w still crash PC with strange display.
> The date showing "2002" because clock -s is not working.

Hmmm, let me look into whether the clock source has changed (I think not), or whether this might have something to do with the new toolchain.

I am wondering whether perhaps the clock command is being built for IBM PC rather than PC-98. I don't recall having seen anything changed to do with that though.

Thank you!

tyama501, 2023-02-06:
Hello @ghaerr ,

Yes, I saw this crash before.
I assume it is using IBM I/O... 
I don't know why.

ghaerr, 2023-02-06:
The clock source has not changed for over a year. There have been what should be immaterial changes to elks/include/arch/io.h to the `inb_p` and `outb_p` macros, we can check and look at the code generated for clock.c to see. The first check should be to ensure that CONFIG_ARCH_PC98 is set, and we can do that by disassembling the elkscmd/sys_utils/clock.o file using `ia16-elf-objdump -D -r -Mi8086 clock.o`. The output should include something like this for the important PC98 functions.

Here is cmos_read, which is for both IBM and PC-98. It looks correct (compare with source in clock.c):
```
00000045 <cmos_read>:
  45:   89 e3                   mov    %sp,%bx
  47:   fa                      cli
  48:   8a 47 02                mov    0x2(%bx),%al
  4b:   0c 80                   or     $0x80,%al
  4d:   ba 70 00                mov    $0x70,%dx
  50:   ee                      out    %al,(%dx)
  51:   e6 80                   out    %al,$0x80
  53:   ba 71 00                mov    $0x71,%dx 
  56:   ec                      in     (%dx),%al
  57:   e6 80                   out    %al,$0x80
  59:   fb                      sti
  5a:   c3                      ret
```

Here is the code for read_calendar, specific for PC-98. It seems correct:
```
000000b5 <read_calendar>:
  b5:   06                      push   %es
  b6:   55                      push   %bp
  b7:   89 e5                   mov    %sp,%bp
  b9:   8b 5e 08                mov    0x8(%bp),%bx
  bc:   8b 46 06                mov    0x6(%bp),%ax
  bf:   8e c0                   mov    %ax,%es
  c1:   b4 00                   mov    $0x0,%ah
  c3:   cd 1c                   int    $0x1c
  c5:   5d                      pop    %bp
  c6:   07                      pop    %es
  c7:   c3                      ret
```
The size of `clock` built for PC-98 on my system is 4064. Is that what you have on your PC-98 image in /bin/clock?

ghaerr, 2023-02-07:
Hello @tyama501,

> So, the CONFIG_ARCH_PC98 has not been applied.

Yes, that's correct. It looks like I broke it with commit 41c6ca70d41f007fe1b211bbdb6be6e9d8f2de5f on Dec 31, 2022. I will post a PR that should fix the problem.

Thank you!

tyama501, 2023-02-08:
Hello @ghaerr ,

Thank you.

It seems that nx driver in the image downloaded from here is still not working.
nxworld and nxterm go black screen.
https://github.com/jbruchon/elks/actions/runs/4116985420

It works in the image I built myself.

Something is different in makefile?

ghaerr, 2023-02-08:
Hello @tyama501,

Sorry for your trouble with our PC-98 automatic build!

I'm pretty sure the problem is in the GitHub main.yml workflow, where a few .o files are erased for quick rebuilding in the PC-98 pass using build.sh:
```
# Build PC-98 kernel, some user land files and image
if [ "$1" = "auto" ]; then
    echo "Building PC-98 image..."
    cp pc98-1232.config .config
    make kclean || clean_exit 9
    rm elkscmd/sys_utils/clock.o
    rm elkscmd/basic/*.o
    rm elkscmd/sys_utils/ps.o
    rm elkscmd/nano-X/*.o <--- this should be elkscmd/nano-X/*/*.o
    make -j1 || clean_exit 10
fi
```

It seems this has likely been the case for quite some time. I'll post a PR to fix it.

Thank you!

closed 2023-02-08 by ghaerr

tyama501, 2023-02-09:
Thank you @ghaerr ,

It worked.