ghaerr/elks #1502: NX demos fail to start

toncho11, 2022-12-31:
I tried all of the NX demos and it says:

- no mouse
- could not open graphics 

Is it possible that the lack of mouse is actually causing the graphics lib to fail?
I am testing using a 1440 image on IBM 5160 with a VGA card.
I do not have a serial card on this computer.

Also should nxterm require a serial mouse?

toncho11, 2022-12-31:
Usually the nxdemos require just a VGA card and enough RAM.
There should be at least one demo that is able to run whiteout requiring a mouse.

toncho11, 2022-12-31:
I am using some old image of ELKS: 5b5567f2 24/05/2022

ghaerr, 2022-12-31:
> Is it possible that the lack of mouse is actually causing the graphics lib to fail?

NX is not so much a graphics library but instead a windowing environment, which requires a mouse. When the serial open fails in the mouse driver because of no serial port, the failure return causes NX to exit.

The NX demos can be compiled to not use a mouse, as they are on the PC-98 (see nanox/Makefile::drivers/mou_null.o). This is done using the "null" mouse driver.

I'm not sure if it would be a good idea to allow NX to run without a mouse, as users might get confused as to how to exit the program (ESC usually works). Any error messages also would be erased by the graphics clear screen, thus adding to confusion, IMO.

Nano-X itself is pretty much designed to require a mouse, so currently the null mouse driver must be compiled in for systems without a serial port. As stated above, I hesitate a bit to think it a good idea to allow the serial mouse driver to return OK with no mouse.

toncho11, 2022-12-31:
I just tested on latest build. 
I had to insert my XT-IDE to avoid the network probing bug ... otherwise it halts ... 

Only the nxlandmine is available in the 1440 now.
Still there should be some program that does not require mouse in order to test that graphics is available on ELKS. 
Maybe calling BASIC with some program ... but currently that is not available. 

I think there can be a simple message "Mouse not detected. Expect problems! Do you wish to continue (Y/N)". This way we keep testing NX even on computers without a mouse.

tyama501, 2022-12-31:
Oh, I thought nxterm and nxworld also available.

nxclock and nxdemo need to rebuild.

tyama501, 2022-12-31:
Sorry, yes, the nxlandmine is the only program that need mouse.

ghaerr, 2022-12-31:
Looking at the mouse driver, one can set the mouse port using `export MOUSE_PORT=/dev/ttyS?`. One should be able to set this to /dev/null, but I just tested this and it doesn't work. I'll look into finding a way that NX programs can be run without a mouse by allowing the serial driver to become the null driver using MOUSE_PORT=none or something like that.

> Oh, I thought nxterm and nxworld also available.
> nxclock and nxdemo need to rebuild.

nxlandmine is available on 1200k and 1440k floppies, and nxworld and nxterm are available when CONFIG_APP_NANOX is set. Nxclock was removed from CONFIG_APP_NANOX since the 1440k floppies ran out of space. It should probably be added back under CONFIG_APP_OTHER for HD images.

toncho11, 2022-12-31:
Yes, some graphics demos are always useful to show off ELKS, but you don't always have a serial port available.
HD images should have every program that is in working condition.

Thank you @ghaerr !

toncho11, 2023-01-01:
Happy New Year!

A simple Y/N message "Mouse not detected, continue (Y/N)?" should also do the job, no?
Or as you suggested "Mouse not detected, use MOUSE_PORT=none".

ghaerr, 2023-01-01:
Hello @toncho11, Happy New Year! 

> A simple Y/N message "Mouse not detected, continue (Y/N)?" should also do the job, no?
Or as you suggested "Mouse not detected, use MOUSE_PORT=none".

It gets complicated fast reading from the console in Nano-X since the keyboard is in raw mode, etc, so I'd rather not ask questions at NX startup. I like your idea of at least changing the mouse open error message to "Mouse not detected, use MOUSE_PORT=none". I'll put together something that should work for all NX binaries for your testing.

toncho11, 2023-01-02:
You can add "[FATAL] No mouse detected, use ..." because it is not clear that it is a blocking issue. I thought the graphics is the blocking issue.

ghaerr, 2023-01-02:
Hello @toncho11,

> because it is not clear that it is a blocking issue. I thought the graphics is the blocking issue.

Have you tested this? There is now a two-line error message, where the first part states there is no mouse found on /dev/ttyS1 serial port, and the second line gives a suggested workaround. I removed the "Cannot open graphics" error message, as it was inaccurate.

Thank you!

toncho11, 2023-01-02:
Thank you @ghaerr.
A default built 1440 has only nxlandmine now?

ghaerr, 2023-01-02:
> A default build 1440 has only nxlandmine now? 

Yes - unfortunately the 1440k image had run out of space to include more stuff. The CONFIG_APP_NANOX (which is not set by default on 1440k) adds nxterm and nxworld.

However, with some recent enhancements, there is more space available on the 1440k image. I've added an item on my TODO for consideration to add more NX demos, versus other programs, for the final 0.7.0 distribution. We can discuss that elsewhere when the time comes.

closed 2023-01-06 by ghaerr