ghaerr/elks #1545: NEC PC-98 enhancements to ELKS

tyama501, 2023-02-25:
Continued conversation from https://github.com/jbruchon/elks/issues/1482

tyama501, 2023-02-25:
Hello @ghaerr ,

I started making PC-98 bus mouse driver for Nano-X since the serial mouse is not general for PC-98.
Here is the initial draft but it seems not working at all yet. I don't see any change to the mouse cursor.
https://github.com/tyama501/elks/commit/1c78749e02de5620a5650dfa508ab96e6bfc00ab

PC-98 bus mouse is attached to 8255A and I confirmed with the "ELKS Basic" these I/O's are working on the emulator.
I assume *dx and *dy in MOU Read is the differential value including negative value. Is that correct?

Thank you.

ghaerr, 2023-02-25:
Hello @tyama501,

Nice! I don't know whether the mouse driver itself works internally, but we're going to need to make some other changes in order to get this to work with Nano-X. The basic problem is that all the other mouse drivers are implemented by opening a file (usually a serial port), of which the file descriptor is passed back to Nano-X. In your case, there is no file descriptor, so we can't 'return 0' as you're doing in the Open routine, as that then makes the main server call `select` on that file descriptor, which in this case is actually `stdin`.

Instead, you will need to implement the Poll entry point, which can then be polled directly using an ifdef PC98 in nanox/srvmain.c. If the Poll returns true, a mouse event will be read by calling Read, which will insert the mouse event into the Nano-X queue, and things will start working.

Take a look at drivers/mou_dos.c to see how a possible Read and Poll could be implemented. Note in that driver, Poll always returns 1, and the Read determines whether any input is present. Your driver will need to implement Poll to only return 1 if a new mouse changed event is present, and Read can then return it.

The Open routine should return -2, which will (falsely) indicate that a mouse isn't present, but that will allow the nanox/srvmain.c::GsSelect() code to work, as you will see.

Then, we can change nanox/srvmain.c by doing something like this:
```
#if UNIX
void
GsSelect(GR_TIMEOUT timeout)
{
    fd_set  rfds;
    int     e;
    int setsize = 0;
    struct timeval to;

#if CONFIG_ARCH_PC98   // <--- new code here
    if (mousedev.Poll()) {
        GsCheckMouseEvent();
        return;
    }
#endif
    /* Set up the FDs for use in the main select(): */
    ...
```

The above change will poll the mouse driver instead of calling `select()`, which waits for keyboard or timeout. As I think of it, we may need to change the timeout so that it never hangs, allowing the mouse to be continually polled, since a mouse event won't cause select() to return. The following code could be added for that:

```
    /* convert wait timeout to timeval struct*/
#ifdef CONFIG_ARCH_PC98
    if (timeout == GR_TIMEOUT_BLOCK) // <-- add this line
          timeout = 10;
#endif
    if (timeout == GR_TIMEOUT_POLL) {
        to.tv_sec = 0;
        to.tv_usec = 0;
    } else {
        to.tv_sec = timeout / 1000;
        to.tv_usec = (timeout % 1000) * 1000;
    }
```

This could get a little tricky running `nxtetris` because that program depends on timeouts in select(). But we'll worry about that after you get the mouse working and test it with the other Nano-X programs.

Thank you!

tyama501, 2023-02-26:
Hello @ghaerr ,

Thank you, it started to work with this modification.
https://github.com/tyama501/elks/compare/1c78749...69d7d7e

I needed to include autoconf.h to work "if CONFIG_ARCH_PC98".
I used return -3 for Open to distinguish from "none".
Also, I removed x_before, y_before since the counter is cleared every time.

I thought MOU_Poll should be ok to always return 1 since the I/O routine never blocks,
and I haven't implemented the timeout yet.

Now it seems I cannot do anything except mouse curser move, so I may be wrong.
Here is a video for nxterm.
I cannot even exit with ESC key.

https://user-images.githubusercontent.com/61556504/221420871-1d52c855-b821-4cdd-8529-1417f79a0855.mp4

ghaerr, 2023-02-26:
Hello @tyama501,

Nice job getting the mouse working! :)

> I thought MOU_Poll should be ok to always return 1 since the I/O routine never blocks
> Now it seems I cannot do anything except mouse curser move, so I may be wrong.
> I cannot even exit with ESC key.

The problem is that, since MOU_Poll always returns 1, then the following code doesn't ever enter the `select` code to read the keyboard or handle timeouts:
```
#if CONFIG_ARCH_PC98
	if (mousedev.Poll()) {
		GsCheckMouseEvent();
		return; // <-- any mouse event will return before select which is a problem if Poll always returns 1
	}
#endif
```
You could try removing the `return` above, but that would mean that select would hang until timeout or a keypress is made, and moving the mouse would not work during that wait. That's why I suggested that you add the code `if (timeout == 0) timeout = 10;` before the timeout calculations - so that after mouse check it also enters select loop to read keyboard or process timeout. (The timeout should not be processed if there is a mouse event though, so this gets complicated and doesn't really work well).

The other approach would be to do the mouse calculations in Poll (not Read), and return 0 if the mouse location has not moved. If it has, store the result in static variables and then return 1, and Read will then return those values. This will still have the problem of hanging in select eventually, so you probably need to add the `if (timeout == 0)` code anyways.

Thank you!

ghaerr, 2023-02-26:
Another problem with Poll always returning 1 is that means Nano-X will always read a mouse event, even if the mouse hasn't changed. That ends up propagating all the way to the nxtetris or other application, which isn't really great. So I think we should probably have Poll return 0 until the mouse status actually changes, and have that change read once in Read().

tyama501, 2023-03-04:
Hello @ghaerr ,

The mouse worked with this modification for MOU_Poll, MOU_Read and the timeout :)
https://github.com/tyama501/elks/commit/d9d68e64c78a9dfd8770abda47df33c2838d6fe9 

Here is the nxtetris video.
https://user-images.githubusercontent.com/61556504/222924899-433b8f34-943b-4b5b-848f-54be57865e8c.mp4

I will test on real PC-98.

Thank you!

ghaerr, 2023-03-04:
Hello @tyama501,

Very nice job implementing the mouse from scratch on PC-98! :) Please send a PR when you are ready, thank you!

tyama501, 2023-03-10:
Hello @ghaerr ,

I am still on the emulator, but it seems that nxworld is something wrong with the mouse.

It draws many square lines when moving the mouse without pressing the button.
Is this right operation?
Also, I can zoom in by pressing left button, but I don't know how to zoom out.

https://user-images.githubusercontent.com/61556504/224380249-19ea4114-aaea-4367-8294-67f4ce3d5d47.mp4

There might be graphic driver issue.

nxlandmine works fine.

Thank you.

tyama501, 2023-03-10:
I tried DOSBox-X with IBM image but it seems that no serial mouse...

ghaerr, 2023-03-10:
Hello @tyama501,

> It draws many square lines when moving the mouse without pressing the button.
Is this right operation?

No - this is a likely bug in your mouse driver. The squares are "selection" squares meant to select an area to zoom to. This is done programmatically in nxworld by pressing the button, then moving the mouse with button down.

I suspect the problem in your driver has to do with the mouse button and x/y state in Poll vs Read when we changed that for operation as a polling driver. As I mentioned, the best way would be to collect the mouse state in Poll and save it if something changed. Then just use the saved state for returning in Read. IIRC you are re-reading the mouse status in the Read routine; this could be an issue.

> Also, I can zoom in by pressing left button, but I don't know how to zoom out.

Look at the source for handling keystrokes (dokeydown), I think the answer is 'c'. There may not be a way to zoom out with the button.

Thank you!

tyama501, 2023-03-10:
> As I mentioned, the best way would be to collect the mouse state in Poll and save it if something changed. Then just use the saved state for returning in Read. 

Thank you @ghaerr ,
I think I have done as you mentioned last time.

Here is my code.
https://github.com/tyama501/elks/blob/merge_pc98/elkscmd/nano-X/drivers/mou_pc98.c

And it is working with other application.

ghaerr, 2023-03-10:
Hello @tyama501,

Your driver looks good, I must have been looking at the code prior to your update. (BTW, buttons_before seems to be no longer needed, but doesn't affect this issue).

I am not sure now whether the problem is within nxworld, or perhaps something to do with Nano-X. Unfortunately, I don't have an emulator with a mouse driver to test. What you might do to test your driver is enhance the sys_utils/mouse.c test program which will show you visually how your mouse is working. You could then test with mouse up/down and movement to check. It will need a slight rewrite because it depends on a mouse file descriptor and would need to be bound with the mouse driver in order to work. Perhaps copy mouse.c into nano-X/demos and make it work there under the Nano-X build Makefile.

If you can find an IBM compatible emulator with mouse, we can then look further into nxworld to determine what might be wrong. In general, I am guessing that this has something to do with the transition between mouse up/down and mouse movement, since that is what nxworld uses to operate. The bug could be in a driver, nano-X itself, or nxworld at this point.

Thank you!

tyama501, 2023-03-10:
buttons_before is to keep previous state of button pressed or not pressed,
and it will not return 1 to go to read if keep pressing the button.
It will go to read if the button is released.

I will try the testing.

Thank you.

tyama501, 2023-03-10:
Oh, there is serial mouse in DosBox-X.
https://github.com/joncampbell123/dosbox-x/wiki/Guide:Mouse-support-in-DOSBox%E2%80%90X#serial-mouse

tyama501, 2023-03-10:
The serial mouse worked on nxlandmine!!
But it seems there is no nxworld on fd1440-fat.img

ghaerr, 2023-03-10:
> buttons_before is to keep previous state of button pressed or not pressed,
and it will not return 1 to go to read if keep pressing the button.

I see. There could be an issue with that logic that is causing the problem? Nano-X needs to know the current button state regardless of whether the mouse is moved. I notice that the Poll routine writes the button state in `button` even if return 0, so if by chance Nano-X called Read multiple times without calling Poll, this could possibly be an issue.

> I will try the testing.

Testing with `mouse` should hopefully show any issues, but it seems regardless since serial mouse is working the issue is in PC-98 mouse driver.

> But it seems there is no nxworld on fd1440-fat.img of IBM PC.

No, unfortunately it doesn't fit. We could move to compressed executables which would allow for a lot more applications per disk, but some time ago it was reported that the decompression was slow on 8086/8088 machines.

tyama501, 2023-03-10:
I meant the serial mouse is working with nxlandmine.
I haven't test the nxworld since there is no nxworld in the image.
I will test it later when I have time and report.

Thank you.

tyama501, 2023-03-11:
Hello @ghaerr ,

nxworld for IBM also worked on DOSBox-X with the serial mouse.
So, the problem is in my drivers.

Here is the video of the IBM and DOSBox-X.
It's great I could get a reference to compare.

https://user-images.githubusercontent.com/61556504/224462550-9f6b602e-4204-45ba-93ec-c7761a0b0d8b.mp4

Thank you.

tyama501, 2023-03-11:
Polarity of the buttons may be inverted.
[image]

tyama501, 2023-03-11:
I inverted the polarity of the button and also added read button in MOU_read too in case the Poll is not called.
A lot better than before since I can release the button but not perfect.

Video that put the button state in text plane simultaneously.

https://user-images.githubusercontent.com/61556504/224474362-b1bee948-f41a-4d31-8fcd-c30e26118f12.mp4

ghaerr, 2023-03-11:
> I inverted the polarity of the button and also added read button in MOU_read too in case the Poll is not called.
A lot better than before since I can release the button but not perfect.

That's pretty nice being able to display printf debug info on the graphics screen! I am not sure that read is called w/o poll, check that more carefully as it seems the likely problem remains that Nano-X is getting multiple states when there hasn't been a change, for some reason. This could be the result of new code in read not in poll, or vice versa. You might also recheck the main loop to make sure that read is only called after poll, not on its own. Looking at engine/devmouse.c and possibly putting in some debug statements there might help too.

There is a slight third possibility that for some reason the mouse driver calls are "hogging" the CPU and that Nano-X isn't getting an event to remove the XOR rectangle being drawn... that is, if calls to GsSelect always resulting mouse events but the other events don't get sent. This can be checked looking at nanox/srvevent.c.

tyama501, 2023-03-12:
Hello @ghaerr ,

I remember I haven't implemented XOR in my screen driver.
So, I think the problem is in screen driver...

Thank you.

tyama501, 2023-03-18:
Hello @ghaerr ,

I added XOR function.
https://github.com/tyama501/elks/commit/42526433de0cf8994782d307dea7affac9c2fb0a

nxworld worked finally.

https://user-images.githubusercontent.com/61556504/226121705-7b779956-635b-4619-b249-3bea50e6ec81.mp4

I will make PR after testing with the hardware.

Thank you.

ghaerr, 2023-03-18:
Hello @tyama501,

Nice work, the XOR code looks great and I'm glad we've figured out the graphics issues with nxworld :)

Taking a look at the latest mouse driver, I'm wondering whether we should include lines 155-156 that re-read the mouse buttons, in the case that the Poll routine was not called. Unless absolutely necessary (meaning that we can show that Read is in fact called without Poll, which should not happen), reading the buttons twice opens up the possibility of a race condition where the button changes state between Poll and Read for the normal case, and the button event gets dropped.

I would suggest removing that code and checking whether your PC-98 driver still works fine. If so, leave it out to avoid that race case.

Thank you!

tyama501, 2023-03-19:
Thank you @ghaerr ,

It seems that there is no difference with and without the re-read function.
I will remove it.

I found another difference in nxworld with IBM-PC.
When I click right button, it displays latitude and longitude in the left bottom corner,
but it does not clear it when I released the button.
It displays overlapped when I click again.

IBM ver. can clear it when released the button.

ghaerr, 2023-03-19:
> Maybe I need to add XOR in drawpixel() too.

Yes - looking at the `showcoords` routine in demos/world.c, it uses the an XOR GC when drawing the text, and text output on this version of Nano-X uses pixel-by-pixel drawing.

ghaerr, 2023-03-24:
Hello @tyama501,

Looks great, glad you've now got everything working. Please send a PR for update and we'll commit it.

On another issue, a while back you rewrote the serial driver to handle timeouts because the CTS line is not operational on the chip used in the PC-98. After that, I added a mechanism that allows for the timeouts to be handled accurately with the kernel, rather than busy-looping which doesn't work well with different PC-98 CPU speeds. Are you interested in updating the PC-98 serial driver to use that newer mechanism? If so, I can (re)find the code I used that you would use in your driver.

Thank you!

tyama501, 2023-03-25:
Hello @ghaerr ,

OK, it worked on the real hardware, PC-9801BX too.
I will make a PR.
I posted a video with nxworld and nxlandmine on twitter.
https://twitter.com/tyama50/status/1639531608209498112

>Are you interested in updating the PC-98 serial driver to use that newer mechanism? If so, I can (re)find the code I used that you would use in your driver.

Yes, I remember you give me a code before. Sorry I couldn't try it yet.
Would you please repost it here again?

Thank you!

ghaerr, 2023-03-25:
Hello @tyama501,

> Yes, I remember you give me a code before. Sorry I couldn't try it yet.
Would you please repost it here again?

The change is discussed [here](https://github.com/jbruchon/elks/pull/1511#issuecomment-1371436867). It involves using `jiffies` which changes with clock ticks every 1/100 second, and the `time_after` macro. That comment shows the small change that you should be able to make to have the timeouts be accurate at any baud rate for PC-98.

Thank you!

tyama501, 2023-04-01:
Hello @ghaerr ,

> 33.3 ms is required for each character transmitted at 300 baud.
> unsigned long timeout = jiffies + 4*HZ/100;     /* 40ms, 300 baud needs 33.3ms */

1 / 300 = 0.00333 = 3.33ms, isn't it?
If so, may be jiffies + 4*Hz/10; /* 4ms */ is enough.

Thank you.

tyama501, 2023-04-01:
Oh, jiffies cannot be set less than 10ms?

ghaerr, 2023-04-01:
> 1 / 300 = 0.00333 = 3.33ms, isn't it?

Yes, 3.33ms per bit, and 33.3ms for 10 bits (1 start, 8 data, 1 stop).

> jiffies cannot be set less than 10ms?

Yes, the ELKS timer is set at 1/100 second, and jiffies is incremented each timer interrupt, which = 10ms. Thus for safety, 33.3ms was rounded up to 40ms.

tyama501, 2023-04-01:
Thank you. I see. It's 10 bits.

tyama501, 2023-04-08:
I tried CSLIP between ELKS on PC-9801UV21 and ELKS on NekoProject21/W.

There is a lot of retries but I could connect to the PC-98 with telnet. (Very slow with 1200bps) 
[image]

[image]

FTP can be connected but it seems that the file I get is 0bytes. 
[image]

[image]

There may be hardware problem on this PC-98, so I will try another PC-98 when I have time.

ghaerr, 2023-04-08:
Hello @tyama501,

> I tried CSLIP between ELKS on PC-9801UV21 and ELKS on NekoProject21/W.

Can you try just SLIP instead? When I originally started working on ELKS TCP/IP, I noticed that CSLIP has a lot of problems. SLIP on the other hand should work fine with few errors, even up to 38.4K. In some instances, the "fast" version of the IBM PC serial driver must be used. This version, set by CONFIG_FAST_IRQ{4,3}, doesn't go through the normal kernel interrupt mechanism as its too slow and drops characters, especially if there is no UART hardware FIFO. See `serfast.S` for details). However, I would think that you should be able to run 9600 baud no problem.

If you're interested in debugging this more (using SLIP to determine highest baud rates possible) I would suggest adding a `dprintf` when a UART receive buffer overrun or framing error occurs.

Thank you!

tyama501, 2023-04-09:
Hello @ghaerr ,

>Can you try just SLIP instead?

I tried SLIP and I got similar result.
This V30 machine seems to have some trouble on COM port, so I will try on newer machine.

Thank you.

ghaerr, 2023-05-12:
Hello @tyama501,

I'm doing well, thank you. Things are starting to warm up over here in Utah , finally.

All three of your additions look great, go ahead and submit a PR for them. The `unsigned int cylinders` will be fine for all systems, so no need to ifdef that specially for PC-98.

Thank you!

tyama501, 2023-06-04:
Hello @ghaerr ,

I think I have destroyed FAT filesystem a couppl times when I am using vi and do some command unintentionally.

I cannot tell what exactly I did yet since I am not vi user normally, but one possibility is some unintentional repeated command.
Onece this happen, it can never write any file even after rebooting or resetting, although it can boot.

I think I have seen strange tmp file of vi onece.
Is that possible if tmp file gets huge it can destroy filesystem?

ghaerr, 2023-06-04:
Hello @tyama501,

That is very strange. I suppose if a tmp file is huge it could use up all available disk space on a filesystem,
prohibiting further writes, but not sure how vi might do that.

Is there any chance you have a copy of a bad filesystem you can upload here for inspection? Also might be possible to run a DOS chkdsk on it?

Thank you!

tyama501, 2023-06-04:
Thank you @ghaerr ,

Well I'm not sure it was huge or not and I think I already delete it. It's like read-only filesystem but somehow I can delete the file.

It was 1.23Mbytes image and also it was not raw image, so I'll try repro

tyama501, 2023-06-04:
Thank you @ghaerr ,

Well I'm not sure it was huge or not and I think I have already delete the file. It's like read-only filesystem but somehow I can delete files.

I will check if I still have the image later but it was 1.23Mbytes and also not raw image.
Maybe I will try to reproduce problem in 1.44Mbytes when I have time, so you can see it.

tyama501, 2023-06-04:
It seems that I have lost the broken image.

Today I pressed keys randomly on vi with 1.44MB disk and I could make segmentation fault,
but the temporally file was about 30KB.
The file system is not broken, and I could write files after that.

Also this is the different error message that says "reading back from tmp file".
So this is different case, and ok since it is not broken permanently.

[image]

[image]

[image]

tyama501, 2023-06-04:
Oh I remember why I felt the temp file is strange.

It is long file name. How can it be created?

tyama501, 2023-06-04:
Sorry, 8.3 letters are short file name.
My eyes were too tired :)

tyama501, 2023-06-10:
Hello @ghaerr ,

It's 1232 img but I got another image that can read but cannot write.
Well, it says disk is full and I can write it if I remove files,
but I am not sure it is really full,
because I only created a couple of text files, about 8KB.

Here is the disk full image attached.
[fd1232_full.zip](https://github.com/jbruchon/elks/files/11711623/fd1232_full.zip)
I think only files added are microef.bas and bak0607/microef.bas
[image]

Here is the ok image before creating files. It says 127KB Free.
[fd1232_ok.zip](https://github.com/jbruchon/elks/files/11711625/fd1232_ok.zip)
[image]

By the way,
I tried chkdsk on FreeDOS but it seems that it says "cannot get summary" even I checked the latest IBM ELKS image.

fd1440-fat.img for IBM
[image]

fd1440.img for PC-98 that I built.
It also says "Fats are different".
[image]

CHKDSK source code
https://gitlab.com/FreeDOS/base/chkdsk/-/tree/master/SOURCE/CHKDSK

Thank you.

ghaerr, 2023-06-10:
Hello @tyama501,

First, with regards to `chkdsk`, yes, it will probably always error with "FATs are different". This is because ELKS only keeps up a single copy of the FAT, for speed reasons. (We could add dual FAT support but things would likely slow down quite a bit further). Thus, this isn't a good option to use for checking FAT disks, I'm sorry.

It looks like both your 1232k floppy images are pretty standard FAT12 disks with cluster size 1, which is simple.

I would say that it is very likely that the floppies are indeed out of space. Thus, the question may be, "why?". 
Can you document exactly what you did to run out of space? You said you thought `vi` was the cause, can that be duplicated, perhaps `vi` is saving more than we think to a tmp file or not deleting something. There could also be some issue with the last 128k bytes or so of disk where ELKS thinks it is full when it is not quite.

Would it be possible to try adding, say, 512-byte sectors, one at a time using the `cp` of a small file, to see whether the above is true, where you add 8k bytes 512 bytes at a time and check the disk free count?

Thank you!

tyama501, 2023-06-11:
Thank you @ghaerr ,

So it will make FAT different when writing to existing dos partition, right? I will write a notice to my homepage.

This time I think I was doing very standard use of vi, just reading and writing.

I will try copy debugging.

ghaerr, 2023-06-11:
> So it will make FAT different when writing to existing dos partition, right? 

Yes, only the first FAT will be updated by ELKS. This usually isn't a big deal, as to my knowledge DOS (the OS itself) doesn't use the second FAT table directly, although some file repair programs do.

tyama501, 2023-06-11:
Before I test the copy,
I checked the FAT tables of the previous images I posted with free software on windows.

<-- fd1232_full Fat and fd1232_ok Fat -->
[image]

It seems that all the Fat in fd1232_full have been occupied with values at least once,
the software also says it's full.
So, the elks telling full maybe true.
But the question is still how and why it gets full.
(vi could not remove tmp file correctly?)

ghaerr, 2023-06-11:
> It seems that all the Fat in fd1232_full have been occupied with values at least once,

Each 12-bit entry in the FAT which is 0 indicates free space for that sector/cluster.

It would be interesting to see a dump of the FAT table just before the disk is full, or at least just before the last operation that makes the disk full, as each FAT entry which is non-zero, when the cluster size is 1, indicates a disk sector allocated to a file. In the 1232k image, disk sectors are 1K instead of the usual 512-byte sector size of most FAT disks.

ghaerr, 2023-06-12:
Hello @tyama501,

Thank you for the very understandable presentation, I think you're on to something here. I will try to repeat. First guess is it seems that perhaps a file is not being closed properly or somehow a file is being allocated but contents not released for some reason. Not sure if that's the temp file, vi problems, or ELKS.

I am wondering whether running `sync` might change the outcome by flushing the disk?

Thank you!

ghaerr, 2023-06-12:
More testing shows that `vi` is buggy when 3 blocks are remaining on MINIX filesystem: run `vi foo` to create a new file, add some text, like "test", and save using ":wq". No errors, and `foo` is zero-length!

With 4 free blocks remaining, `vi` seems to work OK in the above case.

So there's a definite problem here with vi, I've made a note of it. I am not sure what the FAT issue is, but am guessing it is related to the same reasons why FAT filesystems can sometimes have errors shown with `chkdsk` where a FAT table entry is used but not attached to a directory entry. I am not sure whether this is worth trying to debug before fixing `vi`.

ghaerr, 2023-06-13:
Hello @tyama501,

>  I heard recently that to open the elks disk directly from NEC MS-DOS, some BPB are missing.

I would be very interested in learning more about what version BPB NEC MS-DOS disks require, to see if the ELKS format might be updated to work with it.

> I'll try again the CHKDSK on FreeDOS.

It would be nice to find a CHKDSK that doesn't exit with no further information when the FAT tables don't match, that was the reason for suggesting the MSDOS version. There may be others that work.

Thank you!

tyama501, 2023-06-13:
I think initial disk of elks is FAT is same, and FreeDOS CHKDSK did not say FAT is different,
but it failed to get report for some reason.  I will see why it is.  Also I will look logs on twitter about the NEC MS-DOS.

It's a little off topic but I tried creating minixfs on pc-98 harddisk partition.
I could read write it but sometimes it says " dirty".
What does the "dirty" mean in minixfs?
I can fix these dirty by fsck.

Thank you.

ghaerr, 2023-06-13:
> FreeDOS CHKDSK did not say FAT is different, but it failed to get report for some reason. 

It seemed from your screenshot that these two CHKDSK programs were possibly from the same source. One of the versions could be exiting without the "FAT is different" message?

The MINIX filesystem keeps a "dirty bit" on the superblock (1st sector after boot block) which indicates whether it has been mounted, so that an `fsck` might be performed. ELKS sets that bit as well and writes it on mount, unless mount is read-only. Running `fsck` on the filesystem will reset that bit.

On the next boot or mount, if that bit is still set, ELKS displays a message saying the filesystem is dirty, meaning that `fsck` has not been run since the filesystem changed. Since the dirty bit is set on mount, it doesn't actually mean the filesystem has been changed, but just mounted. I have considered the idea of only setting that bit if the filesystem is actually written, but that isn't correct according to the MINIX filesystem specification. That said, it seems possibly not particularly useful to indicate dirty on only a mount with no write.

tyama501, 2023-06-13:
The screen shots I attached before were using same CHKDSK, but elks images were different.
First one that doesn't say FAT is different, it was 1.44 IBM image that I downloaded from this github that I didn't write any files.
Second one that says FAT is different, it was image that I wrote some files from elks.

Thank you for the explanation for minixfs.
So I understand I don't need to mind it saying "dirty" :)

tyama501, 2023-06-16:
I could build fsck.fat of this tool with FreeBSD.
https://github.com/dosfstools/dosfstools

Although it seems the test not completed with Invalid argument but may be I can debug.
[image]

ghaerr, 2023-06-16:
> Although it seems the test not completed with Invalid argument but may be I can debug.

Hmmm, I'm wondering whether `fsck.fat` can support 1024-byte sector floppies... It is a bit strange why it would be trying to read 1835 bytes. Perhaps you can try it on a 512-byte sector floppy from standard MSDOS to see if it gets any further?

tyama501, 2023-06-17:
Hello @ghaerr ,

I was using long command,
now I think the fsck.fat worked correctly.
It can target image file directly.

Here is the result.
fd1232_full/fd1232.img (Bad image, https://github.com/jbruchon/elks/issues/1545#issuecomment-1585507877)
There are a lot of unused clusters.
119 unused clusters + 1102 (used for files) = 1221, that means full.
[image]

fd1232_ok/fd1232.img (OK image)
There are only 3 unused clusters.
[image]

Thank you.

ghaerr, 2023-06-19:
@tyama501, what do you think the conclusions are here? Here's what I am thinking:

1. `vi` is buggy and doesn't operate properly when less then 3 sectors are left on a MINIX or FAT filesystem.
2. `vi` also sometimes gets stack overflow error (can't repeat but saw in your screenshot).
3. When ELKS reports FAT filesystem sizes, "unused" clusters may not be present in any file, but the free space reported is correct.
4. It is a possibility that when running `vi` with error "trouble writing to tmp file" that unused clusters get allocated (to no file) as a result. These clusters can add up and use up all free space on the filesystem.
5. We don't yet know what system calls `vi` might be making that possibly allow for filesystem sectors to be allocated to a file. The sectors are not released when that file's directory entry is removed (likely after a `create` call), with possibly some error occurring prior to the `close` system call. I suppose this could be possible with an `unlink` executed prior to a `close` after creating a file and writing data. This would have to be tested on FAT.
6. MSDOS programs can also create unused sector entries on FAT filesystems, so the issue may be one of the design of how the FAT filesystems are implemented, rather than an ELKS bug. We need more information on what `vi` is doing in the error case to differentiate. Notably, the MINIX filesystem is not left with unused/allocated sectors when `vi` is run on MINIX fs.

[7 more replies]

closed 2023-08-03 by ghaerr