ghaerr/elks #1549: cannot rename(tmp_fname, "/etc/passwd") in passwd

tyama501, 2023-03-03:
**Description**
- What is expected ?
- What is observed ?

**Configuration**
- Version (commit or release):
- Emulator or HW platform:

**How to reproduce ?**
- Does the problem always happen ?
- List the steps to reproduce the problem.

**Raw data**
- Screenshot(s):
- Log file:

**Additional information**
Add any other context information about the problem here.

tyama501, 2023-03-03:
Oh, 00007aaa seems longfile name.
The longfile name cannot be move in ELKS, can't it?

tyama501, 2023-03-03:
It seems that the name is created with libc/misc/tmpnam.c

tyama501, 2023-03-03:
Sorry, 8 letters so it may be short file name.

ghaerr, 2023-03-04:
Hello @tyama501,

I think the problem here is likely the FAT filesystem not supporting a rename operation properly, since I tried repeating this on MINIX filesystem and it works fine. I'll gen up a FAT filesystem to test on and report back what can be done to fix this.

Thank you!

ghaerr, 2023-03-04:
Hello @tyama501,

I have verified that the problem is that ELKS does not actually support `rename` for any files on FAT filesystems. (ELKS also does not support rename of directories on FAT or MINIX, as described in #583). For this reason, `mv` has a workaround where a file is copied rather than being renamed. I will have to add this for `passwd` as well, and will submit a PR for it to fix this.

Thank you!

tyama501, 2023-03-04:
OK, thank you!

ghaerr, 2023-03-04:
Hello @tyama501, my PR #1550 automatically closed this. Please test and verify it works on your system also. I have tested on QEMU w/FAT filesystem.

Thank you!

closed 2023-03-04 by ghaerr

tyama501, 2023-03-05:
Hello @ghaerr ,

Thank you, it worked.

I have one more question releted to etc/passwd and etc/group.

I login with root and modified etc/passwd, etc/group, added tyma like this and also created /home/tyama.
/etc/passwd
[image]

/etc/group
[image]

Then I reboot, and see /etc directory some files user has been changed to tyama from root like /etc/group.
[image]

Is this correct behavior?
Also the group shows 247 not user3.
I can  modifiy these file from tyama.

Thank you.

tyama501, 2023-03-05:
Well, I think it is ok if this is known and limitation for FAT file system since FAT is not suitable for unix permission.

ghaerr, 2023-03-05:
Hello @tyama501,

The FAT filesystem does not support UNIX-style permissions, and looking at the source in fs/msdos/inode.c::msdos_read_inode(), it sets the files being read as being owned by the current UID/GID:
```
    inode->i_uid = current->uid;
    inode->i_gid = current->gid;
```
You will notice that all files are also mode 755 and directories 777, which is set by the FAT filesystem since there are no FAT file permissions stored comparable with UNIX.

It is a bit strange why not all the files are shown owned by tyama/247, I haven't figured that out yet, but I think `ls` is showing the last UID/GID simulated for each file, based on the process UID/GID that last read or opened the file, possibly from the inode cache.

Looking at include/fs.h, the group ID field `i_gid` is an 8-bit value. So the ELKS /etc/group is incorrect with user0-user3 having group IDs of 500-503. Since user tyama is group 503, 503-256=247, since GID is only a byte wide. I'll fix the /etc/group file contents to lower the max group ID below 256.

Thank you!

tyama501, 2023-03-05:
Thank you @ghaerr ,

Yes, many files are tyrma/247 but some are not.

In /bin/ directory
getty, init, login, sh are owned by root.
So, it seems that important files are still root.

/etc/passwd is also owned by root and it cannot write by tyama which is good.

ghaerr, 2023-03-05:
> In /bin/ directory
getty, init, login, sh are owned by root.
So, it seems that important files are still root.
/etc/passwd is also owned by root and it cannot write by tyama which is good.

Like I said, there isn't any actual UID/GIDs with FAT, and the kernel sets the file's UID/GID based upon the last process that read the file (which is not the same as directory listing). Thus, /etc/passwd should probably change to tyama/247 after you cat it. You might experiment more to see that this is true or not.

tyama501, 2023-03-05:
Hmm here is ls after cat.
Well it remains root after rebooting emulator.
 
[image]

ghaerr, 2023-03-05:
> Hmm here is ls after cat.
Well it remains root after rebooting emulator.

> the kernel sets the file's UID/GID based upon the last process that read the file 

Ok, I had that reversed - for FAT, the filesystem driver sets the file UID/GID at the FIRST time the file is accessed, and doesn't change it afterwards, as the inode is only read once and cached. Thus, for /etc/passwd, this file is checked by /bin/login when first logging in, that's why it doesn't change. Same with /etc/inittab, it is read by /bin/init, thus it remains root/root.

We could consider to change this behavior, but it doesn't really matter, since the file attributes are faked anyways. So at this time, I will just fix /etc/group and /etc/passwd to keep group IDs < 256 so at least the group display works properly on `ls`.

Thank you!

tyama501, 2023-03-05:
〉it doesn't really matter,

I agree.

Thank you.

PR #1550 (2023-03-04): [cmds] Fix passwd command
Fixes #1549.

Fixes `getpwent` bug introduced in #1546.

Reformats very ugly source code in libc/getent.

Adds first pass at standardized ELKS C indentation script. (Needs tabs replacement to 4 spaces also).