ghaerr/elks #1556: login, add more apps

FreeLANMan, 2023-03-08:
Hello, interesting project.
- What is the username and password for the images?
- Can I add new Linux programs just by compiling them with the indicated tool?
I was interested in the system because it works with the PCEmulator for the TTGO VGA32 board.

ghaerr, 2023-03-08:
Hello @FreeLANMan,

> What is the username and password for the images?

You can login using `root` with no password.

> Can I add new Linux programs just by compiling them with the indicated tool?

Yes, please refer to some of the programs distributed with ELKS, they live under the elksmd/ directories, and their Makefiles will show easy ways to build them.

Thank you!

FreeLANMan, 2023-03-12:
Thanks for the answers. 
All fat disks have the same content?

ghaerr, 2023-03-12:
> All fat disks have the same content?

Each FAT disk has the same content as the MINIX disk of the same size. As the image size gets larger from 360k, the same additional programs are added to FAT and MINIX disks. We recommend that you use an 1440k if possible.

closed 2023-03-29 by ghaerr