ghaerr/elks #1560: Unipcemu fails to boot latest 360k fat fd image

Randrianasulu, 2023-03-12:
**Description**
- Failure to find /bin/sh

**Configuration**
fd360-fat image from
https://github.com/jbruchon/elks/actions/runs/4398299501
- Emulator or HW platform: unipcemu

**How to reproduce ?**
- only tried once so far - booting with AT / 286 / 4 MB / VGA  takes 30 min on my hw!

- List the steps to reproduce the problem.
Unipcemu a bit hard to config - I used renamed pcem roms and hand-editing setting.ini

**Raw data**
- 
[image]

- Log file:

**Additional information**
might be emulator problem. I used self-compiled build for linux. 32-bit, too.

ghaerr, 2023-03-12:
Hello @Randrianasulu,

It seems there may be a couple problems - 1) the Unipcemu BIOS is definitely not setup to recognize a 360k floppy,
as BIOS is returning "/dev/fd0: 0 cylinders, -31870 heads...". So check into that. Problem 2) is that ELKS
probed the disk itself and determined (incorrectly but this is tricky with emulators) that the floppy is
CHS 40, 2, 36 - which is an incorrectly formatted 1.44M floppy (see the fd0: ... message). The proper CHS
for 1.44M floppies is 40, 2, 18.

I would believe the issue is the floppy setup in Unipcemu, as neither the BIOS emulation nor ELKS is
coming back with a proper floppy format. You could also try booting the 2.88M or 1.44M floppy images from
GitHub. The reason /bin/sh isn't found is that the disk image format is wrong, so the FAT filesystem isn't in fact
correctly accessible from it.

Thank you!

Randrianasulu, 2023-03-13:
@ghaerr I'll try 1.2 and 1.44 images too, but I tried 720k and very similar error still here. AT machines seems to require their own DOS disk /program for setting CMOS  params, so I currently live with defaults.

ghaerr, 2023-03-13:
> I tried 720k and very similar error still here.

Well, the CHS (cylinders, heads, sectors) for 720k is 80, 2, 9, and ELKS is thinking (by probing) that the CHS is 40, 2, 36, so that's certainly why that's failing. The CHS has to match up. 1.2M is CHS 80,2,15, so that won't work. 2.8M is 80,2,36 so that
could work, since probing has the same H and S but less cylinders. 

When the CHS isn't matching, the wrong sectors of the disk are read into memory, which is why neither /dev/console nor /bin/sh can be found. We need to figure out how to set disk CHS parameters (or expected image size) with Unipcemu.

Randrianasulu, 2023-03-13:
well, isn't simplest way to solve this  to just ask user? :)

Randrianasulu, 2023-03-13:
I mean stop booting with question-prompt

Randrianasulu, 2023-03-13:
used Advanced diagnostics disk from

https://winworldpc.com/product/ibm-pc-diagnostics/at-1x

set diskette to "double sided".

ELKS on startup said I have 2* 360kb disjettes, but 1.2Mb image booted anyway , so I log in as root and was able to execute "ps" command.

So, it works but in somewhat confusing way (on wikipedia there is indication confusion between 360k and 1.2m floppy hardware was real on real hardware too)

ghaerr, 2023-03-13:
> used Advanced diagnostics disk from
> https://winworldpc.com/product/ibm-pc-diagnostics/at-1x

Very interesting that that utility fixed the diskette to double sided! I wonder how it works, by setting some special CMOS RAM that is used by your BIOS?

> ELKS on startup said I have 2* 360kb diskettes, but 1.2Mb image booted anyway , so I log in as root and was able to execute "ps" command.

That's great news its working! We're aware that sometimes ELKS doesn't think the floppy is the same format as the BIOS, which is why the probing occurs. Also, many older BIOS don't have values for the higher density floppies that came later. If an ELKS boot floppy image is used, then ELKS will take the CHS format directly from that boot floppy, rather than believing the BIOS only, which is likely why it is now working, in conjunction with the Advanced Diagnostics.

Are there other problems with booting or can this be considered fixed?

Randrianasulu, 2023-03-13:
@ghaerr I think those values saved in battery-backed  RAM, and apparently you can even use BASIC in original AT rom to do some poking!

https://www.minuszerodegrees.net/5170/setup/5170_setup.htm

But I guess for now this issue can be considered fixed, unless you  still can/desire to implement manual parameter entry early (for unsetup machines).

ghaerr, 2023-03-14:
Wow @Randrianasulu, very interesting set of articles on setting up PC BIOS, thanks :)

> unless you still can/desire to implement manual parameter entry early (for unsetup machines).

ELKS itself probably has no business writing to any battery-backed RAM nor even BIOS Data Area locations. Actually, I have been working to get ELKS less, rather than more, dependent on BIOS and its data areas. That said, I'm guessing this latest issue is more the result that your system BIOS won't work properly unless its been properly setup using one of the mechanisms mentioned in your linked article(s). ELKS is dependent on the BIOS CHS when loading the kernel, but afterwards, as mentioned earlier, ELKS will override the BIOS CHS settings based on the ELKS boot disk image CHS settings and use them instead. I'm not exactly sure where the problem lies, but I'm glad this issue is resolved.

closed 2023-03-18 by ghaerr