ghaerr/elks #1563: Can't compule with self-build toolchain

Randrianasulu, 2023-03-14:
**Description**
Expectation: make produces kernel
Reality:

ia16-elf-ld.gold: error: address of section '.fartext' moves backward from 0x20090 to 0x20000      
arch/i86/kernel/akernel.a(system.o):function setup_arch: error: undefined reference to 'outb_p'    
arch/i86/kernel/akernel.a(system.o):function setup_arch: error: undefined reference to 'outb_p!'   
ia16-elf-nm: 'arch/i86/boot/system': No such file

**Configuration**
- 4376409cdb04f52dff53cd149b2234af0cd434b7
- Emulator or HW platform: i586 host

**How to reproduce ?**
- Does the problem always happen ? only tried once!
- Build toolchain from https://github.com/tkchia/build-ia16 up to at least gcc2, symlink its output dir 'prefix' as 'cross' inside pre-downloaded elks dir.
Try to configure ELKS with some extras. I'll attach config in minute

Randrianasulu, 2023-03-14:
[elks-my-config.txt](https://github.com/jbruchon/elks/files/10967305/elks-my-config.txt)

ghaerr, 2023-03-14:
Hello @Randrianasulu,

When building the toolchain, did you execute `tools/build.sh` from ELKS, or did you build it from @tkchia's repository using another means? The former is preferred, as portions of the ELKS C library end up being embedded within the built gcc toolchain.

Try using this config file, it will build an XMS-supported kernel and has been throughly tested:
[config.xms.zip](https://github.com/jbruchon/elks/files/10971153/config.xms.zip)

If building with this config fails, I suggest removing `cross/` and starting the (long) toolchain build again using `tools/build.sh`.

Thank you!

Randrianasulu, 2023-03-14:
@ghaerr I build toolchain manually by executing their build.sh with different targets outlined on readme, including elks-libc.

Your config builds ok, time to diff :)

ghaerr, 2023-03-14:
> Your config builds ok, time to diff :)

Good to know and glad to hear it. I'm interested in the differences. It is possible you selected too many things to fit in the kernel. Did you use `make menuconfig` to build it from scratch?

Randrianasulu, 2023-03-14:
@ghaerr yes, very possible :) I added unix sockets and serial port as minimum ...

ghaerr, 2023-03-14:
@Randrianasulu, you have to select "Far Text Kernel" which gains us another 64k code (=128k total) otherwise most of the kernel won't fit anymore. That's probably your problem, and menuconfig won't warn you about it.

Randrianasulu, 2023-03-14:
@ghaerr I think my problem lies here:

-CONFIG_HW_COMPAQFAST=y

ghaerr, 2023-03-14:
> -CONFIG_HW_COMPAQFAST=y

Wow, you mean that the OUTB for fast mode on Compaq systems killed your system? It should at least come part way through the boot.

Randrianasulu, 2023-03-14:
@ghaerr no, I mean build fails for me if I select this option

ghaerr, 2023-03-14:
> I mean build fails for me if I select this option

I see. When I set that option, the compilation succeeds, but the link fails due to an undefined `outb_p` function. This is a bug, because the file elks/arch/i86/system.c is missing the line `#include <arch/io.h>`. I will fix that.

Randrianasulu, 2023-03-14:
Also, it was parallel port/cga pushing kernel size above acceptable limit with all other devices compiled in. Now I just need to learn where kernel verbosity at startup set, because this one chat a lot less than release/github action images

ghaerr, 2023-03-14:
There is no verbose startup option - each compiled in driver/console/etc uses up a line or so for each subdevice it's dealing with.

The standard release/github images are built using `ibmpc-1440.config` from the ELKS main directory.

Randrianasulu, 2023-03-14:
ah, those (messages) were on serial console (just as I compiled in)

closed 2023-03-18 by ghaerr