ghaerr/elks #1568: TinyIRC does not comply with PING/PONG and thus will not connect to anything on ELKS

DutchComputerKid, 2023-03-21:
**Description**

TinyIRC doews not comply with PING/PONG transmissions.
- What is expected ?
Upon login at say, UnrealIRCd TinyIRC should acknoledge the connection by using PING + PONG commands for connection verification and keepalive.
- What is observed ?
It does not at all and thus cannot connect unless the server is old enough to not need this.

**Configuration**
- Version (commit or release):
Built from source yesterday, /2029e799b079c722a6bc4847911eb649021be580
- Emulator or HW platform:
V20, 640K, 64MB CF, VGA.

**How to reproduce ?**
- Does the problem always happen ?
Yes
- List the steps to reproduce the problem.
1. Transfer TinyIRC from compiled sources since it does not get loaded into the HD image.
2. Try to connect say to an UnrealIRCd platform

**Raw data**
- Screenshot(s):
[image]
- Log file:
None

toncho11, 2023-03-21:
I thought the principal problem with TinyIRC is that nowadays all IRC servers use encrypted communication. And ELKS' TinyIRC is not designed to handle any encryption. 

I have previously tested it in #870

DutchComputerKid, 2023-03-21:
Exactly, I run an IRC server/relay with some other peeps offering both tls and open connections for use cases like these.
Whenever a new machine gets built is always run to grab IRC and join the chat with the new machine in question.

But yeh, TinyIRC is old, and I have no idea if there are other open source clients that can be built for ia16. So, it is put old hold then, yes?

toncho11, 2023-03-21:
Yes, we can say that it is on hold until we find a client that complies and runs on undecrypted connection.
You can help by trying some clients. See my #870.
You can also give us a list of unencrypted severs to try in the future.

ghaerr, 2023-03-22:
Hello @DutchComputerKid,

@toncho11 is right, the ELKS TinyIRC doesn't work. Its based on some very old source, circa 1996, and IMO should be completely replaced. I know very little about IRC so others need to suggest possible small versions that might make sense for ELKS.

BTW, the "sys_brk(19) fail" on the screenshot indicates that the `tinyirc` client doesn't have enough heap space. That can be easily fixed, but won't solve the problem this issue is opened for.

> I think the principal problem with TinyIRC is that nowadays all IRC servers use encrypted communication. And ELKS' TinyIRC is not designed to handle any encryption.

ELKS could possibly be capable of handling encrypted communications. The typical problem is that many projects today use OpenSSH or something like that, which are huge, and can't be ported to 16-bit architectures. So what we need is a small, more self-contained IRC client, which isn't dependent on a large outside encryption library. Any recommendations would be appreciated, and I can look into their portability after ensuring they're not using OpenSSH.

Thank you!

PR #1569 (2023-03-22): Fix IRCv3 PING/PONG connection issue
IRCv3 specification sends an unique token on each PING which it requires be sent back by replying with PONG.

tested on libera.chat (non IRCv3 PING/PONG) and unrealircd (hybrid, one IRCv3 PING packet followed by original PING packets)

Fixes #1568
closed 2023-03-22 by ghaerr