ghaerr/elks #1580: Help adding content to installed ELKS

jsedanoj, 2023-03-31:
Hello,

  I have installed ELKS using the instructions in the wiki https://github.com/jbruchon/elks/wiki#Installation

I have booted from my floppy, created the partition, formatted the partition (1) and used "sys /dev/hda2" to install to the hard disk. Then, I can boot to ELKS using Minix monitor. So far so good.

The instructions now say:

> **Remarks**
> Please note that using sys creates a minimalistic install. For example when installing from a 360kb image you will miss most of the commands from /bin, and some config files from /etc (such as net.cfg and mount.cfg). You may need to complement your installation by manually adding these files from another ELKS build image (e.g. hd32-fat.img).

How do I do that? I am completely lost.

I have tried mounting the hd32-minix.img file as loop in my linux laptop (`mount -o loop hd32-minix.img /mnt` ), to see what it contains,  but only a 1.8M FS is mounted, containing more or less the same that the floppy image that I have already installed.

Any documentation to read? Any hint?

(1): honestly I needed to use Minix for that... but well... it is done.

ghaerr, 2023-04-01:
Hello @jsedanoj,

The `sys` script will copy and create a bootable filesystem based on the contents of the floppy used when running it. The major difference between the contents of the disk images produced by the ELKS build system is the contents of the /bin directory. 

The `hd32-minix.img` does contain all of the files in /bin. The file `elkscmd/Applications` shows (in a somewhat cryptic manner) the `:other` files that are only included on large images. Most of these are the larger, less used `nano-X` graphics and some games. I'm not quite how sure how to specify how to do this automatically, but copying those files onto your existing partition would get you a full installation.

Another option would be to use `make menuconfig` and select Applications by Disk Size, then select Other Applications. This will create a floppy image with the Other files, which are normally too big to fit on anything less than a 2.88M floppy. Note that the non-Other files will have to be turned off to fit.

I hope this helps.

Thanks you!

jsedanoj, 2023-04-02:
Thanks for the hints. Some doubts...

> The `hd32-minix.img` does contain all of the files in /bin. The file `elkscmd/Applications` shows (in a somewhat cryptic manner) the `:other` files that are only included on large images. Most of [...]

The problem for me is not the part of "copying [...] onto". Little by little using disquettes, I can do it. The problem is... copy from where? I understand hd32-minix.img contains the binaries, but I do not know how to extract them.

However...

> Another option would be to use `make menuconfig` and select Applications by Disk Size, then select Other Applications. This will create a floppy image with the Other files, which are normally too [...]

...however... this gave me an idea. I can build the source, and I suppose the binaries will be kept in a folder before sending them to hd32-minix.img ... will key?

(the destination HD is large enough, I created a 70MB partition)

ghaerr, 2023-04-02:
> The problem is... copy from where?
> I can build the source, and I suppose the binaries will be kept in a folder before sending them to hd32-minix.img

Yes - the complete image is produced in elks-home/target. Thus, you can copy all the binaries from elks-home/target/bin to your image.

jsedanoj, 2023-04-02:
Got it!
Thanks

closed 2023-04-08 by ghaerr