ghaerr/elks #1583: Help choosing keyboard layout

jsedanoj, 2023-04-02:
Hello,

  I have been able to burn the floppy image, boot from it and install it to a hard disk. I am also able to rebuild the system.

Now, I'd like to choose my keyboard layout. I can recall having read something somewhere that it needed a system rebuild, because the layout is embedded in the binary... But first, I do not remember where I did read it. And second, I cannot find any config in "menuconfig" to set this.

Do we have any documentation or any hint about it?

ghaerr, 2023-04-03:
Run `make menuconfig`, select K)kernel, D)rivers, C)character device drivers, then you can select the keyboard nationality under "XT Keyboard Support". Then save and `make`. The kernel will be rebuilt with a console driver with the scan codes selected for the keyboard you chose.

jsedanoj, 2023-04-10:
Hi back,

  I do not know if after the issue has been closed someone will read it.

I managed to build using this hints, and the keyboard keymap changed... the problem for me is that even when my keyboard is Spanish layout, the layout is not correct when I boot from it (I know I have successfully chosen the layout, because... keep reading). So, my guess on what is happening:

1) Maybe my keyboard is not an "XT keyboard". The PC I am using is Compaq SLT 286, so it is not an XT but an AT (286). Anybody knows if this may be the cause? Does ELKS only support XT keyboards?

2) Maybe keys-es.h is not properly maintained. I have played a bit with it, and I have mapped MOST of the keys. The system is up and running now with my new keymap, and perfectly usable.
  2.1) Find my keys-es.h copied to the end, in case it is helpful for anybody.
  2.2) I am still missing one key... specifically, the Spanish key for <> (https://commons.wikimedia.org/wiki/File:KB_Spanish.svg), which lives between the LEFT-SHIFT and the Z. Is there any way to dump the raw scan codes of the keyboard to the console (something similar to Linux `showkey`)?
  2.3) Three characters are still missing from the Spanish layout (https://commons.wikimedia.org/wiki/File:KB_Spanish.svg): the dot · which is on SHIFT-3, the tilde ' that is exactly right to the Ñ and the double-dot-tilde ¨ (in Spanish we say "diéresis", but I do not know the name in English) which lives on SHIFT-tilde (directly right to the Ñ, but shifted). Honestly, I can live without them, I am just curious if this can be solved: they are very very used in Spanish natural language, but obviously I am not going to use this laptop to write documents.

Follows `keys-es.h`
```
/* Keymap:ES:Espanol:Spanish	*/

#ifndef __KEYMAP_ES__
#define __KEYMAP_ES__

#if defined(CONFIG_KEYMAP_ES)

/* Spanish Keyboard add by Nacho Martin. imartin@cie.uva.es */

static unsigned char xtkb_scan[84] = {
    0, 033, '1', '2', '3', '4', '5', '6',           /* 00-07*/
    '7', '8', '9', '0', '\'', 173, '\b', '\t',      /* 08-0F*/
    'q', 'w', 'e', 'r', 't', 'y', 'u', 'i',         /* 10-17*/
    'o', 'p', '`', '+', 015, 0202, 'a', 's',        /* 18-1F*/
    'd', 'f', 'g', 'h', 'j', 'k', 'l', 164,         /* 20-27*/
    '<', 167, 0200, 135, 'z', 'x', 'c', 'v',         /* 28-2F*/
    'b', 'n', 'm', ',', '.', '-', 0201, '*',        /* 30-37*/
    0203, ' ', 0204, 0241, 0242, 0243, 0244, 0245,  /* 38-3F*/
    0246, 0247, 0250, 0251, 0252, 0205, 0210, 0267, /* 40-47*/
    0270, 0271, 0211, 0264, 0265, 0266, 0214, 0261, /* 48-4F*/
    0262, 0263, 0272, 0177                          /* 50-53*/
};

static unsigned char xtkb_scan_shifted[84] = {
    0, 033, '!', '\"', ' ', '$', '%', '&',
    '/', '(', ')', '=', '?', 168, '\b', '\t',
    'Q', 'W', 'E', 'R', 'T', 'Y', 'U', 'I',
    'O', 'P', '^', '*', 015, 0202, 'A', 'S',
    'D', 'F', 'G', 'H', 'J', 'K', 'L', 165,
    '>', 166, 0200, 128, 'Z', 'X', 'C', 'V',
    'B', 'N', 'M', ';', ':', '_', 0201, '*',
    0203, ' ', 0204, 0241, 0242, 0243, 0244, 0245,
    0246, 0247, 0250, 0251, 0252, 0205, 0210, 0267,
    0270, 0271, 0211, 0264, 0265, 0266, 0214, 0261,
    0262, 0263, 0272, 0177
};

static unsigned char xtkb_scan_ctrl_alt[84] = {
    0, 033, '|', '@', '#', '~', '5', '6',
    '7', '8', '9', '0', '<', '>', '\b', '\t',
    'q', 'w', 'e', 'r', 't', 'y', 'u', 'i',
    'o', 'p', '[', ']', 015, 0202, 'a', 's',
    'd', 'f', 'g', 'h', 'j', 'k', 'l', 164,
    '{', '\\', 0200, '}', 'z', 'x', 'c', 'v',
    'b', 'n', 'm', ',', '.', '-', 0201, '*',
    0203, ' ', 0204, 0241, 0242, 0243, 0244, 0245,
    0246, 0247, 0250, 0251, 0252, 0205, 0210, 0267,
    0270, 0271, 0211, 0264, 0265, 0266, 0214, 0261,
    0262, 0263, 0272, 0177
};

static unsigned char xtkb_scan_caps[84] = {
    0, 033, '1', '2', '3', '4', '5', '6',
    '7', '8', '9', '0', '\'', 173, '\b', '\t',
    'Q', 'W', 'E', 'R', 'T', 'Y', 'U', 'I',
    'I', 'P', '`', '+', 015, 0202, 'A', 'S',
    'D', 'F', 'G', 'H', 'J', 'K', 'L', 165,
    '<', 248, 0200, 128, 'Z', 'X', 'C', 'V',
    'B', 'N', 'M', ',', '.', '-', 0201, '*',
    0203, ' ', 0204, 0241, 0242, 0243, 0244, 0245,
    0246, 0247, 0250, 0251, 0252, 0205, 0210, 0267,
    0270, 0271, 0211, 0264, 0265, 0266, 0214, 0261,
    0262, 0263, 0272, 0177
};

#endif

#endif

```

ghaerr, 2023-04-10:
Hello @jsedanoj,

> I do not know if after the issue has been closed someone will read it.

I closed the issue thinking the issue was resolved, sorry. It is now reopened and will remain so until the Spanish keyboard is working correctly for you. I suspect the major issue is that the Spanish keyboard has not been well maintained. I don't think there are differences in the scan codes for the AT vs XT keyboard, so I propose replacing keys-es.h with your version going forward.

> Is there any way to dump the raw scan codes of the keyboard to the console (something similar to Linux showkey)?

Follows is a patch that will display the scan code for any key pressed. You will have to recompile the kernel for this to work.
File elks/arch/i86/drivers/char/kbd-scancode.c:
```
static void keyboard_irq(int irq, struct pt_regs *regs)
{
    static int E0Prefix = 0;
    int code, mode;
    int E0key = 0;
    unsigned int key;
    int keyReleased;

    /* read XT or AT keyboard*/
    code = kb_read();

    printk("[%03o]", code);  // <--- add this line to display scan code in octal
    if (kraw) {
        Console_conin(code & 255);
        return;
    }
```

> Three characters are still missing from the Spanish layout 

You should be able to find these missing scan codes using the method above, then insert them into the xtkb_scan_shifted[] table above.

If you'd like to submit a PR once you have the keyboard working to your satisfaction, please do. Otherwise, post the source to keys-es.h here as you did above and I'll be glad to.

Thank you!

jsedanoj, 2023-04-10:
Thanks.

If I can fix it with your instructions, I'll do a PR.

For problem number 2.2, I'll use your debug mod (printk("[%03o]", code);) to dump the codes. I'll report back ASAP.

However, for problems 2.3, I do not think your instructions apply; because the problem is not that I do not know the scancode of the key... I do know it, it is 0x28 for ´, shifted 0x28 for ¨, and shifted 0x04 for · . The problem is that I cannot find the corresponding ASCII character for those Spanish symbols. I mean:

´ as in áéíóú as in "Pedro Sánchez" (our current Prime Minister, https://es.wikipedia.org/wiki/Pedro_S%C3%A1nchez)

and

¨ as in ü as in pingüino (https://es.wikipedia.org/wiki/Spheniscidae).

Right now, if you check the code above, you'll see I mapped them respectively to <, > and SPACE... because... well... I needed to type _something_.

ghaerr, 2023-04-10:
> The problem is that I cannot find the corresponding ASCII character for those Spanish symbols.

I see. The character display will depend on the PC ROM character set that's part of your video card ROM. For most US systems, that code page is [CP 437](https://en.wikipedia.org/wiki/Code_page_437). Thus, if your system is CP 437, you'll have to pick a glyph from that list. Looking at the CP 437 table, it would appear that á is 0xA0, and that ü is 0x81.

If your character ROM is not CP 437, then we'll need to figure out what code page it has and go from there.

jsedano-emobg, 2023-04-11:
Sorry, I'm a bit lost with CodePages.

Is the CodePage hardcoded in the hardware? In the same SLT 286 laptop I also have MSDOS, and the autoexec.bat says:

```
...
MODE CON CODEPAGE PREPARE=((850) C:\DOS\EGA.cpi)
MODE CON CODEPAGE SELECT=850
...
```

So looks like the CodePage is something selected by the OS when drawing characters. Does ELKS have only 437? Not complaining, only checking.

https://en.wikipedia.org/wiki/Code_page_850 (CodePage for Western Europe, mostly) ... which takes me to... according to this Wikipedia article... maybe I should use 0xEF and 0xF9 (**´** and **¨** in the codepage 850)... this way the keyboard layout would be ok, only missing the right CodePage support... but at least keys-es.h would coherent with Spanish.

Sorry if my questions sound dumb, I am not familiar with this.

About the edit... In real life 2023, nobody uses only the **´** character. What you do is... you type **´**, then you type **a**, and the system converts it to **á** (if you do not have a ). I am sure it does not happen only with Spanish, the same happens for example with the French: **^ a** becomes **â**. If you are using a US keyboard...

[image]

... I guess you can use **` a** very similarly, it should output à... just try it here in the GitHub comment.

However I understand it is not something doable with the ELKS console, this is what you typically do when writing a Word document or an email (although in modern systems you can do this for console as well).

jsedano-emobg, 2023-04-11:
Sorry, jsedano-emobg and jsedanoj are the same person, I just wrote this comment in the company laptop and I was logged as the company github account.

jsedanoj, 2023-04-11:
Here is the PR: https://github.com/jbruchon/elks/pull/1586

For my problem 2.2... I do not think I can add it to the keymap, because the code for the Spanish <> key is 0126, which is 86... but the array is 84 items long. The point is that it is typing _something_ (an "a"), so maybe this https://github.com/jbruchon/elks/blob/master/elks/arch/i86/drivers/char/kbd-scancode.c#L352 is an array-overflow?

For my problem 2.3...
* For ´ and ¨ I used the codes 0xEF and 0xF9, which are the right codes according to CP850 (https://en.wikipedia.org/wiki/Code_page_850), In case we add support for codepages in ELKS in the future. Not a perfect solution, because it dumps weird characters right now using CP437... but that's life, there is no other character in CP437 that matches.
* For · (shift-3)... there is a code 0xFA for that in CP437 https://en.wikipedia.org/wiki/Code_page_437 , so I used it.

In the trip, I've found out that AltGr is not behaving as I expected. I expected Ctrl-Alt and AltGr to be synonyms, and the comment at https://github.com/jbruchon/elks/blob/master/elks/arch/i86/drivers/char/kbd-scancode.c#L336 says the same. But they are not. Ctrl-Alt-something shows the right character, but AltGr-something does not. It is a bit weird for we developers, because in the Spanish layout {}[]\| are all of them in AltGr-something... in the 90s we were used to use Ctrl-Alt-something, but now in 2023 my muscular memory makes me use AltGr-something... dumping a weird character. But I cannot completely understand the code in kbd-scancode.c , so I do not know how much intentional it is.

jsedanoj, 2023-04-11:
> > Is the CodePage hardcoded in the hardware? In the same SLT 286 laptop I also have MSDOS, and the autoexec.bat says:
> 
> On older video cards, the hardware supported only one code page, which was CP 437 for US systems. European systems may have been built using CP 850. Newer systems support replacing the ROM character set, which is what is being done by your DOS autoexec.bat (to CP 850, in this case). ELKS does not currently support a mechanism to replace ROM character glyphs. What this means is that what you're seeing when running DOS may not be the same as when running ELKS, as ELKS will use the startup ROM glyphs only.

Thanks for the explanation. Now that you explain it... I remember creating the glyphs in the 90s to create a car game using plain characters... but I just copied-pasted from the instructions in the tutorial without really understanding what I was doing... maybe this is what I was doing XDDD

> 
> > maybe I should use 0xEF and 0xF9 (´ and ¨ in the codepage 850)... this way the keyboard layout would be ok, only missing the right CodePage support... but at least keys-es.h would coherent with Spanish.
> 
> We could assume that the majority of Spanish keyboard users may also have native CP 850 in their system, but I could only guess. Testing would be the best way. I will leave this decision up to you, since I'm unable to test and don't have a Spanish keyboard or native 850 system.

I am guessing as well... but... I think my ELKS laptop has CP437. Because when I have been creating the keys-es layout, I have been reading the codes in https://en.wikipedia.org/wiki/Code_page_437 and they match what I was creating.

ghaerr, 2023-04-12:
> I do not think I can add it to the keymap, because the code for the Spanish <> key is 0126, which is 86... but the array is 84 items long.

It appears the tables were coded for the original 83-key IBM keyboard, thus 84 entries including the unused scan code 0.

> The point is that it is typing something (an "a"), so maybe this https://github.com/jbruchon/elks/blob/master/elks/arch/i86/drivers/char/kbd-scancode.c#L352 is an array-overflow?

Yep - on a Spanish keyboard, typing <> will result in an an array overflow lookup, and undefined behavior! That's a bug, since your keyboard has more than 83 scan codes, and should be fixed.

A fix for both these issues would be to increase the size of every table for every language by three bytes to 87. For all but Spanish, they would be 0-padded. You're welcome to do this in #1586, or submit a separate PR for this fix. What would you map this key to?

> I've found out that AltGr is not behaving as I expected. I expected Ctrl-Alt and AltGr to be synonyms

It seems the keyboard driver tries to map AltGr to Ctrl-Alt in the following code:
```

        /* Step 4:
         * Perform additional special processing for:
         * ALT_GR                       -> CTL-ALT
         * NUM & code >= KP7            -> SHIFT        handle 10 keypad when numlock
         * ALT-code                     -> code | 0x80
         * CTL-code                     -> code & 0x1f
         * Zero table entries           -> '@'
         */
        if (!mode && (ModeState & ALT_GR))  // <-- map ALT_GR to CTRL-ALT
            mode = 3;           /* CTRL-ALT-.. */
```
However, some code above that handles that as a special case only for DE, SE and FR keyboards which apparently treats an E0-prefixed ALT key as AltGr. It is possible that this is not taking place on your ES keyboard, as for some reason, `defined(CONFIG_KEYMAP_ES)` is commented out:
```
    /* --------------Process status keys-------------- */
    if (!(mode & 0xC0)) {               /* Not a simple scancode or fnkey*/
#if defined(CONFIG_KEYMAP_DE) || defined(CONFIG_KEYMAP_SE) || defined(CONFIG_KEYMAP_FR) /* || defined(CONFIG_KEYMAP_ES) */
        if ((mode == ALT) && E0key)     /* ALT_GR has a E0 prefix*/
            mode = ALT_GR;
#endif
```
I would suggest adding that back in to see if that fixes this new issue.

Let me know when you're ready to commit #1586.

jsedanoj, 2023-04-12:
Cool, I'll do that and report back.

Looks like the comment-out of the CONFIG_KEYMAP_ES was made by @georgp24 in https://github.com/jbruchon/elks/commit/bf2e816aa60c8ec3e31e73bf1333f3d308f9a95d ... but it was never uncommented, so maybe Georg just didn't have a Spanish keyboard to test.

jsedanoj, 2023-04-13:
The <> is already added to other keyboards (https://github.com/jbruchon/elks/blob/master/elks/arch/i86/drivers/char/KeyMaps/keys-se.h#L25 or https://github.com/jbruchon/elks/blob/master/elks/arch/i86/drivers/char/KeyMaps/keys-de.h#L26), so I am just adding it to the Spanish layout.

jsedanoj, 2023-04-13:
The PR is ready now

* I have added the <> to the Spanish layout... but without changing anything else, because other layouts already had the same modification.
* I have enabled AltGr (by uncomenting the already-existing code)... and tested it... and well... it works... so I do not understand why it was commented out.

For me, it is ready to merge

closed 2023-04-17 by ghaerr