ghaerr/elks #1587: ELKS works fine on a 1988 Epson PX-16 laptop with 256 KB of memory

Saddamko, 2023-04-12:
I built an ELKS 0.6.0 image in VirtualBox 7.0.
Ubuntu 22.04.2 LTS, a16-elf-gcc (GCC) 6.3.0 as instructed from https://github.com/jbruchon/elks
There were no crashes or errors during the build, everything compiled successfully.
In make menuconfig, there I was removed everything, related to the network and disabled buffers.
This allowed booting  (root or toor) and even running something on a laptop with 256 KB of memory and two FDDs, the second of which is a GOTEK FDD emulator on a USB drive.
Works fine with FAT12 drives.
Also tried compiling several programs. It looks like this:
/home/vladimir/gcc-ia16/elks/cross/bin/ia16-elf-gcc factor.c -o factor -melks-libc -mcmodel=small
Everything works, thanks for your work, it's  funny!
Please close this issue :)
http://www.vintagecomputer.net/fjkraan/comp/px16/
[image]
[image]
[image]

ghaerr, 2023-04-12:
Hello @Saddamko,

Very nice news to hear ELKS built and ran successfully on your Epson PX-16! And thank you for the pictures, they help show what is happening. I can see that with only 256K RAM, the /etc/rc.sys fails because there's not enough room for /bin/sh, but logging in as toor and running the smaller standalone shell works.

Thank you!

Saddamko, 2023-04-14:
No problem with that. Lack of memory prevents many commands from running, but as an experiment, this is very cool.

closed 2023-04-14 by Saddamko