ghaerr/elks #1589: Support for UMB memory

mirevall, 2023-04-19:
Hi,

I discovered elks recently and, as a fan of both very old pc compatibles and linux, it is like a dream come true for me ! Thanks for the great work ! It works wonderfully on my Ericsson Portable PC that has a 3c509 ethernet card.  -> https://imgur.com/a/EOXEkFh
Unfortunately this computer only has 512k of ram, which is not much when using the ethernet daemons, and I am thinking of buying a lo-tech 1Mb card to upgrade it.

I came across issue #709 where you were discussing UMB support with details on how to implement it, and I don't think it was done in the end. I gave it a shot and came up with this -> https://github.com/mirevall/elks
It seams to work well, on 86Box at least, as I don't have any way to test it on real hardware for now. -> https://imgur.com/a/h4ILdU5

Right now the segments to add are hard coded in malloc.c which is of course very temporary, but I don't know the best way to handle this :
- new config file in /etc?
- new entry in /bootops ?
- compile time config ?

To be honest I haven't looked at the feasibility or complexity of any of it, I wanted to have your opinion on this (or even if UMB support is worth adding, but given how little work is required I think it is)

mirevall, 2023-04-20:
Thank you for your quick response !

>I haven't fully studied your code yet, but it seems we'll also need the ability to deallocate RAM back into the freelist, as well as allow any size allocations within a UMB block area. I'm not sure whether your code does that or not yet.

I don't think anything needs to be added for that to work, It just needs to verify that two free segments are contiguous before merging them, so I added this check.

>(Another idea would be, that instead of "adding" main memory and then UMB memory into the freelist, we add a contiguous 1M RAM into the freelist, then "subtract" memory by permanently allocating "holes" where the EGA video or BIOS is 

That is actually a good idea, as it would solve the last problem I see with my current implementation (there are probably some more that I don't see), which is that if a segment is added from A000 to B000, the memory management will not realize that it is contiguous with conventional memory and treat it as a completely separate segment (from what I understand, using A000 to B000 is possible if you only have CGA graphics, because that's where EGA memory is)
However I am not a fan of the "substract" approach because we would have information about the available memory, and this information would need to be converted to "holes" where memory is not available. The "reserved" segments would also probably show up in meminfo, which would be confusing I think.
So in the end I chose the "add segments" method.

>Supporting something like a comma separated list of hex address/lengths in a /bootopts string(s) would be pretty simple, but the config mechanism doesn't handle lists well.

I really like the idea of changing the config "on the fly" when the machine is upgraded, rather than having to recompile the OS. The implementation looks a lot cleaner too. So I added a new entry in bootopts that uses the syntax ```umb=0xC000:0x800,0xD000:0x2000,...``` to specify the start and size of each segment to add.

I also fixed a little bug : when the first segment in conventional memory was freed, it was merged with the next one (the rest of conventional memory) but not flagged as free, so it ended up as huge segment used by no one :
[image]
I think this never append before because the buffer would be here and never de-allocated, but now the buffer tends to be in UMB because the segments are shorter and therefore more 'appealing' to the algorithm.

I tried "stress testing" the current implementation by using as much memory as possible, everything seams to run fine, but tell me if you see any weird behavior !
To test UMB on emulation, the only thing I found is 86box with the Xi8088 system. This config should work with UMB from C000 to C800 and from D000 to E000 : [86box_config.zip](https://github.com/jbruchon/elks/files/11290979/86box_config.zip)

For the problem with the RAM at A000, it is pretty tricky because the segments need to be in the right order in the list or else they can't be merged, and I also don't have a way to test this on emulation (86box can't put memory here) so I will wait for my card to arrive so I can test that on real hardware (It will be here mid may if ebay is to be believed). I don't think it is a big deal for now, it is a pretty unusual configuration and it should still work, just not optimally.

Sorry for the big wall of text, and thank you for your willingness to explain things to a newcomer !

ghaerr, 2023-04-21:
> I am not a fan of the "substract" approach because we would have information about the available memory, and this information would need to be converted to "holes" where memory is not available.

I'd have to think a bit more about how the normal "hole" would be specified, that is A000:0 to FFFF:0, especially when UMB will be specified as "present", not a hole. But the way you've got this currently implemented seems to work well and is quite small, with the A000 problem exception for the time being. We could just document that in the wiki.

> The "reserved" segments would also probably show up in meminfo, which would be confusing I think.

That's a good point, and I agree that reserved segments should not be shown.

I suppose its just the little details like we're discussing that stopped this from quickly being implementing as it seems like a pretty simple solution... which brings us to the problem of bugs and testing:

> I also fixed a little bug : when the first segment in conventional memory was freed, it was merged with the next one (the rest of conventional memory) but not flagged as free, so it ended up as huge segment used by no one:

Wow, very nice catch!!! That is indeed a nasty bug. I've always wondered whether the memory allocation/free routines actually fully worked, as we've never had test cases written for them on a unit-test basis. Was this fixed by just the two lines 142-143?

> I think this never append before because the buffer would be here and never de-allocated, but now the buffer tends to be in UMB because the segments are shorter and therefore more 'appealing' to the algorithm.

I was guessing that reason is that the UMB segments were added last, and they might be possibly checked first and the buffers were thus allocated in UMB memory. I don't think the allocation algorithm tries to best-fit, but instead first-fit (I need to check on that though).

> So I added a new entry in bootopts that uses the syntax umb=0xC000:0x800,0xD000:0x2000,... to specify the start and size of each segment to add.

Very nice work in `parse_umb`, thanks! I like that you've parsed the entire string with so little code. And agree that runtime options are superior to compile-time options. For some embedded systems, /bootopts may not be available, but frankly we probably don't need to worry about that now as most embedded systems have a single linear chunk of RAM available, not having to emulate the IBM PC's problems.

> For the problem with the RAM at A000, it is pretty tricky because the segments need to be in the right order in the list or else they can't be merged, and I also don't have a way to test this on [...]

I see. I hadn't thought of the problem of the umb= segments having to be ordered to work well, but that's not necessarily a show stopper, as long as we document it somewhere. Agreed we should test on real hardware. 

Some other thoughts on getting A000 RAM working: check the ELKS kernel boot line `Kernel text at ... top 9fc0:0` (on my QEMU emulation), this varies across systems, but notice that the available RAM doesn't actually go to A000 but stops at 9FC0, a few paragraphs short of 9FFF:000F. So a merge wouldn't be possible anyways. Perhaps the `mm_init(...,end)` end value could be checked and something done to handle the special case where UMB is specified at A000. 

Given the progress you've made, you're welcome to submit a PR now and I'll merge it so you don't have to come back to this after a few months. It'd be really great to have UMB support thanks to you before the 0.7.0 release, and unless umb= is specified in /bootopts, nothing really is changed (and one bad bug fixed :)

Thank you!

closed 2023-04-21 by ghaerr