ghaerr/elks #1593: Help setting up slip

jsedanoj, 2023-05-08:
Hello,

I have a 286 with dual boot DOS + ELKS (actually triple boot +Minix). I have configured a Raspberry Pi Zero W as SLIP server with NAT and DNS... so now my 286 has WiFi!! (kind of) [by the way, the next project is to actually embed the raspberry Pi Zero W, which is very small, _into_ the laptop, there is some space inside... but that's another story]

I know the raspberry and the hardware are working because I have used MSDOS mTCP to "browse" the internet and even run an HTTP server on the 286. So the Raspberry is Ok, the null-modem cable is Ok, etc

But ELKS still does not work.

I have configured /etc/net.cfg to use SLIP. Changes from the default:
localhost=192.168.2.2 (the same IP I use in MSDOS)
gateway=192.168.2.1 (the IP of the raspberry)
netmask:255.255.255.0
link=slip
baud=115200

`net show` shows: `ip 192.168.2.2 gateway 192.168.2.1 mask 255.255.255.0 slip 115200 /dev/ttyS0` 
`net start` apparently works ok, runs ktcp and starts the daemon telnetd and ftp -d.

Now the errors:

When I ping from the raspberry to ELKS, ELKS dump every second: `IP: BAD CHKSUM (xxxx) len 20`. The error is repeated every second, so I guess it is showing up for every ICMP PING reaching it. If I try `telnet 192.168.2.2` in the raspberry, the same checksum error is show in ELKS. Obviously, the ping is not answered and the telnet is not connecting.

Any idea?

More data: the null-modem is a simple 3 wire cable (GND, TX, RX). So software flow control. Maybe software flow control isn't supported by ELKS? baud rate is 115200... maybe only some baud rates are supported by ELKS?

Any other idea?

jsedanoj, 2023-05-08:
Ups!

I have just found this https://github.com/jbruchon/elks/blob/master/elks/arch/i86/drivers/char/serial-8250.c#L232

```
 * Fast serial driver for slower machines. Should work up to 38400 baud.
```

Looks like 38400 is actually the fastest possible for ELKS. I have reduced the speed of the SLIP link to 38400 and now it works... still with a lot of checksum errors... but works. I have reduced it to 19200 and now it does not show any checksum error (which, by the way, seems to make telnet go faster... I suppose less errors means less TCP retransmissions, so higher final transfer rate).

closed 2023-05-08 by jsedanoj

ghaerr, 2023-05-08:
Hello @jsedanoj,

I was just reading your initial post when I saw you found an answer. I was going to suggest slowing the baud rate down, and it looks like that was a problem. The real issue has to do with how fast your 286 can handle interrupt-driven incoming data and not drop anything. The kernel stores incoming characters into a ring buffer, and it can also happen that if the system is real slow, and the TCP/IP stack and or receiving application haven't had enough time to run, the incoming character queue can overflow as well, which will end up looking like a bad packet.

ELKS has its standard serial driver, which ends up reading the UART using a standard ELKS interrupt handler, which involves saving lots of registers, switching to the kernel stack, checking for signals, etc. Lots of overhead. There is a 'fast' handler that skips all standard overhead, and runs as fast as possible. This can be enabled by editing `elks/include/arch/ports.h` and uncommenting `CONFIG_FAST_IRQ4` for COM1, then recompiling the kernel. There isn't a `make menuconfig` way of configuring it other than manually.

I would suggest you try this, and you might end up being able to run SLIP at 38400 or higher. ELKS itself can allow baud rates up to 11500, but as you've seen, the hardware is the limiting factor.

Let me know if you need more help.

Thank you!

jsedanoj, 2023-05-11:
Reporting: after the tuning you suggested, it works without error up to 38400 and with some of BAD CHEKSUM errors at 57600.

ghaerr, 2023-05-11:
Hello @jsedanoj,

> after the tuning you suggested, it works without error up to 38400 and with some of BAD CHEKSUM errors at 57600.

Thanks for letting us know. Max 38400 baud matches with what I was able to get out of ELKS on a 1987 vintage Compaq Portable 386 without losing data, when CONFIG_FAST_IRQ4 was set. Not sure why DOS can get 115200, other than it and mTCP must have lower overhead than ELKS.

Thank you!