ghaerr/elks #1598: Porting Lotus 123 to ELKS

toncho11, 2023-05-14:
Apparently the Unix version of Lotus 123 has been patched to work on Linux (a contemporary Linux).

How is done: https://lock.cmpxchg8b.com/linux123.html

Github: https://github.com/taviso/123elf

@ghaerr do you think the patches to the Linux version can be adapted for ELKS?

ghaerr, 2023-05-14:
@toncho11,

> do you think the patches to the Linux version can be adapted for ELKS?

Linux is a 32 (and 64)-bit system, and ELKS only 16 bit, with a max program size of 64-128k. From the following comment, referencing a 1.2Mb object file for Lotus 123 on UNIX, the size was so big it seemed to shock even the guys recovering Lotus for UNIX:

> Yikes - it’s an original unstripped object file from 1-2-3. There are nearly 20,000 symbols including private symbols and debug information.

> That 123.o file is huge, even compressed it had to be split across two disks. Let’s take a closer look…
> Why would Lotus ship this? It’s so big it must have required them to phyiscally ship an extra disk to every customer?

Of course, a versio of Lotus 123 ran on DOS, but I'm guessing it had to use some tricky overlay-loading mechanisms to get it to all fit within 640K.

toncho11, 2023-05-14:
yes, it says "file format elf32-i386"

Damn it :)

closed 2023-05-14 by toncho11