ghaerr/elks #1601: Cross-compiling with gcc-ia16

Vutshi, 2023-05-20:
Hi everyone,

I have a naive question regarding compiling with gcc-ia16.

**Description**
- What is expected ?
Compiling a programme according to the [example](https://htmlpreview.github.io/?https://raw.githubusercontent.com/jbruchon/elks/master/Documentation/html/user/writing_apps_in_C.html)
- What is observed ?
```
ia16-elf-gcc hello.c -o hello -melks-libc -mcmodel=small
hello.c:1:18: fatal error: stdio.h: No such file or directory
 #include<stdio.h>
                  ^
compilation terminated.
```

**Configuration**
- I am trying to cross compile on macOS.

**How to reproduce ?**
- gcc-ia16 was successfully build according to [build steps](https://github.com/jbruchon/elks/blob/master/BUILD.md)

ghaerr, 2023-05-20:
Hello @Vutshi,

`gcc-ia16` needs to be pointed to the location of the C library, you can add something like `-I/path_to_elks/libc/include` to the command line.

For a complete external Makefile to use for cross-compiling ELKS programs check out my [File Manager](https://github.com/ghaerr/fm/tree/master) project, which has its own ELKS [Makefile](https://github.com/ghaerr/fm/blob/master/Makefile.elks).

Thank you!

Vutshi, 2023-05-20:
Hi @ghaerr,

> `gcc-ia16` needs to be pointed to the location of the C library, you can add something like `-I/path_to_elks/libc/include` to the command line.
> For a complete external Makefile to use for cross-compiling ELKS programs check out my [File Manager]

After adding the necessary paths I got the following error:
```
ia16-elf-gcc hello.c -o hello -melks-libc -mcmodel=small -I../libc/include -I../include -I../elks/include
ia16-elf-gcc: error: need runtime-specs from C library to link programs
```
Thank you

ghaerr, 2023-05-20:
Hello @Vutshi,

> ia16-elf-gcc: error: need runtime-specs from C library to link program

I'm not sure why this is, perhaps `ia16-elf-gcc` needs to be in the path, as is setup by `env.sh` in the ELKS tree, specifically the `~elks/cross/bin` directory.

Hello @tkchia,

Can you comment on the reason for this error?

Thank you!

tkchia, 2023-05-20:
Hello @Vutshi, hello @ghaerr,

`gcc-ia16` needs to see the `r-elks.spec` file from the ELKS project, which says how to find ELKS's C library and related files.

Doing a `make libc` inside the ELKS tree should fix this.

Hello @ghaerr,

Should we explicitly document the `make libc` command in `BUILD.md`?

Thank you!

ghaerr, 2023-05-20:
Hello @tkchia,

> Should we explicitly document the make libc command in BUILD.md?

Sure, I'll add such a statement, for those that build the toolchain but don't build ELKS, which normally builds libc and the .spec file.

Thank you!

Vutshi, 2023-05-21:
Hi @ghaerr and @tkchia,

I ran `make libc` from elks directory and in the end got the following error:
```
install: /Users/x2241064/Install/8088/ELKS/elks/include/autoconf.h: No such file or directory
make[2]: *** [install] Error 71
make[1]: *** [install] Error 2
make: *** [libc] Error 2
```
Just to remind you I do it in macOS, so there might be some differences.

Best

tkchia, 2023-05-21:
Hello @Vutshi,

Oops.  I forgot that an extra step is needed.

Try running `make defconfig` to set up a "default" ELKS system configuration, then run `make libc`.

(We should probably modify the build process some time, so that a `make defconfig` / `make menuconfig` / ... step is not needed to build the C library.)

Thank you!

ghaerr, 2023-05-21:
Hello @Vutshi, hello @tkchia,

I would like to confirm that the toolchain installation and ELKS build is actually working... @Vutshi, can you run `make` at the ELKS top-level? This should both build libc as well as ELKS. If that works, there should be no problems with the C library etc as has been discussed previously on this thread.

>  Should we explicitly document the make libc command in BUILD.md?
> (We should probably modify the build process some time, so that a make defconfig / make menuconfig / ... step is not needed to build the C library.)

Perhaps, although since the C library is essentially built into the ELKS build, perhaps the instructions should just state to run `make` from the ELKS top level before attempting to use the toolchain as a cross-compiler? This would avoid possibly confusing extra instructions for a "standalone" cross-compiler which is actually tied into the ELKS build, at least presently.

Thank you!

Vutshi, 2023-05-21:
Hi @ghaerr and @tkchia,

> I would like to confirm that the toolchain installation and ELKS build is actually working... @Vutshi, can you run `make` at the ELKS top-level? This should both build libc as well as ELKS. 

I can not run `make` successfully in macOS. Apparently, there are some differences with Linux. Instead I tried to build libc and succeeded to some extent.

First, I ran `make defconfig` and got errors (probably macOS related):
```
make defconfig
rm -f .config
/Users/x2241064/Install/8088/ELKS/elks/Make.defs:10: ELKS is not configured yet
/Library/Developer/CommandLineTools/usr/bin/make -C config all
/Library/Developer/CommandLineTools/usr/bin/make -C lxdialog all
find: /lib/: No such file or directory
find: /lib64/: No such file or directory
find: /usr/lib64/: No such file or directory
find: /usr/local/lib64/: No such file or directory
-e 
>> Unable to find the Ncurses libraries.
>>
>> You must have Ncurses installed in order
>> to use 'make menuconfig'

make[3]: *** [local-curses.h] Error 1
make[2]: *** [all] Error 2
make[1]: *** [kconfig] Error 2
make: *** [defconfig] Error 2
```
After that, I looked inside Makefile and found this line `config/Configure config.in` which allowed me to configure ELKS. Now `make libc` has executed successfully so I can almost compile things with gcc-ia16. Although this time I get the following error during compilation:
```
ia16-elf-gcc hello.c -o hello -melks-libc -mcmodel=small -I../libc/include -I../include -I../elks/include
ia16-elf-gcc: error trying to exec 'elf2elks': execvp: No such file or directory
```
Best

tkchia, 2023-05-21:
Hello @Vutshi,

Thanks for your update!  It looks like we may also need to tweak the ELKS makefiles to separately build the `elf2elks` utility &mdash; not quite sure how exactly to do that yet.  `elf2elks`'s source code is currently under `elks/tools/elf2elks` and is normally compiled together with the kernel.

Thank you!

Vutshi, 2023-05-21:
Hi @tkchia,

> `elf2elks` is needed to turn your `hello` program from an ELF format file into something that ELKS can run. Its source code is currently under `elks/tools/elf2elks/` and it is normally compiled together with the kernel.

I see. So this is why I got a strange message running the hello program in ELKS :)
[image]

ghaerr, 2023-05-21:
Hello @vutshi, Hello @tkchia,

> So this is why I got a strange message running the hello program in ELKS :)

Yes, `ia16-elf-gcc` by default produces ELF output files, while ELKS uses its own a.out format, which is MINIX-compatible. The `elf2elks` program, which is built when running `make` from the ELKS top-level directory, does this.

> It looks like we may also need to tweak the ELKS makefiles to separately build the elf2elks utility — not quite sure how exactly to do that yet.

I think the whole cross-compilation problem, using `ia16-elf-gcc` to cross-compile for ELKS, is most easily solved by just adding a comment in BUILD.md that one should run `make` from the top-level ELKS directory after building the toolchain from `build.sh`. Then all the dependencies will be automatically handled. If `ia16-elf-gcc` were installed or built outside of ELKS, it might make sense to try to add those dependencies to an external build, but not when the C library and `elf2elks` are themselves part of ELKS.

When @vutshi first posed this issue, I hadn't guessed that ELKS itself hadn't been built, I am sorry.

I will add a statement to BUILD.md saying that to complete a cross-compilation installation of `ia16-elks-gcc`, one needs to also configure and build ELKS by running `make config` then `make` after `tools/build.sh`, because the C library and other tools are contained within ELKS.

Thank you!

Vutshi, 2023-05-21:
@ghaerr, @tkchia thank you guys. I managed to build `elf2elks` and now cross-compilation works. To achieve this on macOS I used an old `.config` because it could not be created automatically. After that `make all` managed to build `elf2elks` before it generated an error. 
Case closed. I will open another issue about macOS related problems. 
Cheers.

closed 2023-05-21 by Vutshi

ghaerr, 2023-05-21:
@Vutshi: Glad to hear its now working. I develop only on macOS, and everything should work. I'd definitely like to hear about errors though. Just now, I deleted .config and then ran `make defconfig` which worked fine... with no errors, so there's something funny happening on your system.

> After that make all managed to build elf2elks before it generated an error.

Can you show that error? That might help determine why you can't build on macOS.

ghaerr, 2023-05-21:
> /Users/x2241064/Install/8088/ELKS/elks/Make.defs:10: ELKS is not configured yet
/Library/Developer/CommandLineTools/usr/bin/make -C config all
/Library/Developer/CommandLineTools/usr/bin/make -C lxdialog all
find: /lib/: No such file or directory
find: /lib64/: No such file or directory
find: /usr/lib64/: No such file or directory
find: /usr/local/lib64/: No such file or directory
-e 
>> Unable to find the Ncurses libraries.
>>
>> You must have Ncurses installed in order
>> to use 'make menuconfig'

If this is the error you're getting, try using Homebrew to install the ncurses library. I believe the command is something like `brew install ncurses` or `brew install libcurses-dev`.

tkchia, 2023-05-21:
Hello @ghaerr,

I believe the `ncurses` dependency is mainly for building the `lxdialog` colour user interface, for use with `make menuconfig`.  This is probably not _really_ required for `make config` or `make defconfig`.  Thank you!

ghaerr, 2023-05-21:
Hello @tkchia,

> This is probably not really required for make config or make defconfig.

Good point and I think you're right. Let me take a look and see how that dependency might be removed, thank you!

Vutshi, 2023-05-22:
Hi @ghaerr,

You are right, this is exactly the error message I see in macOS:
> If this is the error you're getting, try using Homebrew to install the ncurses library. I believe the command is something like `brew install ncurses` or `brew install libcurses-dev`.

I do have `ncurses` installed. Just in case, I have reinstalled it using Homebrew and following the brew advice I did `echo 'export PATH="/usr/local/opt/ncurses/bin:$PATH"' >> ~/.zshrc`. Now I have three versions available:
```
ncurses5.4-config  ncurses6-config    ncursesw6-config 
```
the native one, home-brew, and anaconda versions, respectively. ELKS still doesn't like them )

Best

ghaerr, 2023-05-23:
Hello @Vutshi,

Looking at ~elks/config/lxdialog/Makefile, it appears that it jumps through hoops (but not enough apparently) to create a "local" version of curses.h to complete the compilation required for `make menuconfig`:
```
# Check that ncurses is available.
# Write a custom header for curses.

local-curses.h:
    @x=`find /lib/ /lib64/ /usr/lib/ /usr/lib64/ /usr/local/lib/ /usr/local/lib64/ -maxdepth 2 -name 'libncurses.*'` ;\
    if [ ! "$$x" ]; then \
        echo -e "\007" ;\
        echo ">> Unable to find the Ncurses libraries." ;\
        echo ">>" ;\
        echo ">> You must have Ncurses installed in order" ;\
        echo ">> to use 'make menuconfig'" ;\
        echo ;\
        exit 1 ;\
    fi
    printf > local-curses.h '%s\n' \
        '/* Include curses as available on this system.' \
        ' * Automatically created during compilation.' \
        ' * DO NOT CHANGE.' \
        ' */' ''
    printf '#include %s\n\n' $(CURSES_LOC) >> local-curses.h
```
As if that silliness is not enough, the same Makefile then uses `pkg-config ncurses --libs` to generate the appropriate LDFLAGS.

Perhaps the Makefile should drop the longwinded creation of a local-curses.h and instead run `pkg-config ncurses --cflags` to get the header location (although this doesn't solve the problem of including curses.h vs ncurses.h).

Can you try running `pkg-config ncurses --cflags` (and possibly other options like --libs) to determine whether this might solve the build issue? I am interested to know where you headers might actually be for the three libraries...

@tkchia's statement that we should remove the dependency of ncurses altogether for `make config` and `make defconfig` is an even better idea. To do that, make the following change to ~elks/Makefile:
```
config: elks/arch/i86/drivers/char/KeyMaps/config.in kconfig <--- remove 'config' from this line
    config/Configure config.in
```
Then try running `make config`. This hopefully will result in the system building without the need for ncurses. If this works, I can make the change in the ELKS master Makefile.

Thank you!