ghaerr/elks #1603: [feature] Embedded flash update using "romprg"

cocus, 2023-05-21:
Hi,
You might recall I wanted to update my board's flash using some kind of app instead of having to go thru the hassle of removing the flash chip, placing it on a TL866 clone, flashing it, and putting it back on the board. Well, some time ago I found a solution for this, which is "[romprg](https://github.com/tehKaiN/ROMprg)". This project is for an arduino, but with some tweaks, it works under ELKS. I was able to update my flash with it, even if it's kinda slow.

On ELKS, I was running the "arduino" part, which communicates with a host pc thru serial. Then, the host pc sends the appropriate commands to erase/flash/verify the memory.
In my case, I'm hijacking the complete OS by turning off the interruptions, and letting this software talk directly to the UART, not because I'm insane (which I bet I am), but rather because in my particular case, erasing the flash means killing the actual code where ELKS runs off from.

I can contribute this into a new PR, but I'd like to get people's opinion on this. I don't like having to cherry-pick my changes onto master every single time (like this romprg, some tweaks on the /etc/ files, etc), so having this already merged (and pre-selected on the system apps on the ROMFS) would be amazing.

The host-pc app is also found on the same code repo, and I build it externally (it has its own makefile), and when invoked, you get something like:

```
== romprog ping
Firmware version 1.00
  address bits:       18
  max write length:  128
== romprog id
EEPROM chip identification: manufacturer=0x04, device=0xA4
== romprog erase chip
== romprog flash loader
Writing 0x07ff80-0x07fffb
== romprog flash elks image
Writing 0x06ff80-0x06ffff
== romprog flash romfs.bin
Writing 0x05b180-0x05b1c8
```

(note: the lines that start wiith "==" are printed out by my shell script, not the romprg software; also, I'm writing in 3 separated locations (loader, elks image, romfs.bin) to avoid writing a single "long" binary to speed up this process)

ghaerr, 2023-05-21:
Hello @cocus,

Are you suggesting that `romprog` would be just another program built in say, elks/elkscmd/romprog (if multiple files are required) or elks/elkscmd/misc_utils/romprog.c? If so, sure submit a PR and lets add it to the distribution. I'll let you suggest a way to include romprog on the 8018X distribution using 8018x.config. If you can also contribute a man page on it, or at least a README of how to use it where the source lives (or in the source), that would be great :)

It might be nice to also have the host program source available on ELKS; there are several programs which already do this, and are built by chdir to the source directory and using a 'host' prefix for it (such as for ELKS basic running on the host by typing `make hostbasic`.

Thank you!

cocus, 2023-05-21:
> Are you suggesting that `romprog` would be just another program built in say, elks/elkscmd/romprog (if multiple files are required) or elks/elkscmd/misc_utils/romprog.c?

Well, that's a good question, but I think this should live under its own directory, just because it has the "host" application sources as well. Also, I don't see that many users using this for their PC (although it could possibly be extended to flash BIOSes or BIOS extensions memories)

> I'll let you suggest a way to include romprog on the 8018X distribution using 8018x.config.

Yes, I think just changing the defconfig should suffice; I don't think this app should be on any "group" of apps just now.

> If you can also contribute a man page on it, or at least a README of how to use it where the source lives (or in the source), that would be great :)

Indeed, I'll have a look at that. I think I'll just keep everything under a new dir on elkscmd/

> It might be nice to also have the host program source available on ELKS; there are several programs which already do this, and are built by chdir to the source directory and using a 'host' prefix for it (such as for ELKS basic running on the host by typing `make hostbasic`.

Yes, that's really cool. I didn't have a second look at that, I was building the host app manually, but this is the right way to do it.

ghaerr, 2023-05-30:
Same question here @cocus, is this resolved with your implementation in PR #1604?

cocus, 2023-05-30:
Indeed, all working now! Hopefully someone wants to try this on a PC :)

closed 2023-05-30 by cocus