ghaerr/elks #1605: Options from Boot Monitor to ELKS

jsedanoj, 2023-05-27:
Hello,
  (I think the answer is "no", but I lose nothing by asking...)

Is there any way to pass an argument from the Minix Monitor to the ELKS rs.sys script? I would like to automate different tasks depending on the menu option chosen in the Monitor. Maybe setting the runlevel, maybe setting a env, maybe any other hackaround,... any idea?

Thanks,
Javi

ghaerr, 2023-05-28:
Hello @jsedanoj,

I am not familiar with the Minix Monitor, so I'm not sure how this might apply: There is a way to pass "arguments" to the ELKS /etc/rc.sys script - this would be done by passing environment variables of the form ENVSTRING=string, where the environment variable 'ENVSTRING' could be tested within /etc/rc.sys, and a desired action taken.

ELKS has a mechanism that allows creating environment variables and values by just including them in /bootopts, which file is read at boot time. Lines in /bootopts of the form A=B will cause an environment variable A to be created with the value B, and passed to /bin/init, which will get passed in the environment to the subsequent execution of `sh /etc/rc.sys` execution during startup.

It is unlikely that the Minix Monitor will be writing any files such as /bootopts, but the ELKS boot loader actually reads /bootopts into main memory at 0050:0000 at pre-kernel load time. Thus, if there's a way to get the Minix Monitor to write RAM starting at 0050:0000 with the effective contents of a /bootopts file, ELKS will interpret it the same way.

Thank you!

jsedanoj, 2023-05-30:
Minix Monitor cannot do that, as far as I understand from the manpage https://www.unix.com/man-page/minix/8/monitor/ (maybe an option would be to modify the source to do that, but that may be beyond my skills).

:-(

Any other suggestions? Any other boot manager (for 8086/286) that can do that? I have:

-DOS in partition 1
-Minix in partition 2. Active partition. Minix Monitor with a menu; boots hda1, Minix and hda3 (1) respectively.
-ELKS in partition 3.

Other option would be modify the ELKS boot manager to give the capability to show a menu sending different envs to the ELKS kernel [similar to LILO or Grub] (I understand from your message that the sources are in our codebase... maybe here? https://github.com/jbruchon/elks/tree/master/bootblocks). But again, probably beyond my skills.

(1) Minix uses different names, but that's the meaning.

ghaerr, 2023-05-30:
Hello @jsedanoj,

From the Minix Monitor man page, it looks like there are provisions for setting environment variables and boot options; but it seems the Monitor would have to be modified to write into RAM locations 0050:0000 in order to communicate their intention to ELKS, unfortunately.

> Other option would be modify the ELKS boot manager to give the capability to show a menu sending different envs to the ELKS kernel 

This could work, although there is no space left in the standard ELKS boot blocks, they are within a few bytes of being completely using the 512 bytes maximum. There is a less-used MBR boot manager, which currently displays a `MBR:` prompt and asks for the partition to load, which could possibly be modified to write options to RAM as discussed above. It's source is at https://github.com/jbruchon/elks/blob/master/bootblocks/mbr.S. 

Thank you!