ghaerr/elks #1608: Guidelines for writing device driver

johnathancn, 2023-06-08:
I was reading the doc page for writing kernel mode driver which ends with this disclaimer:
"Since the ELKS kernel is almost 64k in size now, currently there is only very limited space available for additional device drivers."
Is this still a concern? Is this 64k assuming a very complete set of config options that can be trimmed down to make room for additional drivers? 

The disclaimer adds:
"User mode device drivers should be considered as an alternative. "
Is there any documentation for writing user mode device drivers? Or existing examples that can be used as reference?

ghaerr, 2023-06-08:
Hello @johnathancn,

It seems the doc page you have been reading is out of date - we added the ability to build the kernel in "medium mode" a couple of years ago, which allows up to 128k in its code segment. The existing config options allow creating a kernel with all drivers present, as well as small kernels that reside in less than 64k code, so this is no longer the concern it once was.

Also, presently there really isn't a defined way to write user mode device drivers in ELKS without a lot of work. The TCP/IP implementation is in user mode, but uses a specific kernel mode implementation of a mechanism to pass requests back and forth. While this was designed in the days long before a 128k kernel became available, I would not recommend trying to write a user-mode device driver presently.

We'd be happy to help with any pointers for a kernel device driver you may want to implement, just ask.

Thank you!

johnathancn, 2023-06-10:
Thanks for the quick reply

johnathancn, 2023-06-13:
Reopening in case my follow up questions above were missed.

ghaerr, 2023-06-13:
> Given medium model should support multiple code segments, is the 128k limit coming from the toolchain or something about the kernel itself?

It's the toolchain, but the kernel has to also of course load anything produced by the toolchain. `ia16-elf-gcc` compiles every function into a separate segment in medium model, but the linker combines them into one of two 64k segments, thus the limitation. We're only at 75k now total, so no problem really.

> By the way this is the doc pages I was referring to:
https://htmlpreview.github.io/?https://github.com/jbruchon/elks/blob/master/Documentation/index.html

> Are those pages still maintained? Is there any intention of moving the developer facing docs into this github repo?

The HTML you referenced only references documentation that is on this repo, but there has not been much contribution or enhancement to it for some time. I try to keep it up when making changes or for very technical memory layouts, but haven't had time to go through most of it, and its in various incompatible formats. For the past several years, much more interest and updates have been made to the Github Wiki documentation pages.

closed 2023-06-13 by johnathancn